MitchBradley/cforth #9: voc-link, undefined

ellerh, 2016-10-16:
Trying to load order.fth leads to this error:
```
C Forth  Copyright (c) 2008 FirmWorks
ok fload ../../src/cforth/order.fth

voc-link, ?
Error at:    voc-link, |  
ok
```
Maybe I made some mistake during building, but grepping for `voc-link,` doesn't match anything.

MitchBradley, 2016-10-16:
I committed the fix as https://github.com/MitchBradley/cforth/commit/b02ea8cf43681a4468988d85e606922a751ba9cc

Just FYI, a lot of the code in cforth is derived from another of my Forth systems, which is present as the core of https://github.com/MitchBradley/openfirmware .  voc-link, is present in its kernel.

ellerh, 2016-10-17:
> I committed the fix as b02ea8c

Thanks!

> Just FYI, a lot of the code in cforth is derived from another of my Forth systems, which is present as the core of https://github.com/MitchBradley/openfirmware . voc-link, is present in its kernel.

I see. I guess it's the same issue with `find`; it's not defined in cforth but in openfirmware's kernel.fth.

ellerh, 2016-10-17:
Just in case you care about order.fth.
This example causes an error:
```
C Forth  Copyright (c) 2008 FirmWorks
ok fload ../../src/cforth/order.fth
ok wordlist set-current
ok : foo ;
Address exception
```