MitchBradley/cforth #10: missing file "makeit"

leptonix, 2016-10-27:
Trying to build for the ESP8266 I get the error:

(cd ........./nodemcu.firmware && sh makeit)
sh: 0: Can't open makeit
../../src/app/esp8266/targets.mk:101: recipe for target 'nodemcu-fw' failed
make: *** [nodemcu-fw] Error 127

I have build nodemcu, also flashed it to an ESP, and that works, but I can't find the script 'makeit' in nodemcu needed to link app.o to an image

https://github.com/MitchBradley/cforth/blob/6b29926df83b109b41bfc2426da3de2a2f548f4b/src/app/esp8266/targets.mk#L100

MitchBradley, 2016-10-27:
You will need some patches to the nodemcu code base.  I'll post them soon.

The patches omit the Lua stuff from the build and substitute Forth.  Long term, I plan to eliminate the dependency on the nodemcu tree, but that won't happen immediately.

The makeit script, included in the patch set, is very simple.  It just calls "make" after adding the xtensa compiler and the correct version (2.7) of Python to PATH.

MitchBradley, 2016-10-27:
Oh, and since you have successfully built the unmodified nodemcu, that implies that you have already sorted out the PATH stuff in your environment, so "makeit" just reduces to "make".  But you will need the patches anyway, in order to call Forth instead of Lua.

leptonix, 2016-10-27:
Ok, thanks.

quozl, 2016-10-27:
I've got my fork of nodemcu on github with makeit added, see https://github.com/quozl/nodemcu-firmware

leptonix, 2016-10-27:
@quozl

Thanks.

Linking works now, only I get the warning:
Warning: ELF binary has undefined symbol c_sprintf

I flashed the image to an ESP, but it keeps resetting:

    ets Jan  8 2013,rst cause:2, boot mode:(3,6)

    load 0x40100000, len 23808, room 16 
    tail 0
    chksum 0x6c
    load 0x3ffe8000, len 908, room 8 
    tail 4
    chksum 0x1a
    load 0x3ffe838c, len 8, room 4 
    tail 4
    chksum 0x7b
    csum 0x7b

I heard you talking about cforth on the podcast embedded. Was very interesting.

quozl, 2016-10-27:
Hmm, doesn't reproduce for me.  My local branch is based on Mitch's latest public commit b02ea8cf, and my only changes are for building on Ubuntu.  You might look at nodemcu/app/mapfile to see how it compares to mine?  http://dev.laptop.org/~quozl/z/1bzqfi.txt

leptonix, 2016-10-27:
I added "#define c_sprintf os_sprintf" to "sdk-overrides/include/c_types.h". Not where it should be, but it fixes the link warning for now. Unfortunately, the ESP still keeps resetting.

quozl, 2016-10-27:
Contents of the high flash pages can cause this; the SDK start function triggers an exception, and doesn't seem to be happy with some data.  I've had to fix it once or twice by either erasing flash or reflashing specific versions of nodemcu.  I've not yet fully appreciated the cause.  I'll mail you my built binaries; if they also reset, then the high flash pages are your immediate cause.

leptonix, 2016-10-27:
Thanks, the binaries work (CForth built 2016-10-27 19:42 from 41f323e-clean).

After that, I tried again to upload the binaries I compiled, but they still don't work.

I flashed it with
esptool.py write_flash -fm dio -fs 32m -ff 40m 0x00000 bin/0x00000.bin 0x10000 bin/0x10000.bin 0x3e000 /opt/esp-open-sdk-1.5.2/esp_iot_sdk_v1.5.2/bin/blank.bin 0x7e000 /opt/esp-open-sdk-1.5.2/esp_iot_sdk_v1.5.2/bin/blank.bin

quozl, 2016-10-27:
Great, so whatever the problem is you have isolated it to your build.  Progress.  Tell us about your build chain?  Mine is Ubuntu 14.04 on 64-bit with esp-open-sdk b10756d25aae6f6836050bd99aa38e5b27281454.  Probably out of date, haven't checked.

leptonix, 2016-10-28:
I am using Ubuntu 16.04 64-bit. My version of esp-open-sdk was older. Now I have installed the same version (b10756) as you. But the result is then same, it keeps resetting.

The only thing I changed is disabling "CFLAGS += -m32" in build/esp8266/Makefile. If that is enabled, it doesn't compile (/usr/include/features.h:367:25: fatal error: sys/cdefs.h: No such file or directory)

quozl, 2016-10-28:
My build has `CFLAGS += -m32` so I've had a quick look at what packages on my system have `sys/cdefs.h`; best match by atime of build is `/usr/include/sys/cdefs.h`, so please check you have libc6-dev-i386 installed?

leptonix, 2016-10-28:
Thanks, that fixes it. I installed libc6-dev-i386, recompiled with -m32 flag, flashed it to an ESP and it works now (CForth built 2016-10-28 08:42 from c046014).

closed 2016-10-28 by leptonix

quozl, 2016-10-28:
Yay!

MitchBradley, 2016-10-28:
I can explain why it fails without -m32.  The base Forth dictionary is built on the host system and must have the same cell size as the target.  If you omit -m32 on a 64-bit host, the host Forth will have 64-bit cells, whereas the ESP8266 is a 32-bit machine.

MitchBradley, 2016-10-29:
I just pushed a commit that automates the process of getting the right version of the nodemcu-firmware tree and patching it for CForth use.   src/app/esp8266/README.md explains how it works and what the patches do.