MitchBradley/cforth #12: [MAC] Build issues

arkrost, 2016-12-01:
Hi!

I'm having troubles with building cforth under mac.
```
gcc --version
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 7.3.0 (clang-703.0.31)
Target: x86_64-apple-darwin15.6.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

If I'm trying to build bluez64 target I got:
```
~/Documents/cforth/build/bluez64 ❯❯❯ make
CC ../../src/cforth/extend.c
../../src/app/bluez/extend.c:279:2: error: use of undeclared identifier 'res'
        res = ft_open_serial(portnum, 0x4e4c);  // Nod Ring
        ^
../../src/app/bluez/extend.c:280:6: error: use of undeclared identifier 'res'
        if (res)
            ^
../../src/app/bluez/extend.c:281:10: error: use of undeclared identifier 'res'
                return res;
                       ^
../../src/app/bluez/extend.c:283:2: error: use of undeclared identifier 'res'
        res = ft_open_serial(portnum, 0x4e4d);  // Nod Backspin
        ^
../../src/app/bluez/extend.c:284:6: error: use of undeclared identifier 'res'
        if (res)
            ^
../../src/app/bluez/extend.c:285:10: error: use of undeclared identifier 'res'
                return res;
                       ^
6 errors generated.
make: *** [extend.o] Error 1
```

If I'm trying to build host-serial-macos64 target I got the following.
```
~/Documents/cforth/build/host-serial-macos64 ❯❯❯ make
MAKING FORTH
CC main.o io.o nullbi.o dictfile.o mallocl.o lineedit.o linux-kbd.o glops.o sha256.o ftdi.o forth.o compiler.o syscall.o floatops.o extend.o -L/usr/local/Cellar/libusb/1.0.20/lib -lusb-1.0 -lobjc -Wl,-framework,IOKit -Wl,-framework,CoreFoundation -L/usr/local/lib -lglfw -framework Cocoa -framework IOKit -framework CoreFoundation -framework CoreVideo -o forth
cc -g -O -DBITS64 -DFLOATING -DMOREFP -DMAXDICT=0x800000 -DOPENGL -I/usr/local/Cellar/libusb/1.0.20/include/libusb-1.0 -I/usr/local/include -I. -I../../src/cforth -I../../src/app/host-serial -I../../src/lib -I../../src/app/host-serial/libftdi   -DNOSYSCALL -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=0 -DUSE_FTDI -o forth main.o io.o nullbi.o dictfile.o mallocl.o lineedit.o linux-kbd.o glops.o sha256.o ftdi.o forth.o compiler.o syscall.o floatops.o extend.o -L/usr/local/Cellar/libusb/1.0.20/lib -lusb-1.0 -lobjc -Wl,-framework,IOKit -Wl,-framework,CoreFoundation -L/usr/local/lib -lglfw -framework Cocoa -framework IOKit -framework CoreFoundation -framework CoreVideo
Undefined symbols for architecture x86_64:
  "_glBegin", referenced from:
      _glop in glops.o
  "_glBindBuffer", referenced from:
      _glop in glops.o
  "_glBindTexture", referenced from:
      _glop in glops.o
  "_glBufferData", referenced from:
      _glop in glops.o
  "_glCallList", referenced from:
      _glop in glops.o
  "_glClear", referenced from:
      _glop in glops.o
  "_glClearColor", referenced from:
      _glop in glops.o
  "_glClearDepth", referenced from:
      _glop in glops.o
  "_glColor3d", referenced from:
      _glop in glops.o
  "_glColor4d", referenced from:
      _glop in glops.o
  "_glColorPointer", referenced from:
      _glop in glops.o
  "_glDepthFunc", referenced from:
      _glop in glops.o
  "_glDisable", referenced from:
      _glop in glops.o
  "_glDisableClientState", referenced from:
      _glop in glops.o
  "_glDisableVertexAttribArray", referenced from:
      _glop in glops.o
  "_glDrawArrays", referenced from:
      _glop in glops.o
  "_glDrawElements", referenced from:
      _glop in glops.o
  "_glEnable", referenced from:
      _glop in glops.o
  "_glEnableClientState", referenced from:
      _glop in glops.o
  "_glEnableVertexAttribArray", referenced from:
      _glop in glops.o
  "_glEnd", referenced from:
      _glop in glops.o
  "_glEndList", referenced from:
      _glop in glops.o
  "_glFlush", referenced from:
      _glop in glops.o
  "_glFrontFace", referenced from:
      _glop in glops.o
  "_glFrustum", referenced from:
      _glop in glops.o
  "_glGenBuffers", referenced from:
      _glop in glops.o
  "_glGenTextures", referenced from:
      _glop in glops.o
  "_glGetBooleanv", referenced from:
      _glop in glops.o
  "_glGetDoublev", referenced from:
      _glop in glops.o
  "_glGetFloatv", referenced from:
      _glop in glops.o
  "_glGetIntegerv", referenced from:
      _glop in glops.o
  "_glInitNames", referenced from:
      _glop in glops.o
  "_glLightf", referenced from:
      _glop in glops.o
  "_glLightfv", referenced from:
      _glop in glops.o
  "_glLighti", referenced from:
      _glop in glops.o
  "_glLightiv", referenced from:
      _glop in glops.o
  "_glLineWidth", referenced from:
      _glop in glops.o
  "_glLoadIdentity", referenced from:
      _glop in glops.o
  "_glLoadName", referenced from:
      _glop in glops.o
  "_glMaterialfv", referenced from:
      _glop in glops.o
  "_glMateriali", referenced from:
      _glop in glops.o
  "_glMaterialiv", referenced from:
      _glop in glops.o
  "_glMatrixMode", referenced from:
      _glop in glops.o
  "_glNewList", referenced from:
      _glop in glops.o
  "_glNormalPointer", referenced from:
      _glop in glops.o
  "_glOrtho", referenced from:
      _glop in glops.o
  "_glPixelStorei", referenced from:
      _glop in glops.o
  "_glPointSize", referenced from:
      _glop in glops.o
  "_glPopMatrix", referenced from:
      _glop in glops.o
  "_glPopName", referenced from:
      _glop in glops.o
  "_glPushMatrix", referenced from:
      _glop in glops.o
  "_glPushName", referenced from:
      _glop in glops.o
  "_glRenderMode", referenced from:
      _glop in glops.o
  "_glRotatef", referenced from:
      _glop in glops.o
  "_glScaled", referenced from:
      _glop in glops.o
  "_glSelectBuffer", referenced from:
      _glop in glops.o
  "_glShadeModel", referenced from:
      _glop in glops.o
  "_glTexCoord2f", referenced from:
      _glop in glops.o
  "_glTexImage2D", referenced from:
      _glop in glops.o
  "_glTexParameteri", referenced from:
      _glop in glops.o
  "_glTexParameteriv", referenced from:
      _glop in glops.o
  "_glTranslated", referenced from:
      _glop in glops.o
  "_glVertex2d", referenced from:
      _glop in glops.o
  "_glVertex2i", referenced from:
      _glop in glops.o
  "_glVertex3d", referenced from:
      _glop in glops.o
  "_glVertex3i", referenced from:
      _glop in glops.o
  "_glVertex4d", referenced from:
      _glop in glops.o
  "_glVertex4i", referenced from:
      _glop in glops.o
  "_glVertexAttribPointer", referenced from:
      _glop in glops.o
  "_glVertexPointer", referenced from:
      _glop in glops.o
  "_glViewport", referenced from:
      _glop in glops.o
  "_gluLookAt", referenced from:
      _glop in glops.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [forth] Error 1
```
But I'm sure I have glfw3 installed. Moreover I've tried to build glfw3 from sources.

quozl, 2016-12-01:
For the bluez64 error, my patch is https://github.com/quozl/cforth-1/commit/a4ca717def4b3709f4ca87303f403dc6efb3d6a1 also referenced [here](http://quozl.us.netrek.org/cforth-milo-champions-band/).

arkrost, 2016-12-01:
Oh, I see. Can you suggest me the way how to build the minimal version of cforth, i.e. without glfw, bluez ant etc?

MitchBradley, 2016-12-01:
The bluez builds only work on Linux.  MacOS and Windows have different APIs for Bluetooth.

Sorry about all that OpenGL cruft in host-serial-macos64.  It was for some factory diagnostics at a previous job, for a product that didn't make it to market.  To build without it for macos64, replace build/host-serial-macos64/Makefile with the following simplified version.  Let me know if it works; I can't test it because I no longer have a Mac.

# Builds a MacOS host Forth application with serial port tools
all: default
TOPDIR=../..
CONFIG += -DBITS64
CONFIG += -DFLOATING -DMOREFP
include $(TOPDIR)/src/app/host-serial/targets.mk

MitchBradley, 2016-12-01:
Haha, looks like I got marked down.  The line "Builds a MacOS host Forth ..." starts with a '#', i.e. it's a Makefile comment, but Markdown treated it as a heading.

quozl, 2016-12-01:
Heh.  Been there.  You probably know by now, but if not the trick is to enclose lines in triple backticks (```);

```
# Builds a MacOS host Forth application with serial port tools

all: default
TOPDIR=../..
CONFIG += -DBITS64
CONFIG += -DFLOATING -DMOREFP
include $(TOPDIR)/src/app/host-serial/targets.mk
```

I've some old Macs from a few years ago, but nothing recently upgraded.  I can offer ESP32 testing, but haven't had time to port yet.

Reviewed 9a96d1f54987ed86268d69caa6103fdc725cb806 and 0d00d26cdaece84a420912ae8304eff08c4d84d5.  Like 'em.

arkrost, 2016-12-02:
I confirm that it compiles fine with this makefile. Thank you!

closed 2016-12-02 by arkrost