MitchBradley/cforth #16: Problem with postpone

ellerh, 2016-12-03:
This example leads to an error:
```
shell> cat test.fth
: bar postpone if postpone then ;
: foo [ bar ] ;
shell> cforth test.fth 
Stack Underflow
```

MitchBradley, 2016-12-03:
The system is in interpretation state after **[**, so when **BAR** executes **IF**, there is an undefined situation according to the standard, as the interpretation semantics of **IF** are undefined.

In CForth (and also in my native Forth that underlies Open Firmware), when you execute **IF** (or **BEGIN** or **DO**) in interpretation state, a temporary definition is created and the system switches to compilation state.  When the control structure is completed, that definition is closed and executed, and the system returns to interpretation state.  This lets you run full-speed control structures on the command line in one step, without the syntactic overhead of explicitly creating a definition.  It's an extremely useful feature; I use it very frequently in hardware debugging scenarios.  It's entirely "legal" under the standard, because of the undefined interpretation semantics for control flow words.

If **BAR** were immediate, you could use **BAR** inside **FOO** without the brackets and thus compile the control structure inside **FOO**.  That formulation would be a standard program.

ellerh, 2016-12-03:
> The system is in interpretation state after [, so when BAR executes IF, there is an undefined situation according to the standard, as the interpretation semantics of IF are undefined.

Well, I disagree with that interpretation of the standard. IMO, it's obvious that when BAR is called that it should execute the compilation sematics of IF (that was appended to BAR when POSTPONE IF was compiled).

Closing this as there's no point in language lawyering.

closed 2016-12-03 by ellerh