MitchBradley/cforth #20: teensy 3.2

schallee, 2016-12-27:
Greetings,

I am trying to build for the teensy 3.2 using the arduino 1.6.11 + teensyduino 1.30 tool chain (on osx). I can successfully compile and run quozl's arm-teensy3 branch (ba48256599897973c17544d4c4c2b637945eb915) but I am having issues building MitchBradley/cforth (8d3e1e6aeae8cc500902f2c1e1210be1559f9849) for the teensy.

The default compile target compiles with warnings and produces app.o:

[make_default_target.txt](https://github.com/MitchBradley/cforth/files/673608/make_default_target.txt)

Attempting to compile app.elf (which I believe to be the next step) fails as it depends on start.o which is nowhere to be found:

[make_app_elf.txt](https://github.com/MitchBradley/cforth/files/673612/make_app_elf.txt)

There also does not appear to be any target for app.hex to use with teensy loader. I can certainly objcpy as needed but a make target would be useful;)

Thanks!

quozl, 2016-12-27:
Reminds me I should really merge my [old branch](https://github.com/quozl/cforth/commits/arm-teensy3) with @MitchBradley's later work.

quozl, 2016-12-27:
@schallee, I think I see your problem now ... my aim was a standalone source tree with everything needed to build and flash, which is why I've an `app.hex` target, as well as `mk20dx128.c` and `mk20dx256.ld`.  But probably @MitchBradley's aim was to have Teensy and CForth source trees side by side on a system, and have the Teensy source tree bring in app.o from the CForth source tree.

I've rebased my old branch on 8d3e1e6, tested, and pushed to; [quozl/cforth-1 arm-teensy3](https://github.com/quozl/cforth-1/tree/arm-teensy3)  Next is merging @MitchBradley's Teensy work and mine to figure out if there's anything for me to propose as a pull-request.  A quick look at the `ccalls` shows there's more functionality in @MitchBradley's Teensy work.

MitchBradley, 2016-12-27:
James is correct; I was doing the
        top-level build in PJRC's Teensy subtree of the Arduino tree.
        
        I like what I see in quozl/cforth-1
          arm-teensy3 
      
      I'm looking forward to the pull request when James is ready.
      
      
      On 12/27/2016 3:29 AM, James Cameron wrote:
      
    
      @schallee,
          I think I see your problem now ... my aim was a standalone
          source tree with everything needed to build and flash, which
          is why I've an app.hex target, as well
          as mk20dx128.c and mk20dx256.ld.
          But probably @MitchBradley's
          aim was to have Teensy and CForth source trees side by side on
          a system, and have the Teensy source tree bring in app.o from
          the CForth source tree.
      I've rebased my old branch on 8d3e1e6, tested, and pushed
          to;  Next is merging @MitchBradley's
          Teensy work and mine to figure out if there's anything for me
          to propose as a pull-request. A quick look at the ccalls
          shows there's more functionality in @MitchBradley's
          Teensy work.
      —
            it on GitHub, or mute
            the thread.

quozl, 2016-12-28:
Thanks.

I've rebased against the upstream platform code,
https://github.com/quozl/cforth-1/commits/arm-teensy3

I've then squashed several commits and made a pull request, #21.  Works for me, but I may have broken something of yours, Mitch!

MitchBradley, 2016-12-28:
Very nice.  Merged.
        
        On 12/28/2016 4:52 AM, James Cameron wrote:
      
    
      Thanks.
      I've rebased against the upstream platform code,
        
        https://github.com/quozl/cforth-1/commits/arm-teensy3
      I've then squashed several commits and made a pull request,
        #21.
          Works for me, but I may have broken something of yours, Mitch!
      —
            it on GitHub, or mute
            the thread.

schallee, 2016-12-29:
Thanks! I'll give it a try when I have a chance here.

schallee, 2016-12-29:
Compiles and loads fine on my teensy 3.2. 

BTW: Though a beginner at forth and little time for hobbies, I have a hankering to try to port to the Atmel Samd21 at some point. Adafruit has ported micro python to it so most of the boiler plate could be taken from there. I'd be curious to know your thoughts on such.

Thanks again!

quozl, 2016-12-29:
Thanks for the test on Teensy 3.2.

Atmel SAMD21 would not be difficult.  Sparkfun has some lower priced boards than the Adafruit ones.  A quick porting method is to write a sketch to blink an LED, then enable verbose building in the Arduino IDE, and look at the linker map and use the GCC ARM objdump to disassemble the result.  Then copy the critical code from somewhere.