MitchBradley/cforth #23: Documentation ... (is there any?)

paraplegic, 2017-01-06:
Hi Mitch,

I stumbled onto Cforth and I'd like to try it on a couple of Arm boards ... (BBB, Pi etc) in native mode.  I see you have BSP's in the tree for the Teensy and various ARM boards, and I built the bluez version on Debian (64bit) without difficulty.  What I need to progress would be some docs.  I see from one of your threads that the Cforth is a derivative of OpenFirmware ... but I can't find any docs on the supported wordlists, extensions or anything.  Do the docs exist or do I have to inspect code to figure it all out?

Cheers, and thanks.

Rob Sciuk

ps:  I sent you an email with a patch for the OBP Tiny forth separately.

paraplegic, 2017-01-07:
Ahh, I spent some time with the github repository, and stumbled upon the EXCELLANT readme in the forth directory, containing the standard references, the glossary and provenance of each word.  That will most certainly suffice for my purposes.  Also spending some time with the source code, I see why cforth's speed is much higher than my forth, given the optimal nature of the inner interpreter, and the generous use of very efficient primitives in the big loop.  I particularly like the next as continue, since it does not get any faster than that.  Nice.  I look forward to learning more, and I apologize for the noise, I should have studied the code before asking.  (LMGTFY).

closed 2017-01-07 by paraplegic

MitchBradley, 2017-01-07:
Thanks for the kind words.
        
        CForth was intended to be as fast as possible within the
        constraint of portability.  I have done assembly language
        direct-threaded Forths that are marginally faster, at the
        expense of having to write new kernel code for each processor. 
        See https://github.com/MitchBradley/openfirmware/tree/master/cpu
        for some of the kernels (there are others that I no longer
        maintain, including the original 68K and the SPARC one that
        drove a lot of the early development).
        
        You can get even faster by using native compilation with
        peephole optimization across word boundaries, but that has the
        disadvantage of making decompilation and source-level debugging
        much harder.  Subroutine threading, which seems like it should
        be a speed win, turns out not to be a win in and of itself,
        because subroutine calls and returns cause unnecessary memory
        accesses.  But subroutine threading does open the door to native
        code generation, so it can be the starting point for a
        super-fast Forth.  Modern caches and superscalar CPU
        implementation techniques can sometimes "bury" the call/return
        penalty, so it can be difficult to analyze what is going to be
        faster these days.
        
        Modern CPUs are so fast that trying to micro-optimize the Forth
        interpreter often is not worthwhile.  For the things that I
        usually do, involving lots of I/O operations, the access to the
        I/O devices is such a bottleneck that you just don't notice the
        Forth interpreter overhead.
        
        Having said that, you might wonder why CForth is so carefully
        coded for speed.  It's because I originally wrote it back in the
        day when a 10 MHz, single-issue, multi-cycle memory access
        system with 1 MB of memory was considered fast.
        
        On 1/7/2017 7:46 AM, Rob Sciuk wrote:
      
    
      Ahh, I spent some time with the github repository, and
          stumbled upon the EXCELLANT readme in the forth directory,
          containing the standard references, the glossary and
          provenance of each word. That will most certainly suffice for
          my purposes. Also spending some time with the source code, I
          see why cforth's speed is much higher than my forth, given the
          optimal nature of the inner interpreter, and the generous use
          of very efficient primitives in the big loop. I particularly
          like the next as continue, since it does not get any faster
          than that. Nice. I look forward to learning more, and I
          apologize for the noise, I should have studied the code before
          asking. (LMGTFY).
      —
          this thread.
            it on GitHub, or mute
            the thread.

paraplegic, 2017-01-07:
>     <reply+0025a512ca6e339cc82bbb7052b49b0aa0bf391e71b3dd6d92cf000000011488f34
>     State change <state_change@noreply.github.com>
> c0b-11e6-9aed-b52498112777.png","action":{"name":"Open in GitHub","url":"https://github.com/MitchBradley/cforth"}},"updates":{"snippets":[{"icon":"PERSON","message":"@paraplegic in #23: Ahh, I spent some time with the

Likewise for your reply.  As for "back in the day", I'm a bit of a 
greybeard myself, and my first Forth ran on an HP3000 16bit minicomputer, 
so I get the whole "architecting for speed and efficiency" thing.

I will have another patch for CForth (this time) to get rid of CLANG 
warnings and a socket fix for the FreeBSD/Clang platform shortly.

I look forward to getting a X compilation environment going, and getting 
down with some ARM dev boards, and I *JUST* ordered a RISC-V board in an 
arduino package 8-).  THAT should be a hoot.

Back with some patches shortly.

Cheers,
Rob Sciuk

paraplegic, 2017-01-08:
Hi Mitch,

I did a bit of work on cforth on the FreeBSD/clang platform and I've 
attached a patch for a couple of clang nags, and a link problem over a 
missing include.  I created a freebsd_x64 directory as a symbolic link to 
the bluez64 directory, and it seems to just work so long as you use gmake 
rather than the native bsd make.  The symbolic link creation is *NOT* part 
of the patch, so if you apply it, the bluez64 build directory remains 
untouched, and only a a couple of source files are hit.

Speed looks pretty good, and I'll try to get some timings at some point. 
Also, cforth has a robust feeling to it.  Next I'll try to come up to 
speed with your c interface, which is much cleaner than the one I devised. 
Damn, it looks like I'll have to abandon my OneFileForth in favour of 
cforth, and get through the learning curve as rapidly as I can.

I like the see <word> ... I implemented mine as taking an execution token 
on the stack ( ' <word> view ) ... and more importantly, I like how you 
worked directly from the standard such that documentation kind of takes 
care of itself.  My last forth was a fig style which really didn't care 
much for the standards process ...

Thanks for making this available, and your kind reply regarding the 
design decisions you made.

Cheers,
Rob Sciuk
(Paraplegic on github).
rob@controlq.com

-- 
-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
Robert S. Sciuk                                         rob@controlq.com
Principal Consultant                                        905.706.1354
Control-Q Research                  97 Village Rd. Wellesley, ON N0B 2T0
diff --git a/src/cforth/compiler.c b/src/cforth/compiler.c
index ba4e34e..42f2bea 100755
--- a/src/cforth/compiler.c
+++ b/src/cforth/compiler.c
@@ -314,7 +314,7 @@ int num_initial_matches(char *adr, cell len,
 		}
 		if (nmatches == matchnum) {
 		    *namelen = name_len(xt);
-		    *namep = to_name_adr(xt);
+		    *namep = (char *) to_name_adr(xt);
 		}
 		++nmatches;
 		dictp = XT_FROM_CT(*to_link(xt), up);
diff --git a/src/cforth/forth.c b/src/cforth/forth.c
index 349fbf9..a94d88c 100755
--- a/src/cforth/forth.c
+++ b/src/cforth/forth.c
@@ -1237,7 +1237,7 @@ execute:
 
 /*$p file-size */  case FILE_SIZE:  /* fid -- ud ior */
     sp -= 2;
-    tos = pfsize(tos, &sp[0], &sp[1], up);
+    tos = pfsize(tos, (u_cell *) &sp[0], (u_cell *) &sp[1], up);
     next;
 
 /*$p to-ram */  case TO_RAM:
@@ -1264,7 +1264,7 @@ execute:
 
 /*$p file-position */  case FILE_POSITION:     /* fid -- ud ior */
     sp -= 2;
-    tos = pfposition((void *)tos, &sp[0], &sp[1], up);
+    tos = pfposition((void *)tos, (u_cell *) &sp[0], (u_cell *) &sp[1], up);
     next;
 
 /*$p w/o */         case W_O:  push(1);  next;
diff --git a/src/lib/sfc-client.c b/src/lib/sfc-client.c
index 1be6481..e54a424 100644
--- a/src/lib/sfc-client.c
+++ b/src/lib/sfc-client.c
@@ -7,6 +7,9 @@
 #include <sys/socket.h>
 #include <sys/types.h>
 #include <arpa/inet.h>
+#ifdef __FreeBSD__
+ #include <netinet/in.h>
+#endif
 
 
 static int sockfd = -1;

MitchBradley, 2017-01-08:
Thanks, I applied your patches. 
        I'm glad it works on BSD.  I have used the BlueZ stuff on
        Raspberry Pi and Linux; it's nice to know it is reasonably
        portable.  Although I guess I'm not too surprised, since it
        basically only depends on being able to open BTPROTO_HCI and
        BTPROTO_L2CAP sockets.  I guess we can thank other OSs for
        implement the BSD socket interface largely verbatim.  That said,
        
        
        Note that sfc-client.c is unlikely to be useful.  It was an
        interface to a shop floor database server in a factory in China,
        for Forth-based diagnostics of a now-defunct product.  That C
        program was ultimately implemented directly in Forth.
        
        The test engineer for that factory told us that it would be
        impossible for us (my colleague Blake and I) to implement the
        interface to that database server in our Forth diagnostic suite,
        as the only way to talk to it was with a Java program.  As I
        recall, it took us about one afternoon to figure out the network
        protocol and implement it.  You opened a socket to the server
        and sent a single-line request with a message code followed by
        semicolon-separated argument fields, and the server sent back a
        single line response that was either OK or an error message.
        
        I was Vice Chair of the ANS FORTH committee, so I sort of had to
        implement the standard just to make sure it worked.  The first
        version of CForth was heavily influenced by F83.
        
        On 1/7/2017 6:15 PM, Rob Sciuk wrote:
      
    
      
        Hi Mitch,
      
      
      
        I did a bit of work on cforth on the FreeBSD/clang platform and
        I've 
      
        attached a patch for a couple of clang nags, and a link problem
        over a 
      
        missing include. I created a freebsd_x64 directory as a symbolic
        link to 
      
        the bluez64 directory, and it seems to just work so long as you
        use gmake 
      
        rather than the native bsd make. The symbolic link creation is
        *NOT* part 
      
        of the patch, so if you apply it, the bluez64 build directory
        remains 
      
        untouched, and only a a couple of source files are hit.
      
      
      
        Speed looks pretty good, and I'll try to get some timings at
        some point. 
      
        Also, cforth has a robust feeling to it. Next I'll try to come
        up to 
      
        speed with your c interface, which is much cleaner than the one
        I devised. 
      
        Damn, it looks like I'll have to abandon my OneFileForth in
        favour of 
      
[... 370 lines cut ...]
      
        @@ -7,6 +7,9 @@
      
       #include <sys/socket.h>
      
       #include <sys/types.h>
      
       #include <arpa/inet.h>
      
      
        +#ifdef __FreeBSD__
      
      
        + #include <netinet/in.h>
      
      
        +#endif
      
       
       
       static int sockfd = -1;
      
      
            it on GitHub, or mute
            the thread.

paraplegic, 2017-01-08:
> Date: Sun, 08 Jan 2017 00:26:53 -0800
> From: Mitch Bradley <notifications@github.com>
> Reply-To: MitchBradley/cforth
>     <reply+0025a5120251890448f1252be648d63ba7aa83ed5221ca2e92cf000000011489b9c
>     d92a169ce0be13fbb@reply.github.com>
> To: MitchBradley/cforth <cforth@noreply.github.com>
> Cc: Rob Sciuk <rob@controlq.com>,
>     State change <state_change@noreply.github.com>
> 
> Thanks, I applied your patches. 
> I'm glad it works on BSD.  I have used the BlueZ stuff on
> Raspberry Pi and Linux; it's nice to know it is reasonably
> portable.  Although I guess I'm not too surprised, since it
> basically only depends on being able to open BTPROTO_HCI and
> BTPROTO_L2CAP sockets.  I guess we can thank other OSs for
> implement the BSD socket interface largely verbatim.  That said,

Your notes on Raspberry Pi and Linux lead me to the BlueZ64 to begin with, 
so thanks for the hint.  As mentioned, you could conceivably create a 
FreeBSD_x86 build by simply adding a symbolic link to Bluez64.

> 
> 
> Note that sfc-client.c is unlikely to be useful.  It was an
> interface to a shop floor database server in a factory in China,
> for Forth-based diagnostics of a now-defunct product.  That C
> program was ultimately implemented directly in Forth.

I noted from the context the Java dependency, but broken is broken and 
fixed is fixed.  Now the code remains useless, but not broken 8-).

Over the years I've put in a lot of effort to avoid Java, and also Windows 
wherever possible, and managed to eke out a career nonetheless.

> The test engineer for that factory told us that it would be
> impossible for us (my colleague Blake and I) to implement the
> interface to that database server in our Forth diagnostic suite,
> as the only way to talk to it was with a Java program.  As I

Aha!  A challenge.

> recall, it took us about one afternoon to figure out the network
> protocol and implement it.  You opened a socket to the server
> and sent a single-line request with a message code followed by [...]

Nice.

I did a PPC board bring-up on an ATCA chassis a few years back, and I put 
my own forth into U-Boot to decipher a means to interrogate the Chassis ID 
and the slot number of the blade to set the MAC id's on the blade 
according to policy.  I never told anyone about the use of Forth, just 
said it was U-Boot, and quietly got it done.  I was surrounded by 
engineers who were devoted C++ guys, and were freaked out by change (and 
Forth specifically).

> 
> I was Vice Chair of the ANS FORTH committee, so I sort of had to
> implement the standard just to make sure it worked.  The first
> version of CForth was heavily influenced by F83.

I remember that you were on the committee, and F83 had a big part in my 
first introduction to Forth (Laxen&Perry?).  Actually I probably ran a 
cartridge based Forth on a Commodore 64 back "in the day".  I wrote my 
first forth back in 1983, and while it was VERY portable, it was not fast. 
I re-wrote it for embedded, and added the Nintendo DS to my platform list 
(arm 7), and the last version was the OneFileForth on github, which was 
designed, as yours, for education purposes.  Nowhere near as nice as 
cforth.

Thanks
[... 462 lines cut ...]
> CPU\n\u003e implementation techniques can sometimes \"bury\" the call/return\n\u003e penalty, so it can be difficult to analyze what is going to be\n\u003e faster these days.\n\u003e \n\u003e Modern CPUs are so fast that
> trying to micro-optimize the Forth\n\u003e interpreter often is not worthwhile.  For the things that I\n\u003e usually do, involving lots of I/O operations, the access to the\n\u003e I/O devices is such a bottleneck that
> you just don't notice the\n\u003e Forth interpreter overhead.\n\u003e \n\u003e Having said that, you might wonder why CForth is so carefully\n\u003e coded for speed.  It's because I originally wrote it back in the\n\u003e
> day when a 10 MHz, single-issue, multi-cycle memory access\n\u003e system with 1 MB of memory was considered fast.\n\u003e \n\u003e On 1/7/2017 7:46 AM, Rob Sciuk wrote:\n\u003e \n\u003e \n\u003e Ahh, I spent some time
> with the github repository, and\n\u003e stumbled upon the EXCELLANT readme in the forth directory,\n\u003e containing the standard references, the glossary and\n\u003e provenance of each word. That will most certainly
> suffice for\n\u003e my purposes. Also spending some time with the source code, I\n\u003e see why cforth's speed is much higher than my forth, given the\n\u003e optimal nature of the inner interpreter, and the generous
> use\n\u003e of very efficient primitives in the big loop. I particularly\n\u003e like the next as continue, since it does not get any faster\n\u003e than that. Nice. I look forward to learning more, and I\n\u003e
> view\n\u003e it on GitHub, or mute\n\u003e the thread.\n\u003e \n\u003e \n\u003e \n\u003e \n\u003e \n\u003e \n\u003e \n\u003e \n\u003e{\"api_version\":\"1.0\",\"publisher\":{\"api_key\":\"05dde50f1d1a384dd78767c55493e4bb\",\"name\":\"GitHub\"},\"entity\":{\"external_key\":\"github/MitchBradley/cforth\",\"title\":\"MitchBradley/cforth\",\"subtitle\":\"G
> itHubrepository\",\"main_image_url\":\"https://cloud.githubusercontent.com/assets/143418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png\",\"avatar_image_url\":\"https://cloud.githubusercontent.com/assets/143418/158421
[... cut ...]
> pfposition((void *)tos, (u_cell *) \u0026sp[0], (u_cell *) \u0026sp[1], up);\r\n next;\r\n \r\n /*$p w/o */ case W_O: push(1); next;\r\ndiff --git a/src/lib/sfc-client.c b/src/lib/sfc-client.c\r\nindex 1be6481..e54a424
> 100644\r\n--- a/src/lib/sfc-client.c\r\n+++ b/src/lib/sfc-client.c\r\n@@ -7,6 +7,9 @@\r\n #include \u003csys/socket.h\u003e\r\n #include \u003csys/types.h\u003e\r\n #include \u003carpa/inet.h\u003e\r\n+#ifdef
> __FreeBSD__\r\n+ #include \u003cnetinet/in.h\u003e\r\n+#endif\r\n \r\n \r\n static int sockfd = -1;\r\n"}],"action":{"name":"View Issue","url":"https://github.com/MitchBradley/cforth/issues/23#issuecomment-271128648"}}}
> 
> ?