MitchBradley/cforth #31: In case of build with floating point math, the mathlib is not listed in the linker (STM32F103)

ivand58, 2017-09-15:
Observed in **STM32F103** target build:
When the CForth is build with floating point support, the ```-lm``` is missing in the linker flags.

MitchBradley, 2017-09-15:
Do you have a patch?

ivand58, 2017-09-15:
I am still working on it... Currently I am blocking by the RAM size: 

> region `RAM' overflowed by 2340 bytes 

It is linked with:

`arm-none-eabi-gcc -mcpu=cortex-m3 -mthumb -mfloat-abi=soft -o app.elf -T../../src/platform/arm-stm32f103/stm32_flash.ld    tstartup_stm32f10x.o tstm32f10x_usart.o tstm32f10x_rcc.o tstm32f10x_gpio.o tstm32f10x_adc.o tmisc.o tsystick.o ttmain.o mallocembed.o tconsoleio.o  tsystem_stm32f10x.o tgpio.o tadc.o  tembed.o ttextend.o -lc -lgcc -lm -specs=nosys.specs
`