MitchBradley/cforth #34: ESP32 no rule to make app.o

tomberek, 2018-08-21:
I'm still gaining familiarity with this toolchain, not sure yet the correct way to resolve:
```
...
CC build/lwip/netif/slipif.o
CC build/lwip/netif/etharp.o
CC build/lwip/netif/ethernet.o
CC build/lwip/netif/lowpan6.o
CC build/lwip/netif/ethernetif.o
CC build/lwip/port/freertos/sys_arch.o
CC build/lwip/port/netif/wlanif.o
CC build/lwip/port/netif/ethernetif.o
CC build/lwip/port/debug/lwip_debug.o
CC build/lwip/port/vfs_lwip.o
AR build/lwip/liblwip.a
make[1]: *** No rule to make target 'app.o', needed by 'libmain.a'.  Stop.
/home/user/esp32/esp-idf/make/project.mk:472: recipe for target 'component-main-build' failed                                                           
make: *** [component-main-build] Error 2
```
app.o does exist in the cforth/build/esp32 directory, but not in the sdk_build subdirectory.

MitchBradley, 2018-08-21:
My build setup isn't easily accessible at this instant, so I can't say for sure, but I think it is as simple as copying - or perhaps linking - the app.o in the cforth build directory to the esp-idf directory.  As time permits I will dredge up my last build setup and tell you for sure.

MitchBradley, 2018-08-22:
Well, that was tedious.  I finally got the build working again with the latest version of esp-idk.  I ended up having to use an absolute path to app.o; the relative one failed miserably.  Try the latest commit 54d5b1b788 and see if it helps  you.

tomberek, 2018-08-23:
Almost there...
```
Rebooting...                                                                                                                                 
ets Jun  8 2016 00:22:57                                                                                                                     
                                                                                                                                             
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)                                                                                       
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00                                                                     
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4916
load:0x40078000,len:8920
load:0x40080400,len:5844
entry 0x400806f8
I (143) cpu_start: Pro cpu up.
I (144) cpu_start: Single core mode
I (144) heap_init: Initializing. RAM available for dynamic allocation:
I (147) heap_init: At 3FFAE6E0 len 00011920 (70 KiB): DRAM
I (153) heap_init: At 3FFC87B0 len 00017850 (94 KiB): DRAM
I (160) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (166) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (172) heap_init: At 400898DC len 00016724 (89 KiB): IRAM
I (179) cpu_start: Pro cpu start user code
I (196) cpu_start: Starting scheduler on PRO CPU.

CForth built ? from Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.                                      
Core 0 register dump:
PC      : 0x400d67e3  PS      : 0x00060930  A0      : 0x800d2d58  A1      : 0x3ffb2190                                                      
0x400d67e3: get_arg at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/lib$/stdio/vfprintf.c:2274

A2      : 0x00000001  A3      : 0x00000073  A4      : 0x3ff9ac88  A5      : 0x00000000                                                      
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800f4634  A9      : 0x3ffb2380                                                      
A10     : 0x00000001  A11     : 0x3ffc3a84  A12     : 0x00000000  A13     : 0x00000000                                                      
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c                                                      
EXCVADDR: 0x00000001  LBEG    : 0x400d4838  LEND    : 0x400d4851  LCOUNT  : 0x00000000  
```

MitchBradley, 2018-08-23:
The fact that it printed "CForth built" is a very good sign.  I can't look at it right now because it's late and I'm sleepy.  Maybe I can look at it tomorrow, but no promises because Hurricane Lane is bearing down and could knock out our power or worse.

tomberek, 2018-08-23:
I'm in!
```
I (330) heap_init: At 400918B8 len 0000E748 (57 KiB): IRAM
I (336) cpu_start: Pro cpu start user code
I (130) cpu_start: Starting scheduler on PRO CPU.

CForth built 2018-08-23 12:59 from b0d30cf
Type a key within 2 seconds to interact
3  . 2 +
3  . 2 +
6  1 2 3 + + .
1  1 .ub dup + ;
dub isn't unique 1 .
1
c  . dub
```
Looks like an issue with line endings or line-feed. I'm using `make monitor` from the esp sdk. But that's something I should be able to track down when I have a moment. (by the way, thanks for being responsive and working these issues, bravo!)

MitchBradley, 2018-08-23:
I suspect the problem is that the serial terminal application invoked by "make monitor" doesn't handle key-at-time input very well.

After finding where I had hidden my esp32 module :-) I did "make flash" and then hooked up a proper terminal emulator program (I use TeraTerm on Windows and "screen" on Linux and MacOS).  It worked fine:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4956
load:0x40078000,len:8920
load:0x40080400,len:5844
entry 0x400806f8
I (177) cpu_start: Pro cpu up.
I (177) cpu_start: Single core mode
I (178) heap_init: Initializing. RAM available for dynamic allocation:
I (181) heap_init: At 3FFAE6E0 len 00011920 (70 KiB): DRAM
I (187) heap_init: At 3FFCEBF0 len 00011410 (69 KiB): DRAM
I (194) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (200) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (206) heap_init: At 4009179C len 0000E864 (58 KiB): IRAM
I (213) cpu_start: Pro cpu start user code
I (7) cpu_start: Starting scheduler on PRO CPU.

CForth built 2018-08-23 08:13 from 0e22116
Type a key within 2 seconds to interact
ok here .
3ffc8a70
ok see see
: see
   ' ['] (see) catch if
      drop
   then
;
ok : foo 5 + . ;
ok 3 foo
8
ok

```
BTW there is a new commit.  it lets you override the download port via the command line, e.g.

`COMPORT=/dev/ttyUSB0 make flash
`

Lacking an override, the port configured in sdkconfig will be used.  Saves having to rebuild if the port changes.

tomberek, 2018-08-23:
Awesome. That was my guess, but I was heading out the door. Closing this
issue.

closed 2018-08-23 by tomberek