MitchBradley/cforth #35: Compile error for glfw-linux because of duplicate object file

FrankRuben, 2019-02-15:
Hi,

trying to compile the glfw-linux tree, I got the following error:

    glops.o: In function `glop':
    xxx/cforth/build/glfw-linux/../../src/cforth/glops.c:19: multiple definition of `glop'
    glops.o:/xxx/cforth/build/glfw-linux/../../src/cforth/glops.c:19: first defined here

It seems that `glops.o` is added to variable `MYOBJS` twice, once in xxx/cforth/build/glfw-linux/Makefile and then again in xxx/cforth/src/app/glfw/targets.mk, which is included from the first Makefile. As quick fix it worked for me to just use the `sort` function of `make`, which will also remove the duplicate entry. But for someone more knowledgeable with the build could just fix the duplicate addition at the correct place.

For the workaround, edit xxx/cforth/src/cforth/targets.mk and search for `@echo MAKING FORTH`, then change the C-compiler invocation to `@$(CC) $(CFLAGS) -o $@ $(sort $(HOSTOBJS)) $(BASEOBJS) $(LIBS)` (and to have it all nice, also change the `echo` in the line above. `cforth` compiled just fine for me after this change.

Plus - no compile error, just a hint for anyone running into the same issue: compiling on Ubuntu 18.04, I got the error:

    Package libusb-1.0 was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libusb-1.0.pc'
    to the PKG_CONFIG_PATH environment variable
 
Having installed `libusb-1.0` the following invocation of `make` helped: `PKG_CONFIG_PATH=xxx/libusb/lib/pkgconfig/:$PKG_CONFIG_PATH make`.

Thanks for cforth and regards, 
    Frank

MitchBradley, 2019-02-16:
I just pushed a fix for the duplication. Let me know how it works for you.

I also commented out the libusb dependency which was only needed for a particularly arcane test setup that I deployed in a factory in China.

FrankRuben, 2019-02-16:
Hi,
thanks for the quick fix - compile now worked fine without any error.
Regards, Frank