MitchBradley/cforth #36: Connect to Wifi ?

TRiXER88, 2019-03-18:
If I type wifi-sta-on it says it needs a SSID when I type wifi-sta-ssid "SSID" then it says again it needs a SSID. I tried all combinations. maybe you can help me.

Greetings Lukas

quozl, 2019-03-18:
`wifi-sta-ssid` is a `defer`, a vectored execution, analogous to a callback.  You might do this before `wifi-sta-on`;

```forth
:noname  " SSID" ;  to wifi-sta-ssid
```

When not set, the word `must-set-ssid` is executed, which aborts with message _wifi-sta-ssid needs to be set_.

TRiXER88, 2019-03-19:
First of all thanks. I think it worked. But a Question, how can I show the IP Address? And can I connect via Telnet or something?

MitchBradley, 2019-03-19:
The phrase `ipaddr@ .ipaddr` will show the IP address.

There is no canned Telnet server but there is a framework for acting as web server.  serve-sensors.fth is an example of a web service that displays the state of a collection of sensors relevant to a plant-growing system.  The key thing you have to define is a word to plug into the defer word "homepage".  When you browse to port 80, homepage is called emit HTML that will be displayed in the browser.

TRiXER88, 2019-03-19:
Ok thanks. if I Type ipaddr@ .ipaddr I have Ip address 0.0.0.0 :/

TRiXER88, 2019-03-19:
Ok got my IP Address. The serve-sensors file is not working for me but thanks. I try to get a remote connection.

MitchBradley, 2019-03-19:
serve-sensors.fth will not  work unless you have all of those hardware sensors.  Use it as an example.