MitchBradley/cforth #37: Make error

TRiXER88, 2019-05-15:
Hello,

I have an Issue with Making the ESP32 Cforth Build. Sorry for asking if its dumb but Im a newbie with make stuff and so on...

[image]

quozl, 2019-05-15:
> bits/libc-header-start.h: Datei oder Verzeichinis nicht gefunden

File or directory not found.

Looking at my own Ubuntu 18.04 development system, with `locate` and `dpkg -S`, perhaps you haven't installed the `libc6-dev-armel-cross` package?  This is a wild guess, I've not tried to reproduce your problem.

TRiXER88, 2019-05-15:
Hello, 

thanks for your answer... Sadly Its not the Problem. I have just installed the libc6-dev-armel-cross package and it still doesnt work, still the same error message...

TRiXER88, 2019-05-15:
I used your Tutorial to Build it but It wont work.

quozl, 2019-05-15:
I guess you've have to uncover why it doesn't work; I did that tutorial with Ubuntu 16.04 two years ago, and things will have changed since.  What architecture are you running your Ubuntu operating system on?  If it is x86_64, please install the `libc6-dev` package and remove the `libc6-dev-armel-cross` package.

TRiXER88, 2019-05-15:
I found a solution for my problem... I installed the 32bit libs with:

`sudo apt-get install gcc-multilib`

Then I installed Python Pip because it wasnt installed somehow...

then I installed all esp-idf requirements with Python Pip and the requirements.txt from esp-idf

But now I become another error and I dont know what to do...

[image]

quozl, 2019-05-15:
The error you have looks like a problem with the `esp-idf`.  The clone is of HEAD.  My last successful build was bf3093feca1f470bdd6caf45adf3b71492b84a19 in the `esp-idf` repository; years ago, so that gives you a known working baseline.  Mitch may have built it more recently.  You might try a less broken commit of `esp-idf`, or find what the problem is.

When I try it, I get a different error.

(Could you please copy and paste the error rather than use a screen image?  It takes me much longer to handle a screen image, and I can't easily search for errors without first doing an OCR.  Look at the `script` command on Ubuntu for capturing errors.)

TRiXER88, 2019-05-15:
I Try but for some reasons the Copy and Paste from my VM to my Host Maschine dont work ...

TRiXER88, 2019-05-15:
Toolchain path: /home/ubuntu/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Toolchain version is not supported: 8d95cad
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 4.8.5
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
Python requirements from /home/ubuntu/esp-idf/requirements.txt are satisfied.
WARNING: Toolchain version is not supported: 8d95cad
Expected to see version: crosstool-ng-1.22.0-80-g6c4433a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Compiler version is not supported: 4.8.5
Expected to see version(s): 5.2.0
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
CC build/bootloader/soc/src//lldesc.o
/home/ubuntu/esp-idf/components/soc/src/lldesc.c: In function 'lldesc_setup_link':
/home/ubuntu/esp-idf/components/soc/src/lldesc.c:29:35: error: 'NULL' undeclared (first use in this function)
     dmadesc[n - 1].qe.stqe_next = NULL;
                                   ^
/home/ubuntu/esp-idf/components/soc/src/lldesc.c:29:35: note: each undeclared identifier is reported only once for each function it appears in
/home/ubuntu/esp-idf/components/soc/src/lldesc.c: At top level:
cc1: warning: unrecognized command line option "-Wno-frame-address" [enabled by default]
/home/ubuntu/esp-idf/make/component_wrapper.mk:289: recipe for target 'src//lldesc.o' failed
make[3]: *** [src//lldesc.o] Error 1
/home/ubuntu/esp-idf/make/project.mk:582: recipe for target 'component-soc-build' failed
make[2]: *** [component-soc-build] Error 2
/home/ubuntu/esp-idf/components/bootloader/Makefile.projbuild:41: recipe for target '/home/ubuntu/cforth-1/build/esp32/sdk_build/build/bootloader/bootloader.bin' failed
make[1]: *** [/home/ubuntu/cforth-1/build/esp32/sdk_build/build/bootloader/bootloader.bin] Error 2
Makefile:26: recipe for target 'forth.elf' failed
make: *** [forth.elf] Error 2

TRiXER88, 2019-05-15:
Now I installed the newset Build from all stuff I need... Now he gives me this error message when I execute it normal:

[image]

And this If I execute it with sudo because he said he have no permission to do something so i thought sudo is right...

[image]

MitchBradley, 2019-05-15:
I'm sorry that you are having so much trouble.  I just made some improvements and fixes in the Makefiles so that it might "just work".  Try pulling the latest version of cforth and then

```
cd cforth/build/esp32
make

```

TRiXER88, 2019-05-16:
Ok I will try thanks ! Apreciated