MitchBradley/cforth #38: Is there a Way to communicate over WiFi ?

TRiXER88, 2019-05-21:
Hey, is there a way to switch the Terminal from Serial to Wifi ? So that I can connect with Putty or Something and control the ESP32 ? The ESP32 is connected to the WiFi but how to switch the Terminal to WiFi ?

Thanks 

Lukas

MitchBradley, 2019-05-21:
There is a telnet server in my openfirmware tree at https://github.com/MitchBradley/openfirmware/blob/master/ofw/inet/telnetd.fth

It would require some porting to make it work with cforth.  I don't have time to do that right now.

Telnet is the old, non-secure, protocol that people used back when the precursor to the Internet was a friendly place.

At one point I had some code that would let you use a web page as a console.  Type into a text box and the commands get executed on the esp and results displayed back in the text box.  I can't seem to find that code at the moment.

If you just want to send a command or two, instead of a full interactive session, the web server framework has a feature that lets you embed a Forth command in a URL.  The command will be executed and any output will be returned as the web page contents, which the browser will display.  The URL syntax is  "?forth&cmd=WHATEVER".  WHATEVER can be any forth command line, like, say, "here ."

If there are a few specific commands that you want to use a lot, you can make a web page with buttons that send such URLs.

quozl, 2019-05-21:
On the ESP8266 I've a TCP command-line server similar to telnetd, usable with the telnet client on Linux, but with several functional limitations.  It was enough for testing without a serial port.  See https://github.com/quozl/cforth-1/blob/master/src/app/esp8266/telnetd.fth ... it would need rewrite for ESP32 stack.

TRiXER88, 2019-05-22:
Ok, thanks for the Answers... @MitchBradley Thanks for the Code, the Code for the Webserver would be nice to have as well if you find it.

@quozl Thanks but I need it for the ESP32, the ESP8266 is running with forth and telnet at my office

quozl, 2019-05-22:
It should be possible to change it to work on the ESP32.  Just depends on how much time you have.  :grin:

MitchBradley, 2019-05-22:
I was unclear about what I lost.  The specific thing I lost was a text box for the interactive "terminal".  The webserver code that lets you run Forth from URLs is in the tree.  If you look in src/app/esp32/serve-sensors.fth you will see a rather elaborate application that controls a plant growing machine.  It has a bunch of environmental sensor whose values are presented via a web page, and a timer tick reads the sensors and adjusts various valves and pumps accordingly.  The web server setup code is at the end and there are various snippets of HTML interspersed through the file.  If  you study that file you can learn a lot about how the web server setup is supposed to work, and then remove bits until you have just what you need.

TRiXER88, 2019-05-22:
@quozl  Depends on how much skill I have in Forth 🤣

TRiXER88, 2019-05-22:
> I was unclear about what I lost. The specific thing I lost was a text box for the interactive "terminal". The webserver code that lets you run Forth from URLs is in the tree. If you look in [...]

Thanks for the Info... I will have a look !