MitchBradley/cforth #39: Possible problem with number conversion for number > 2^64

Bushmills, 2019-08-13:
After compilation of glfw-linux version of cforth for x86_64 Linux, I started testing number ranges and output. First attempt was -1 u.  which produced the expected and correct output of 2^64-1. Thusly encouraged I went on with  -1. ud. and got 204963821196358365382482626504915156991. While this is a nicely long number, it still struck me as not quite what I was expecting. I therefore did
-1. hex ud., which showed the possible problem in better detail:  1111111111111111ffffffffffffffff.
Examining then #, the culprit appeared to be mu/mod, which i then tested like this:
18446744073709551616.  10 mu/mod  d. .      ( producing 1844674407370955161 6, good )
36893488147419103232. 10 mu/mod  d. .       (producing 1844674407370955161 6, not good )
73786976294838206464. 10 mu/mod  d. .       (producing 1844674407370955161 6, again )
it appears as if mu/mod doesn't take upper half of a double into account.

MitchBradley, 2019-08-13:
Thanks.  I fixed it and will upload the fix after I clean it up.

MitchBradley, 2019-08-13:
Fixed by https://github.com/MitchBradley/cforth/commit/3ca22a64bfdf5720de95a042a711016699b114cb

closed 2019-08-13 by MitchBradley

Bushmills, 2019-08-13:
Call that a quick fix! Thanks.

MitchBradley, 2019-08-13:
Good problem reports result in fast fixes!

MitchBradley, 2019-08-13:
Try https://github.com/MitchBradley/cforth/commit/8b31c6c80590e69d1218de00df4a50cef0e7872b .  I realized that all the difficult double precision arithmetic code is unnecessary now that modern compilers routinely support double-width data types.  Might as well let the compiler do the work, now that C compilers are very very good.  There was a time when that wasn't the case...