MitchBradley/cforth #43: Address exception with pictured number conversion,  host-serial-linux64

Bushmills, 2019-09-24:
Trying to output rather long binary numbers, I encountered a possible buffer overflow when converting more than 100 digits. As the 64 bit versions ought to be able to handle 128 digits, in my humble opinion, that buffer should be large enough to accommodate such a number.
Please observe the issue by executing test2, which causes "Address exception" on my (2nd gen) i7-2640M system, running cforth in the host-serial-linux64 flavour. test1 doesn't exhibit that problem.

    ( binary )   ( irrelevant, shows with any radix )
    : test1  -1. <# #100 0 do # loop #>  2drop ;
    : test2  -1. <# #101 0 do # loop #>  2drop ;

Quick and easy workaround in this case was to output two 64 bit binary numbers, concatenating their output, but I feel that this issue deserves a somewhat more general fix.

MitchBradley, 2019-09-24:
Fixed by the above commit.  Thanks for the report.

closed 2019-09-24 by MitchBradley

Bushmills, 2019-09-24:
Record breaking speed of fixing again! Thank you.

MitchBradley, 2019-09-24:
Again, good reports result in easy fixes.