MitchBradley/cforth #46: Problems transfering sonoff MQTT example for ESP32

meineid-git, 2019-09-27:
Hi,
first I have to say that I am impressed by cforth and all the effort you put into it and the fact that you are sharing this.

I try to transfer the \ESP8266\sonoff MQTT example from ESP8266 to ESP32, but it seems lacking some binding to certain c-routines like tcp-poll (showing up in \ESP8266\tcp-new.fth) with your release version of 2017.

You may say, that I have to recompile the binaries including these routines, but I must admit that:
- I am not sure whether I can do this (or am I completely on the wrong route?)
- I do not want to do it, because it will lead me deeper into a jungle of further obstacles, taking me further away from my actual goal 
- I wanted to ask, if you could eventually provide a new release with that binding, if you consider this to be a useful for other users as well

Your feedback is very much appreciated!

MitchBradley, 2019-09-27:
The TCP interface is different for ESP32, sure to the fact that ESP8266 requires a strict callback approach whereas ESP32 can support a more conventional socket interface.When I have a chance I will make an MQTT version for ESP32. It will probably be sometime next week since I am traveling and do not have access to an ESP32 for testing

MitchBradley, 2019-10-05:
Try this commit 6781153c1638a11a70fb86b69750eda51e04f9a0 . src/app/esp32/mqtt-test.fth shows how to use MQTT on ESP32