MitchBradley/cforth #49: make error 2 for a ESP8266F

Jos-Ven, 2020-01-08:
Hi, In an attempt to try cforth on a ESP8266F

I tried the following:
git clone https://github.com/MitchBradley/cforth 
cd  cforth/build/esp8266
make

The make command ends with:
 inflating: libwpa.a
rmdir /home/pi/esp/nodemcu-firmware/sdk//patch
rm -f /home/pi/esp/nodemcu-firmware/sdk/esp_iot_sdk_v1.5.4.1/lib/liblwip.a
touch /home/pi/esp/nodemcu-firmware/sdk/.patched-1.5.4.1
CC ../../src/cforth/makename.c
CPP ../../src/cforth/forth.c
MAKENAME
TCC ../../src/app/esp8266/consoleio.c
/home/pi/esp/nodemcu-firmware/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gc                                          c: 1: /home/pi/esp/nodemcu-firmware/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-                                          elf-gcc: Syntax error: "(" unexpected
../../src/common.mk:29: recipe for target 'tconsoleio.o' failed
make: *** [tconsoleio.o] Error 2
HPdebian $

How can that error be solved?
Kind regards,
Jos

MitchBradley, 2020-01-08:
You appear to be compiling on a Raspberry Pi.  The compilation setup is for an x86 machine.  I don't know if there is an ESP8266 toolchain that runs on a Pi.

Jos-Ven, 2020-01-08:
Hi, Thank you for your quick answer.

The Debian runs on a DELL PC
By using that PC I was able to install punyforth
See https://github.com/zeroflag/punyforth/wiki/Build-environment-setup

HPdebian $ ls # Shows:
cforth  esp-open-rtos  esp-open-sdk  nodemcu-firmware
'
No idea if that Build-environment-setup influenced the cforth-setup

Kind regards Jos

quozl, 2020-01-08:
Yes, it seems likely that the setup does conflict.  I see the path in the error.

Jos-Ven, 2020-01-08:
What path? What should I do to solve it?
Kind regards Jos

MitchBradley, 2020-01-08:
I was confused by the "/home/pi" in the directory names.  The username pi is the default on raspberry pi, but not often seen elsewhere.

Is your Dell PC a 32-bit system?  The toolchain that is embedded in the copy of esp-open-sdk that nodemcu-firmware uses is for x86_64.

Since you seem to have succeeded in building a toolchain, you could just tell cforth to use it instead of the one that it normally uses.  

```
cd cforth/build/esp8266
XTGCCPATH=~/esp/esp-open-sdk/xtensa-lx106-elf/bin/ make

```

Jos-Ven, 2020-01-09:
Hi, That did the trick.
After putting the ESP8266F into the flash mode
sudo COMPORT=/dev/ttyUSB0 make download # also worked.
After a rebooting the ESP8266F I see through the comport:
___________
d$l ãrl$
CForth built 2020-01-09 10:18 from d348a96-dirty
Type a key within 2 seconds to interact
ok
___________
Many thanks to you all for your time and helping me with the installation problem.
Kind regards, Jos

closed 2020-01-09 by Jos-Ven