MitchBradley/cforth #52: Build fail on arm-stm32f103

andrewtholt, 2020-02-09:
Hi,

I have just acquired a 'blue pill' stm32f103 and am trying to build cforth for this.

My first guess is that the cforth has grown too large to fit in  the 128k of this device.  Any help much appreciated.

When I do I get the following errors:

./forth kernel.dic ../../src/cforth/load.fth
.( isn't unique
./makeccalls <../../src/cforth/extend.c >tccalls.fth
./forth forth.dic tccalls.fth ../../src/app/arm-stm32f103/app.fth
Test for SFR Passed
Test for RAM Passed

gpio-open ?
Error at:    out_pp #13 gpioc gpio-open  |  to led-gpio

gpio-clr ?
Error at: : led-on  ( -- )  led-gpio gpio-clr  |   ;

gpio-set ?
Error at: : led-off ( -- )  led-gpio gpio-set  |   ;

gpio-open ?
Error at:    ain -rot  gpio-open  |  drop     \ Configure the GPIO for analog

adc-open ?
Error at:    1 adc-open  |  to adc            \ Fire up the ADC

adc-start ?
Error at:    adc-time adc-channel adc adc-start |  

adc-done? ?
Error at:    begin  adc adc-done?  |   until

adc-get ?
Error at:    adc adc-get |  
CC ../../src/cforth/embed/makebi.c
./makebi app.dic
TCC ../../src/cforth/embed/rodict.c
TLD tembed.o
TCC ../../src/platform/arm-stm32f103/textend.c
Linking app.elf ...
arm-none-eabi-ld --gc-sections -static -T../../src/platform/arm-stm32f103/stm32_flash.ld tstartup_stm32f10x.o tstm32f10x_usart.o tstm32f10x_rcc.o tstm32f10x_gpio.o tstm32f10x_adc.o tmisc.o tsystick.o ttmain.o mallocembed.o tconsoleio.o  tsystem_stm32f10x.o tgpio.o tadc.o  tembed.o ttextend.o -L"/usr/lib/gcc/arm-none-eabi/6.3.1/thumb/v7-m/" -lgcc -o app.elf
arm-none-eabi-ld: section .ARM.exidx LMA [000000000800be54,000000000800be5b] overlaps section .data LMA [000000000800be54,000000000800be9b]
../../src/platform/arm-stm32f103/targets.mk:105: recipe for target 'app.elf' failed
make: *** [app.elf] Error 1

MitchBradley, 2020-02-09:
fixed by 815214ee7e65f21214d17b9f1f3f307d3ab5d33b

closed 2020-02-09 by MitchBradley

andrewtholt, 2020-02-10:
Hi,

Thanks Mitch, I'll give it a try as soon as I can.

Regards,
Andrew

andrewtholt, 2020-02-10:
it builds fine, thanks.

Regards,
Andrew