MitchBradley/cforth #58: esp8266 How do I change app

andrewtholt, 2020-05-21:
Hi,

I have feeling I'm doing something stupid, but I can't see what.

In src/app/esp8266/app.fth it includes car.fth

car.fth has a word 'car-init'

using sifting I can find this on my running board

So I comment out
```
fl car.fth
```
In app.fth, and rebuild, and flash the target:

```
make clean
make
COMPORT=/dev/ttyUSB2 make download 
```

And car-init is still there.

I also made some small, debugging, changes to wifi.fth, and app.fth which are not being changed either.

As usual any help is much appreciated.

Regards,
Andrew

quozl, 2020-05-21:
I'd check that 0x00000.bin and 0x10000.bin have changed date and time on filesystem.  I don't think they are in the _clean_ target.

andrewtholt, 2020-05-21:
Hi, 

Thanks for the quick reply.  I think I have found it.

I ran md5sum on the .bin files and app.dic, rmemoved them, made a change, make, compare the sums, and no change

I noticed in the make output
```
../tools/esptool.py elf2image .output/eagle/debug/image/eagle.app.v6.out 
```
So I found and removed it.

Now all is well.

Thanks.

Just out of curiosity what time zone are you in ?  I'm in the UK

A file is built : .output/eagle/debug/image/eagle.app.v6.out

Th

closed 2020-05-21 by andrewtholt

MitchBradley, 2020-05-21:
James (quozl) lives in Australia

quozl, 2020-05-21:
Yeah, I'm in UTC+11 this month, UNIX TZ=Australia/Sydney, but I'm about six hours drive, 450km from Sydney in a rural and regional area.  I _think_ I know where MitchBradley is.  :grin:

andrewtholt, 2020-05-22:
Hi,

I think I know too.

I met Mitch a couple of times when I worked for Sun, and had a few email
conversations over the years.

Regards,
Andrew

andrewtholt, 2020-08-18:
> The ideal workflow would be for you to push your branch to a fork in your
> github account, but in this case I am probably going to pick and choose
> bits of your code, so it would be okay for you to just mail me the patch.
> Commit your changes then do "git diff master >cforth.patch", then send me
> the cforth.patch file.

Hi,

Thanks.

Patch attached.

Regards,
Andrew