MitchBradley/cforth #59: I2C scanner

andrewtholt, 2020-05-21:
Hi,

Not so much an issue, as a question.

micropython and amforth both (and Linux) have a mechanism for 'scanning' an i2c bus and reporting on what it finds.  I was wondering if anybody had already written such a thing for cforth ?

Regards,
Andrew

MitchBradley, 2020-05-21:
I never bothered to create such a tool because it is so easy to write  #127 0 do  i (some I2c command) loop

quozl, 2020-05-21:
Same here; if the device I wanted didn't respond on the bus, I'd just
try again with a range of addresses at the ok prompt.  Almost always
it was because a datasheet specified the address without the extra
bit.

I didn't have to do this more than once, which is my criteria for
taking up dictionary space.  But if there's a situation that needs
it, I'm interested to hear about it.

andrewtholt, 2020-05-25:
Thanks for the advice.  I like the simple approach.  I guess something like:

#127 do i 
   read address 0 
  test for ACK if
   i . cr
  then
loop
;

MitchBradley, 2020-05-25:
Something like that.  It depends on how the I2C interface library is structured.   At the protocol level, all you have to do is send a start followed by the address byte, then look for ack, then issue i2c stop to get back to idle.  That level of control might or might not be possible in some libraries, but for any library there will be some simple op that will return a flag or throw an error if you get no ack. I have,  in the past, attempted to create a standardized interface, but lately I have tended to mimic whatever is available in the SDK I am using on the particular system.