MitchBradley/cforth #63: Can't use a terminal in cforth on an ESP8266F

Jos-Ven, 2020-10-14:
Hi, I am trying to install cforth on an ESP8266F using a dell PC (not a RPI)
running Debian GNU/Linux 10 (buster)
It seems all went well, but after 
sudo COMPORT=/dev/ttyUSB0 make download  
and rebooting the ESP8266F then the system led keeps flashing.
In minicom only appear the following endless repeating sequence:
Bnâ|
    âl
      Bûoï$l
I wonder how it could be solved.

wiping the ESP8266F with
python esptool.py erase_flash 
does not solve it.

Many thanks in advance,
Kind regards, Jos

Perhaps the following long logfile might help for this problem

pi@HPDEBIAN:$ cd $home
pi@HPDEBIAN:$ cd esp
pi@HPDEBIAN:$ cd cforth/build/esp8266
pi@HPDEBIAN:$ XTGCCPATH=~/esp/esp-open-sdk/xtensa-lx106-elf/bin/ make
(cd /home/pi/esp \
&& git clone https://github.com/nodemcu/nodemcu-firmware.git \
&& cd /home/pi/esp/nodemcu-firmware \
&& git branch cforth 7b83bbb \
&& git checkout cforth \
&& git apply --whitespace=fix /home/pi/esp/cforth/src/app/esp8266/*.patch \
&& tar -xzf tools/esp-open-sdk.tar.gz \
)
Cloning into 'nodemcu-firmware'...
remote: Enumerating objects: 19, done.
remote: Counting objects: 100% (19/19), done.
remote: Compressing objects: 100% (19/19), done.
remote: Total 20730 (delta 6), reused 1 (delta 0), pack-reused 20711
Receiving objects: 100% (20730/20730), 108.88 MiB | 6.71 MiB/s, done.
Resolving deltas: 100% (14146/14146), done.
Switched to branch 'cforth'
/home/pi/esp/cforth/src/app/esp8266/0001-Use-CForth-not-Lua-as-the-extension-language.patch:179: trailing whitespace.
static void
warning: 1 line adds whitespace errors.
cd /home/pi/esp/nodemcu-firmware && make --no-print-directory sdk_patched
mkdir -p "/home/pi/esp/nodemcu-firmware/cache/"
wget --tries=10 --timeout=15 --waitretry=30 --read-timeout=20 --retry-connrefused http://bbs.espressif.com/download/file.php?id=1469 -O /home/pi/esp/nodemcu-firmware/cache/esp_iot_sdk_v1.5.4_16_05_20.zip || { rm -f "/home/pi/esp/nodemcu-firmware/cache/esp_iot_sdk_v1.5.4_16_05_20.zip"; exit 1; }
--2020-10-14 15:34:49--  http://bbs.espressif.com/download/file.php?id=1469
Resolving bbs.espressif.com (bbs.espressif.com)... 119.9.92.91
Connecting to bbs.espressif.com (bbs.espressif.com)|119.9.92.91|:80... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://bbs.espressif.com/download/file.php?id=1469 [following]
--2020-10-14 15:34:50--  https://bbs.espressif.com/download/file.php?id=1469
Connecting to bbs.espressif.com (bbs.espressif.com)|119.9.92.91|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 3251308 (3.1M) [application/octet-stream]
Saving to: â/home/pi/esp/nodemcu-firmware/cache/esp_iot_sdk_v1.5.4_16_05_20.zipâ

/home/pi/esp/nodemc 100%[===================>]   3.10M   777KB/s    in 4.1s

2020-10-14 15:34:55 (777 KB/s) - â/home/pi/esp/nodemcu-firmware/cache/esp_iot_sdk_v1.5.4_16_05_20.zipâ saved [3251308/3251308]

(echo "868784bd37d47f31d52b81f133aa1fb70c58e17d  /home/pi/esp/nodemcu-firmware/cache/esp_iot_sdk_v1.5.4_16_05_20.zip" | sha1sum -c -) || { rm -f "/home/pi/esp/nodemcu-firmware/cache/esp_iot_sdk_v1.5.4_16_05_20.zip"; exit 1; }
/home/pi/esp/nodemcu-firmware/cache/esp_iot_sdk_v1.5.4_16_05_20.zip: OK
mkdir -p "/home/pi/esp/nodemcu-firmware/sdk/"
(cd "/home/pi/esp/nodemcu-firmware/sdk/" && rm -fr esp_iot_sdk_v1.5.4.1 ESP8266_NONOS_SDK && unzip /home/pi/esp/nodemcu-firmware/cache/esp_iot_sdk_v1.5.4_16_05_20.zip ESP8266_NONOS_SDK/lib/* ESP8266_NONOS_SDK/ld/eagle.rom.addr.v6.ld ESP8266_NONOS_SDK/include/* )
Archive:  /home/pi/esp/nodemcu-firmware/cache/esp_iot_sdk_v1.5.4_16_05_20.zip
   creating: ESP8266_NONOS_SDK/include/
  inflating: ESP8266_NONOS_SDK/include/airkiss.h
  inflating: ESP8266_NONOS_SDK/include/at_custom.h
  inflating: ESP8266_NONOS_SDK/include/c_types.h
[... 360 lines cut ...]
/home/pi/esp/nodemcu-firmware/tools/esptool.py --port /dev/ttyUSB0 -b 115200 write_flash -fm=dio -fs=32m 0x00000 0x00000.bin 0x10000 0x10000.bin
esptool.py v1.2-dev
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0240
Writing 28672 @ 0x0... 28672 (100 %)
Wrote 28672 bytes at 0x0 in 2.5 seconds (91.6 kbit/s)...
Writing 331776 @ 0x10000... 330752 (99 %)
A fatal error occurred: Timed out waiting for packet header
make: *** [../../src/app/esp8266/sdk.mk:46: download] Error 2
pi@HPDEBIAN:$
pi@HPDEBIAN:$ # The next attempt was OK
pi@HPDEBIAN:$
pi@HPDEBIAN:$ sudo COMPORT=/dev/ttyUSB0 make download
/home/pi/esp/nodemcu-firmware/tools/esptool.py --port /dev/ttyUSB0 -b 115200 write_flash -fm=dio -fs=32m 0x00000 0x00000.bin 0x10000 0x10000.bin
esptool.py v1.2-dev
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0240
Writing 28672 @ 0x0... 28672 (100 %)
Wrote 28672 bytes at 0x0 in 2.5 seconds (91.6 kbit/s)...
Writing 331776 @ 0x10000... 331776 (100 %)
Wrote 331776 bytes at 0x10000 in 28.9 seconds (91.9 kbit/s)...
Leaving...
pi@HPDEBIAN:$

MitchBradley, 2020-10-14:
I'm not exactly sure what you mean by the designation ESP8266F.  Is it perhaps the ESP-12F module?

You could try flashing with some other apps to see what happens, or try different settings for -fm and -fs.  Maybe your module has a different flash size.  -fs 32m is for a 4 MByte FLASH.  Some modules, if I recall correctly, have only 1 MByte of FLASH, which would be -fs 8m  (8 Megabits).  There might even be modules with only 512KBytes of FLASH, which would be -fs 4m .  It has been awhile since I used any of the really small modules.

Jos-Ven, 2020-10-15:
Hi,  Thanks for the quick response.
It is indeed an ESP-12F module and has 4MB Flash. The use of -fs 32m is possible.
I have no Arduino, but I am able to flash punyforth.
Somewhere in their phyton script they use:
("python esptool.py -p %s write_flash -fm %s -ff 40m 0x%x %s" % (self.port, self.flashmode, address, path))
("python esptool.py -p %s write_flash -fs 32m -fm %s -ff 40m %s" % (self.port, self.flashmode, ' '.join("0x%x %s" % each for each in tupl)))

MitchBradley, 2020-10-16:
You could try adding the -ff 40m argument to the command line.  https://github.com/espressif/esptool#flash-frequency---flash_freq--ff

MitchBradley, 2020-10-16:
Look near the end of cforth/src/app/esp8266/sdk.mk to see the esptool command line.  You could add the "-ff 40m" argument to the LOADCMD line, and also try changing FM?=dio to FM?=dout .

MitchBradley, 2020-10-16:
The punyforth README says they use qio mode by default.  The dio mode that cForth defaults to is generally safer than qio, but there are no guarantees, so you should try all the modes dout , dio , and qio

Jos-Ven, 2020-10-16:
Hi,
I tried all the 3 modes with and without the "-ff 40m" argument.
Unfortunately, it did not solve the problem.

MitchBradley, 2020-10-17:
What is the result of
`/home/pi/esp/nodemcu-firmware/tools/esptool.py -p /dev/ttyUSB0 flash_id`

MitchBradley, 2020-10-17:
Another thing to try.

Edit /home/pi/esp/nodemcu-firmware/app/user/user_main.c   Down near the end there are these lines
```
    #ifndef NODE_DEBUG
    system_set_os_print(0);
    #endif
```
Comment out the #ifndef and #endif like this
```
    //#ifndef NODE_DEBUG
    system_set_os_print(0);
    //#endif
```

The rebuild with
```
rm *.bin
COMPORT=/dev/ttyUSB0 make download
```
The purpose of that is to enable some debugging messages from the system startup code.

Then connect the serial port and reboot the ESP8266 with the reset button.  For me, that gives:
```
flash_used_end:402617e8
fs.start:70000,max:38a000
_flash_used_end:402617e8
fs.start:80000,max:37a000
_flash_used_end:402617e8
fs.start:90000,max:36a000
mount res: 0, 0

CForth built 2020-10-06 20:08 from cda7400
Type a key within 2 seconds to interact
```

Jos-Ven, 2020-10-17:
For the flash_id:
root@HPDEBIAN:$/home/pi/esp/nodemcu-firmware/tools/esptool.py -p /dev/ttyUSB0 flash_id
esptool.py v1.2-dev
Connecting...
Manufacturer: 1c
Device: 3016
root@HPDEBIAN:$
---------------

After editing user_main.c, removing rm *.bin I got the following:

pi@HPDEBIAN:$ COMPORT=/dev/ttyUSB0 make download
/home/pi/esp/nodemcu-firmware/tools/esptool.py --port /dev/ttyUSB0 -b 115200 write_flash -fm=qio -ff 40m -fs=32m 0x00000 0x00000.bin 0x10000 0x10000.bin
esptool.py v1.2-dev
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0040
Writing 28672 @ 0x0... 28672 (100 %)
Wrote 28672 bytes at 0x0 in 2.5 seconds (91.7 kbit/s)...
Writing 331776 @ 0x10000... 331776 (100 %)
Wrote 331776 bytes at 0x10000 in 28.9 seconds (92.0 kbit/s)...

As soon as I start minicom and reset the ESP8266 then I got again
the repeating unreadable output in minicom and the flashing systemled.

quozl, 2020-10-17:
Once I had something similar, and it was solved by reflashing nodemcu, then flashing cforth; so I think it was boot configuration in flash disagreeing with some reality.

Jos-Ven, 2020-10-17:
What was the exact syntax to reflash the nodemcu?

quozl, 2020-10-17:
Don't recall, sorry.  See https://nodemcu.readthedocs.io/en/release/flash/

Jos-Ven, 2020-10-17:
Hi, Thanks for the hint. After downloading and flashing I got the AT commands back.
ready
AT+GMR
00200.9.4

OK
So that went well. 
But then after COMPORT=/dev/ttyUSB0 make download
I got the same problem again. Sorry.

MitchBradley, 2020-10-17:
What external circuitry is connected to the module?

MitchBradley, 2020-10-17:
If you start a terminal emulator at 74880 baud, you can see the early startup messages from ESP8266.  I use TeraTerm Pro on Windows which lets me set any baud rate that I want, not just the standard ones.  This reference http://cholla.mmto.org/esp8266/weird_baud/ tells one way to do it on Linux.  The early messages that I see are:

```
load 0x40100000, len 24680, room 16
tail 8
chksum 0xfb
load 0x3ffe8000, len 960, room 0
tail 0
chksum 0x37
load 0x3ffe83c0, len 8, room 8
tail 0
chksum 0x12
csum 0x12
rf cal sector: 1019
```
Then it switches to 115200 and the CForth startup messages are gibberish.

Jos-Ven, 2020-10-17:
Hi, 
See http://home.kpn.nl/~josv/CircuitEsp8266Minimal.jpg for the used circuit.

The startup msg. at 74880 is:
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 24952, room 16
tail 8
chksum 0x8c
load 0x3ffe8000, len 944, room 0
tail 0
chksum 0x36
load 0x3ffe83b0, len 8, room 8
tail 0
chksum 0xc2
csum 0xc2
rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

Jos-Ven, 2020-10-17:
Hi, Thanks to your support it seems the problem is solved after searching for :
rst cause:2, boot mode:(3,7)
An extra capacitor of 470 uF between the 5V and the GND did the trick!
Many thanks, kind regards, Jos

closed 2020-10-17 by Jos-Ven