MitchBradley/cforth #66: Constant rebooting

arblake, 2020-11-20:
I flashed a device and it constantly reboots, thus:
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4828
load:0x40078000,len:8784
load:0x40080000,len:5780
entry 0x400802dc
I (293) cpu_start: Pro cpu up.
I (293) cpu_start: Single core mode
I (294) heap_init: Initializing. RAM available for dynamic allocation:
I (297) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (303) heap_init: At 3FFCE810 len 000117F0 (69 KiB): DRAM
I (309) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (316) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (322) heap_init: At 40091FCC len 0000E034 (56 KiB): IRAM
I (328) cpu_start: Pro cpu start user code
I (123) cpu_start: Starting scheduler on PRO CPU.

CForth built Undefined word encountered
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4008a172  PS      : 0x00050033  A0      : 0x4008a07b  A1      : 0x3ffd2400  
A2      : 0xa5a5a5a5  A3      : 0x00000000  A4      : 0x000002de  A5      : 0x3ffc05d0  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x400820a8  A9      : 0x3ffc7a7c  
A10     : 0x3ffd24f3  A11     : 0x00000001  A12     : 0x80088646  A13     : 0x3ffc05c0  
A14     : 0x3ffc0604  A15     : 0x00000001  SAR     : 0x00000019  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xa5a5a5a5  LBEG    : 0x400d58ac  LEND    : 0x400d58c5  LCOUNT  : 0x00000000  

Backtrace: 0x4008a172:0x3ffd2400 0x4008a078:0x3ffd2410

Rebooting...

MitchBradley, 2020-11-20:
The line "CForth build Undefined word encountered" means that something went wrong in the compilation phase.  Without knowing a lot more about what you tried, I can't debug further.  I need to know which version of the cforth tree you are using, which build directory you are working in, what system you are compiling on, and any modifications that you have made.

arblake, 2020-11-21:
I tried again, now this :
E (64) boot: Chip CPU frequency rated for 160MHz. Modify CPU frequency in menuconfig
user code done
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4828
load:0x40078000,len:8784
load:0x40080000,len:5780
entry 0x400802dc
E (64) boot: Chip CPU frequency rated for 160MHz. Modify CPU frequency in menuconfig
user code done

it is: cforth WIP: built in build/ESP32:  compiled on a linux box: no changes: device is ESP32-WROOM-32  
I did successfully install on a "HiGrow" device which has the same chip.
The error implies a CPU frequency error and correcting with menuconfig ... but where is menuconfig?

MitchBradley, 2020-11-21:
menuconfig is a tool that the ESP IDF development environment uses to modify the sdkconfig file, which sets up a lot (more than 700) configuration parameters that control how the application is built.  The easiest way to solve the 160 MHz problem is to edit build/esp32/sdk_build/sdkconfig directly.  Change these lines:

```
CONFIG_ESP32_DEFAULT_CPU_FREQ_160=
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=240

```
to

```
CONFIG_ESP32_DEFAULT_CPU_FREQ_160=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=
CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=160

```

MitchBradley, 2020-11-21:
I added a target to the makefile so you can say

`make menuconfig`

in cforth/build/esp32 .  It might work for you, but it is a complicated process to compile it, so if something goes wrong, I really don't have time to support it.  I can support direct edits of sdkconfig.

arblake, 2020-11-21:
I simply edited the sdkconfig file. All is working. 
Next challenge is to discover how to drive io pins , ADC etc and wifi
Thanks

closed 2020-11-21 by arblake