MitchBradley/cforth #67: resize-file

Jos-Ven, 2020-11-30:
In an ANS forth system RESIZE-FILE can be used to create quickly a large file.
The following code also works to create a large file:

decimal
768                constant /UdpData
0 value fd-file    variable fsize
create &UdpData    /UdpData allot

: init-file-space ( size -- )
  &UdpData /UdpData erase
   /UdpData /mod  /UdpData * &UdpData + &UdpData
     do    &UdpData /UdpData fd-file  write-file 1 ms
               if   fsize off leave
               then
           /UdpData
     +loop
   &UdpData swap fd-file write-file 1 ms
      if   fsize off
      then
      fd-file file-size 3drop  ;

  s" test.txt" 2dup type  r/w create-file drop to fd-file
  154659 init-file-space  fd-file close-file ls

\ RESIZE-FILE  ( ud fileid -- ior ) would still be much nicer.

MitchBradley, 2020-11-30:
You can extend a file efficiently with reposition-file.  Simply reposition to the new end of file, minus one, then write a single character.  RESIZE-FILE could be implemented thusly for the enlarge case, but I do not have time right now to do it.

Jos-Ven, 2020-11-30:
Ok, thanks for the hint.

closed 2020-11-30 by Jos-Ven