MitchBradley/cforth #68: Workflow Git

Jos-Ven, 2020-12-04:
Hi Mitch,
I Updated interface.h and extend.c and tried to update them by using:
$ git add src/app/esp8266-rtos/extend.c src/app/esp8266-rtos/interface.h 
$ git commit -m "added: esp_clk_cpu_freq esp_set_cpu_freq esp_deep_sleep"
$ git push 

The files were updated https://github.com/Jos-Ven/cforth/tree/WIP/src/app/esp8266-rtos
On the same page there is a message that tells:
This branch is 1 commit ahead of MitchBradley:WIP

Is this workflow is right and convenient for you to handle the proposed update?

MitchBradley, 2020-12-04:
So far so good. The next step is to create a pull request.  Start by clicking on the "Pull request" link to the right of "This branch is 1 commit ahead".  Describe the benefits / implications of your changes in the "Leave a comment" area, then click on the green "Create pull request" button.

Jos-Ven, 2020-12-05:
It was no use on my local PC. (msg: Already up to date.)
Then I found and activated a pull at: https://github.com/Jos-Ven/cforth/tree/WIP
I hope it will be alright now.

closed 2021-01-19 by Jos-Ven