MitchBradley/cforth #70: Adding stuff etc to esp8266-rtos

andrewtholt, 2020-12-17:
Hi,

I have found time to look at cforth again and the new goodness that you have added.

Given that it's now on top of Esprissif's extended FreeRTOS I was looking at adding words for things like MQTT and looking in to lower power modes.

Not wanting to create a parallel implementation of these things  I was wondering what your plans were for features.

Regards,
Andrew

MitchBradley, 2020-12-17:
I'm not working on cforth at this instant so anything you do in the near future is not likely to be redundant.

The MQTT implementation in src/app/esp32/mqtt.fth is likely to work in esp8266-rtos, perhaps unmodified.

Let me know what you discover, and provide PRs for any exciting new features that you develop.

andrewtholt, 2020-12-18:
Hi,

I'm hoping to make a start sometime over the christmas break.  I'll kep you
posted .

Regards,
ANdrew

Jos-Ven, 2021-01-30:
Hi, In the past I reported that I could not connect to my router when it was set to
the '11N protocol only' in my router.
After adding 'esp_wifi_set_protocol' in 
~/cforth/src/app/esp8266-rtos/sdk_src/main/interface.c it works.
Here is what I did, perhaps it could be better:

    if (strlen(password)) {
        wifi_config.sta.threshold.authmode = WIFI_AUTH_WEP;
// When I set in my router to 11N ONLY then:
    esp_wifi_set_protocol(ESP_IF_WIFI_STA, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N); // Works
//  esp_wifi_set_protocol(ESP_IF_WIFI_STA, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G);   // Fails
    }
Regards, Jos

MitchBradley, 2021-01-30:
Please read this https://www.section.io/engineering-education/wep-encryption/

Jos-Ven, 2021-01-30:
Hi, Thanks for the hint, I know about the security problem of web Encryption.
It is off in my router.
Despite that fact I am able to connect to my router using WPA2-PSK.
It seems the line WIFI_AUTH_WEP is overruled.
Also note that it I did not change that line in interface.c.

Kind regards, Jos