MitchBradley/cforth #73: multitasking/cororoutines

andrewtholt, 2021-01-26:
Hi Mitch,

I seem to recall asking you this question a very long time ago, when I was working at Sun, certainly last century.

Did you ever implement a co-operative task system for cforth ? 

I see that pause is a deferred word, IIRC it was in openfirmware too.

Thanks

MitchBradley, 2021-01-26:
Yes, I have used cooperative MT with an active pause.  I don't use it often, but all of my Forth systems are set up so pause can do the right thing.

andrewtholt, 2021-01-27:
Hi Mitch,

Is there any way I can make use of this ?

MitchBradley, 2021-02-07:
I updated the multitasking code for CForth.  Pull the latest commit and look at src/lib/test-tasking.fth

andrewtholt, 2021-02-08:
Thanks for that.  I'll take a look ASAP.

Regards,
Andrew

Jos-Ven, 2021-02-08:
That sounds great. 
Unfortunately, I am not able to compile tasking.fth under esp8266-rtos
the words   asleep  (pause  and link   are missing.
It would be nice if it is also possible to use tasking.fth with esp8266-rtos!

Kind regards, Jos

andrewtholt, 2021-02-08:
A quick scan would suggest that you replace (pause with (pause)

Haven't actually tried it yet, that's from observation.

Jos-Ven, 2021-02-08:
Hi Andrew,

Thanks for the quick scan.
Can't replace (pause with (pause) since
(pause) is not defined under esp8266-rtos.

Kind regards, Jos

MitchBradley, 2021-02-08:
asleep, (pause, and link are defined in forth.c according to the part of the commit shown below.  You will need to recompile to get them.

[image]

MitchBradley, 2021-02-08:
(pause is the correct name.

Jos-Ven, 2021-02-09:
i Mitch,
Sorry, that did not work. Here is what I did:
If I clone the main branch then after:
cd /home/pi/cf/cforth/build/esp8266-rtos
bash: cd: /home/pi/cf/cforth/build/esp8266-rtos: No such file or directory

The esp8266-rtos directory seems only to be present in the branch WIP and not in the branch MAIN
After: 
git clone https://github.com/MitchBradley/cforth --branch WIP
cd /home/pi/cf/cforth/build/esp8266-rtos
make
make flash

\ Then in cforth I saw:
sifting pause
        In vocabulary  forth
(40285388) pause
ok see pause
defer pause is
: noop

;
Kind regards, Jos

Jos-Ven, 2021-02-09:
Hi Mitch,

After I made the changes manually in forth.c for the words 
asleep (pause and link
Then it was possible to compile and run test-tasking.fth 
The counter was indeed increased when the background was running.
To be sure multiple tasks could also be handled I added 1 more task.
Here is the code:

\needs multi fl tasking.fth

\ Tests and examples for cooperative multitasking

\ Global variable used by the test tasks
Global counts0
Global counts1
Global counts2

\ Explicit creation of word and task

: do-count0 begin 1 counts0 +!  pause  again  ;
task: count-task0
' do-count0 count-task0 fork
count-task0 wake

: do-count1 begin 2 counts1 +!  pause  again  ;
task: count-task1
' do-count1 count-task1 fork
count-task1 wake

\ Combined creation of task with word to execute
\ background counters3  begin  3 counts0 +!  2 counts1 +!  1 counts2 +!  pause  again  ;

background counters2  begin  pause  again  ;

: .counts  ( -- )  ." counts = " counts0 @ .d counts1 @ .d  counts2 @ .d cr  ;

.counts
pause
.counts
pause
.counts

: run-background  ( -- )  begin  pause  key? until  key drop  ;

.( Type a key to return to prompt) cr
run-background
.counts
\  end code

Output when counters3 is active:
counts = 0 0 0
counts = 3 4 1
counts = 6 8 2
Type a key to return to prompt
counts = 1332300 1776400 444100

Output when counters2 is active:
ok fl test-tasking.fth
counts = 0 0 0
counts = 0 2 0
counts = 0 4 0
Type a key to return to prompt
counts = 0 656194 0

\ Manual changing:
ok multi
ok pause .counts
counts = 0 656200 0
ok pause .counts
counts = 0 656202 0
ok pause .counts
counts = 0 656204 0

It seems pause does not run all tasks
Here is the link to the changed forth.c:
https://github.com/Jos-Ven/cforth/commit/f13c871bc969efaba1fd847017e2e37ca40329a1
I hope the problem can be solved.

Thanks in advance, kind regards, Jos