MitchBradley/cforth #79: esp8266 not working after flash

andrewtholt, 2021-09-13:
Hi everybody,

Once again I have returned to take a look at cforth.

Built latest esp32, works like a charm.  MQTT publishing, very cool.

Tried building esp8266-rtos.  Good start it built.  Tried flashing onto a device, and

```
COMPORT=/dev/ttyUSB0 make download
```
It flashed without complaint:

```
....
Wrote 647088 bytes (398451 compressed) at 0x00010000 in 6.0 seconds (effective 866.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 97...
Wrote 3072 bytes (97 compressed) at 0x00008000 in 0.0 seconds (effective 9122.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Connected to a terminal:
```
���������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������
```

and so on.

I have a tried a couple of other 8266 boards and they behave the same.

I have a feeling of deja vu, and this is something simple. but ....

Thanks,
Andrew

MitchBradley, 2021-09-13:
Try changing the flash mode to dio instead of qio.  Or maybe the other way around.  It is in the esptool command line.

andrewtholt, 2021-09-13:
Thanks, I'll give it a try.

⁣Get BlueMail for Android ​

quozl, 2021-09-13:
When I last had that problem it was necessary to use esptool to erase
flash.

andrewtholt, 2021-09-14:
I erased the flash, and connected to the device.  As you would expect nothing printed.

I changed the flash mode (default is dio), and reduced the baud rate (IIRC this has caused problems) my command is:

```
FM=qio COMPORT=/dev/ttyUSB0 ESPBAUD=115200 make flash
```

Same result.

quozl, 2021-09-15:
Just guessing, but what changed since it last worked?  I've had
problems which seem to be related to GCC upgrade, but I've not
finished analysis yet.

MitchBradley, 2021-09-15:
If you set the terminal emulator baud rate to 74880 you can see messages from the bootloader; sometimes that can give a clue about what is going wrong.  Yes I realize that is a strange baud rate that not all terminal emulators can handle.  Tera Term Pro on Windows can do it.  There is no such entry in the drop down menu but you can type it in manually.

andrewtholt, 2021-09-15:
Hi,

It's a while since I tried this.  SO many things have changed 🙄

andrewtholt, 2021-09-15:
I set the baud rate as suggested, and hit reset.  This is what I saw:

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x60000014, len 8, room 16
tail 8

Does that tell you anything ?

andrewtholt, 2021-09-15:
Attempting to show if the h/w was working I flahed it with a binary of micropython I've used before.  This was successful

I reflashed with cforth, same problem.

andrewtholt, 2021-09-15:
What's puzzling to me is:

```
esptool.py v2.4.0
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: bc:dd:c2:23:57:37
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 10464 bytes to 7045...
Wrote 10464 bytes (7045 compressed) at 0x00000000 in 0.6 seconds (effective 134.8 kbit/s)...
Hash of data verified.
Compressed 647088 bytes to 398451...
Wrote 647088 bytes (398451 compressed) at 0x00010000 in 35.1 seconds (effective 147.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 97...
Wrote 3072 bytes (97 compressed) at 0x00008000 in 0.0 seconds (effective 1926.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
So it writes it, verifies it and all looks good

andrewtholt, 2021-09-15:
I have just built the esp8266 (not esp8266-rtos) binary.  Flashed that.

When I hit reset a got a load of garbage and a message saying 'formatting filesystem, please wait"  I did and the OK prompt appeared.

I tried the same thing, again with esp8266-rtos, and same issue.

Jos-Ven, 2021-09-15:
There are several boards for an esp8266 
Which do you have?

andrewtholt, 2021-09-15:
By trying several times I have esp8266 working on some boards.  These two work:

wio link

The other looks like an Arduino with D1 printed on the component side, and Arduino compatible, Based ESP8266  I believe it came from AZ Delivery in Germany

andrewtholt, 2021-09-15:
It's strange that I can get the esp8266  working on some boards, but the esp8266-rtos on none

Jos-Ven, 2021-09-15:
According:
https://github.com/SeeedDocument/Wio_Link/raw/master/resource/Wio%20Link%20v1.0%20sch.pdf
That board contains a ESP8266EX

The one with D1 sounds a bit like a:
WeMos D1 WiFi UNO ESP8266 IOT IDE Compatible Board
https://www.instructables.com/Arduino-WeMos-D1-WiFi-UNO-ESP-8266-IoT-IDE-Compati/

On my ESP12-F systems trouble can happen when other software was flashed 
on it.

In a worst-case scenario.
I have to start putty and connect it to the COM port with speed 74880 bauds.
Reboot the ESP12-F
And look for the boot message and search the internet for a solution.

That takes a lot of time.

andrewtholt, 2021-09-15:
This is getting more wierd.

I dleted cforth, and cloned it from git.  Abuild of esp8266 now fails with:

```
TCC ../../src/app/esp8266/extend.c
../../src/app/esp8266/extend.c:401:21: fatal error: esp_spi.h: No such file or directory
 #include "esp_spi.h"
```

MitchBradley, 2021-09-15:
ESP8266 booting is sort of involved - there is a ROM bootloader that reads several other FLASH artifacts including a partition map and often a secondary bootloader.  If those things don't match just right, booting can fail.  It's not enough for the esptool flash step to succeed - the various things that it flashes have to be consistent with each other.  Sometimes you think things are just fine, but the only reason that it worked because some other application already put something that you need into FLASH.

If I had time to dig down into this, I'm sure I could sort out the problem, but unfortunately I am way behind on some other stuff at the moment.

MitchBradley, 2021-09-16:
I had forgotten to check in esp_spi.h .  I just fixed that.  It is a simple file with some function prototypes.

andrewtholt, 2021-09-18:
Thanks Mitch.  Now builds OK.

andrewtholt, 2021-09-19:
Just having another attempt to flash esp8266-rtos on a device.  Spotetd the following message in the output from esptool.py

```
WARNING: Suspicious segment 0x60000014, length 8
```
Does this give any hint to anybody ?

Jos-Ven, 2021-09-19:
Perhaps the following might help:
It is what I do if a ESP12-F does not work as expected (could be 
dangerous in your case!).
After erasing the flash memory I had also to send a file with 4MB of 
0xFF to the ESP12-F
Next flash esp_init_data_default.bin to the happened ESP12-F. This is 
explained at:
  https://arduino.stackexchange.com/questions/33590/endless-loop-on-boot-after-reflashing-esp-12e-with-at-firmware
and http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory
In my case I flash:
~/esptool.py --port /dev/ttyUSB0 --baud 115200 write flash 0x3fc000 
esp_init_data_default.bin
Sometimes that did not do the trick
Than I flash the AT command set to the ESP12-F. See:
https://robertoostenveld.nl/esp8266-at-firmware/

andrewtholt, 2021-09-28:
Some progress.

I can flash the esp8266 and it works.

The esp8266-rtos version still fails to boot.

andrewtholt, 2021-12-13:
More progress.  It was my environent.  Don't lnow what but after a tidy up it all seems to work now.  Sorry about that

closed 2021-12-13 by andrewtholt