MitchBradley/cforth #82: setsockopt problem

Jos-Ven, 2021-12-07:
In cforth/src/app/esp32/server.fth I found:
      \ Set SO_LINGER so lwip-close does not discard any pending data
      \ 8 &linger $80 $fff r@  setsockopt drop

When I tried that option in a new word setsockopt always returns -1
So, I do not think the option SO_LINGER is active.

The used code is:

\ create &linger 1 , 5 ,  \ on , 5 seconds, see server.fth
create &linger2 1 , 5 ,
: set-socketopt ( SocketAfterAccept - )
   >r 8 &linger  $80 $fff r@ setsockopt cr .s drop \ setsockopt always returns  -1
   \  8 &linger2 $80 $fff r@ getsockopt cr .s drop \ getsockopt crashes cforth
   r> drop ;

: http-responder  ( timeout -- )
   timed-accept if  exit  then  
   dup set-socketopt
   req-buf /req-buf  rot lwip-read    ( len )
   req-buf swap  handle-request
   1 ms   ;

How can the problem at setsockopt be solved ?
Thanks in advance, Jos

MitchBradley, 2021-12-07:
[This thread ](https://www.esp32.com/viewtopic.php?t=6688) says:

> The reason for the ENOPROTOOPT result is that in ESP-IDF we don't build LWIP with LWIP_SO_LINGER enabled, so the feature isn't available.

I think that you would need to configure esp-idf to turn it on.  I'm not sure about this, but I think the right way would be to go into <esp-idf>/components/lwip/include/lwip/port/lwipopts.h and, in the  Socket options section, add this line:

```
#define LWIP_SO_LINGER 1
```

Jos-Ven, 2021-12-07:
Thank you, now it works!
Kind regards, Jos

closed 2021-12-07 by Jos-Ven