MitchBradley/cforth #86: Flashing a Sonoff Basic R1

andrewtholt, 2022-03-29:
Hi,

I have a 'spare' sonoff R1 and I am trying to flash cforth onto it.

I have read the notes 
```
Sonoff S20 - WiFi mains switch with ESP8266 inside    
    
FM=dout FS=8m make download    
    
FM=dio will appear to download but the firmware won't run     
```
I am using the following to flash it:
```
esptool.py --port /dev/ttyUSB0 -b 115200 write_flash -fs=8m -fm=dout 0x00000 0x00000.bin 0x10000 0x10000.bin
```
I have flashed other things (Tasmota and cforth onto esp8266 boards) recently bit for information the esptool version is

```
$ esptool.py version

esptool.py v3.2-dev
3.2-dev
```
Any advice/hints/tips etc is appreciated.

Thanks

Andrew

andrewtholt, 2022-03-29:
Spoke to soon, works but on startup it sometimes does not.  Also, when it does start, can't switch relay

andrewtholt, 2022-03-30:
OK, some progress.  I removed most of the code from app.fth leaving just the bare minimum.  I have a utility word

```
#32 constant /msg-buffer
#255 constant /topic-buffer

/msg-buffer   buffer: msg-buffer
/topic-buffer buffer: topic-buffer

: publish ( topic n msg n -- )    
    msg-buffer place    
    topic-buffer place    
    msg-buffer count topic-buffer count 0 0 mqtt-publish-qos0    
;    

```
used like this
```
 s" /home/office/sonoff/POWER" s" OFF" publish
```

And sure enough a subscriber gets the message.  Followed shortly by
```
wdt reset
load 0x40100000, len 24960, room 16 
tail 0
chksum 0x08
load 0x3ffe8000, len 960, room 8 
tail 8
chksum 0xc8
load 0x3ffe83c0, len 8, room 0 
tail 8
chksum 0xdd
csum 0xdd
```

Any suggestions ?
BTW is anybody reading this ?

MitchBradley, 2022-03-30:
Yes I am reading it but I don't have much time to do anything about it.

MitchBradley, 2022-03-30:
If it is a watchdog reset then something is probably blocking so the core wifi code cannot run periodically.

Normally, when you run mqtt-loop, it calls do-tcp-poll which includes "ms" inside the loop, which returns to the ESP8266 core code to give it time to do its wifi stuff.

But since you did not tell us what your main loop looks like, I can't tell what is going on.  Are you just sitting at the ok prompt?  Or what?

You have to give me a lot of information for me to be able to help you without wasting huge amounts of my time.

andrewtholt, 2022-03-31:
I understand that you have a day job Mitch, I was wondering if others (who post here) had seen this.

I interrupt the auto start.

I then enter

```
' mqtt-start catch
```
After a short delay it returns

I then run the publish command, my subscriber sees the message.

Then the reset.

I have only seen the watchdog a few times.  I also see a reset.

I'm beginning to suspect power issues since I am  running the sonoff from external power, i.e. not mains.

If I get time I'm going to try a few things today.

As always, thanks for your assistance.

Jos-Ven, 2022-04-07:
In the past I had some power issues when the ESP8266 or ESP32 were connecting to my router.
They were solved with a using 2 capacitors.
One of 330 uF a 100 nF parallel over the first one for small peaks.

MitchBradley, 2022-04-07:
I too have had problems that were solved with a capacitor.  In my case it was an ESP32 on a D1 Mini form factor module.  The module had too little capacitance on 3V3, so when the radio turned on the 3v3 rail dipped and caused the module to reset.  A 220uF cap from 3V3 to GND fixed it.  The quality "Devkit C" modules from Espressif seem to have about 50uF on 3V3, so that might be enough, assuming a good quality cap that has its rated capacitance.