MitchBradley/cforth #87: MOre of a question: FreeBSD

andrewtholt, 2022-05-18:
Hi,

I'm taking a look at FReeBSD at the moment.  I was wondering if tou have ever build cforth for a BSD os (not including SunOS 4.x of course)

Regards,
Andrew

andrewtholt, 2022-05-19:
FYI Had a few minutes so I gave it a try.  Figured the macos version was closest, so in directory build/host-serial-macos64

defined __APPLE__ in the makefile and ran gmake

```
CC ../../src/cforth/lineedit.c
CC ../../src/cforth/getc-kbd.c
CC ../../src/lib/sha256.c
MAKING FORTH
CC main.o io.o nullbi.o dictfile.o mallocl.o lineedit.o getc-kbd.o sha256.o forth.o compiler.o syscall.o floatops.o extend.o -o forth
ld: error: undefined symbol: fmod
>>> referenced by floatops.c:68 (../../src/cforth/floatops.c:68)
>>>               floatops.o:(floatop)

ld: error: undefined symbol: sin
>>> referenced by floatops.c:72 (../../src/cforth/floatops.c:72)
>>>               floatops.o:(floatop)
```
So I deleted -DFLOATING from the make file and tried again.

Built fine, runs.  Just need to sort out floatops.  Though, I confess, for most of my uses I don't need them.

MitchBradley, 2022-05-19:
I found [this patch to FreeBSD]( https://lists.freebsd.org/pipermail/freebsd-standards/2003-April/000053.html).  If you can't figure out how to get ecvt from the library, maybe you could add the source from that patch directly to the CForth tree.