MitchBradley/cforth #93: Catching undefined words during a build

quozl, 2022-11-12:
When one makes a mistake, such as referencing an undefined word in _app.fth_, the build continues and returns zero exit status.

Would would be a nice way to catch this?

I've seen `$do-undefined` and `.not-found` in _interp.fth_ and `(.not-found)` in _util.fth_.

I could create a flag for return status, define a `where2` which calls `where1`, and then call `finished-pop` as the last line, but my sense of waste is triggered. :grin:

MitchBradley, 2022-11-12:
.not-found is probably the thing to amend.  The trick is to distinguish between interaction and compiling from the command line, doing the exit only in the latter case.  Let me think about that.

MitchBradley, 2022-11-12:
interactive? tells you whether or not you are interactive, so if it returns false, you could do the exit with status code thing.

MitchBradley, 2022-11-12:
```
: (.not-found)  ( -- )
   cr  type ."  ?"  cr  where1
   interactive?  0=  if  1 finished-pop  then
;
```
seems to work

closed 2022-11-13 by quozl

MitchBradley, 2022-11-13:
PR that patch and I will merge it.