MitchBradley/cforth #95: Does CForth have any words to compile to RAM, Flash or EEPROM?

snowyu, 2022-12-13:
The [FlashForth](https://flashforth.com/) has `ram`/`eeprom`/`flash` words to switch.
And [mecrisp-stellaris](https://mecrisp.sourceforge.net/) has `compiletoflash`/`compiletoram` words to switch.

quozl, 2022-12-13:
Not that I am aware of.  At the moment, new words are added to a RAM dictionary.

MitchBradley, 2022-12-13:
CForth's memory model is that of C.  The part of the dictionary that is precompiled goes into C read-only data, which would typically be FLASH on an embedded computer.  When you add definitions at runtime, they go into read-write data, which is RAM.  C has no memory space that corresponds to EEPROM - and anyway EEPROMs are typically too small to be used for definitions.

snowyu, 2022-12-14:
Thanks your explain. It would be great if we can abstract the structure that runs directly on the Flash IO layer (RAM as a buffer). It is useful when the Flash size of the device is much larger than the RAM.

There are a number of libraries on platformio that provide direct access to Flash:

https://registry.platformio.org/libraries/khoih-prog/FlashStorage_SAMD
https://registry.platformio.org/libraries/khoih-prog/FlashStorage_STM32
https://registry.platformio.org/libraries/khoih-prog/FlashStorage_RTL8720

MitchBradley, 2022-12-14:
Great as in "I will pay money for someone to design and implement this feature" or great as in "if somebody spends their time, without compensation, to implement it, I might use it"?

snowyu, 2022-12-15:
The definition of the word "great" depends on one's understanding. For former US President Donald J. Trump, building a high wall similar to a prison is "making America great again". For me, "great" means To make things more useful to more people. Use your actions and intentions to help more people, and be full of gratitude. As Socrates said: "True greatness lies in compassion and kindness".

Anyway, Great thanks to Open Source.