MitchBradley/cforth #96: ESP32 and platformio

andrewtholt, 2022-12-16:
Hi,

Simple question, the README.md file lists (in the region of line 57) four targets.  The last of which is ESP32.

So using the example 

```pio run --environment pico```

which build fine BTW, and using

```pio run --environment esp32```

I get 

Error: Unknown environment names 'esp32'. Valid names are 'teensy35, target_makebi, host_meta, teensy36, featheresp32, teensy31, teensy40, host_forth, adafruit_feather_m0, pico, teensy32, host_makeccalls, host_makename, esp01'

I'm guessing this is a work in progress ?

MitchBradley, 2022-12-16:
Which of those "valid names" would you think is most likely to be an esp32?  There are about a million different ESP32 boards, with more being developed every day, and there is no way that we can make targets for every one of them.  You will have to edit platformio.ini to configure for the specific board you care about.

closed 2022-12-16 by MitchBradley

andrewtholt, 2022-12-19:
Hi Mitch,

d'oh featheresp32, hit the button too quick.  I'll investigate further.

Thanks for all your help over the last year.

Have a merry Christmas, and a happy new year Mitch.
Regards,
Andrew