MitchBradley/cforth #98: Embedded component build for ESP32

jonsmirl, 2023-03-11:
Is there any interest in making a component build for the ESP32 family which could be embedded into other ESP32 apps?  Maybe in a form where it could be hosted in the Espressif component repository.  https://components.espressif.com/components  This would also make cforth more visible to ESP developers.

My idea is that a cforth make command would copy all of the correct source files into an ESP component directory but not compile them.  And then you would use the esp-idf would do all of the compiling.  There are many flavors of esp32 chips now and the cforth build only supports the original. This copied source would be packaged into the component distribution. ESP components are distributed in source form and they compile when the app does, that allows the debugger to work both on the component and the main app.

For esp32 command line support, there would be a tiny app in the cforth build which just sets up the serial port and then calls the embedded component. Similar to the existing esp32 build.  There is already a component in that build, spiffs.

The goal here is to be able to target devices without a serial port but which have a touch screen. Such as these: https://www.lilygo.cc/collections/all

This also means cforth will be ported onto IDF 4.4 and the new build system. I've compiled with 4.4 and the needed touch ups are minor.

quozl, 2023-03-11:
I predict there would be people who would use it.  Another ecosystem to consider is Zephyr.

Jos-Ven, 2023-03-11:
I would love to see a faster and clearer update policy instead of 
multiple copies on various sites.
However I guess that Mitch is too busy for that.

Jos

MitchBradley, 2023-03-11:
@Jos-Ven  I am spending more than my available spare time on the FluidNC project.  I would be happy to add you as a collaborator so you and @quozl could push updates without my intervention.  I would hope that you would review each other's work when I am being unresponsive.

Jos-Ven, 2023-03-11:
Thank you. Of course I am willing to help.

Allow me just one question:
In the past:
cforth/src/app/esp8266_rtos and cforth/src/app/esp32 were only present 
in the
WIP branch some time ago they were copied into the master branch.
Does that mean the WIP branch should not be updated anymore?
Note: Some extensions were removed in the main branch.

Jon Smirl wrote: I've compiled with 4.4 and the needed touch ups are minor.
That sounds very good. I hope you will make the update available for us.

Kind regards Jos

jonsmirl, 2023-03-11:
My goal is to integrate forth into esp-matter (https://github.com/espressif/esp-matter) so that it can be used to script the Matter devices. My first try was Micropython, but it is too large to share an ESP with Matter (forces PSRAM). I also tried ueforth which I have compiled into my app on IDF 4.4, but now I am missing all of the Arduino libraries so that's a mess. 

My more basic problem is using cforth 'make' to copy the right source files into the IDF component. I am lacking knowledge of exactly what to copy over.

End goal is to do 'idf.py add-component cforth' and be done with it. Then have a readme at the top of the component explaining how to use it as a component.

Changes were minor. Note I did not get to the point of testing, the build system doesn't support the esp32s3.
~~~
 cell my_lwip_write(cell handle, cell len, void *adr)
 {
-    return (cell)lwip_write_r((int)handle, adr, (size_t)len);
+    return (cell)lwip_writev((int)handle, adr, (size_t)len);
 }
 
 cell my_lwip_read(cell handle, cell len, void *adr)
 {
-    return (cell)lwip_read_r((int)handle, adr, (size_t)len);
+    return (cell)lwip_readv((int)handle, adr, (size_t)len);
 }

diff --git a/src/app/esp32/interface.h b/src/app/esp32/interface.h
index d59a331..52ab719 100755
--- a/src/app/esp32/interface.h
+++ b/src/app/esp32/interface.h
@@ -204,13 +205,14 @@ void gpio_isr_handler_add(gpio_num_t gpio_num, gpio_isr_t isr_handler, void *arg
 void vTaskSuspendAll( void );
 BaseType_t xTaskResumeAll( void );
 
+#ifdef NOTIME
 typedef long time_t;
 typedef long suseconds_t;
 struct timeval {
   time_t      tv_sec;
   suseconds_t tv_usec;
 };
-
+#endif

diff --git a/build/esp32/sdk_build/components/spiffs/mutex.c b/build/esp32/sdk_build/components/spiffs/mutex.c
index 84aea9c..0cff223 100644
--- a/build/esp32/sdk_build/components/spiffs/mutex.c
+++ b/build/esp32/sdk_build/components/spiffs/mutex.c
@@ -34,6 +34,7 @@
 #include "esp_attr.h"
 #include "mutex.h"
 
+extern void _frxt_setup_switch( void );
 
 #define portEND_SWITCHING_ISR(xSwitchRequired) \
 if (xSwitchRequired) {   \

 ~~~
I disabled these because they wouldn't compile
~~~
diff --git a/src/app/esp32/extend.c b/src/app/esp32/extend.c
index 96b6193..254c246 100755
--- a/src/app/esp32/extend.c
+++ b/src/app/esp32/extend.c
@@ -302,6 +302,7 @@ cell ((* const ccalls[])()) = {
   // Like Posix sockets but the socket descriptor space is not
   // merged with the file descriptor space, so you cannot
   // do a  that encompasses both
+  #ifdef SOCKET
        C(lwip_socket)          //c socket         { i.proto i.type i.family -- i.handle }
        C(lwip_bind_r)          //c bind           { i.len a.addr i.handle -- i.error }
        C(lwip_setsockopt_r)    //c setsockopt     { i.len a.addr i.optname i.level i.handle -- i.error }
@@ -321,7 +322,7 @@ cell ((* const ccalls[])()) = {
 
        C(my_select)            //c lwip-select    { i.sec a.exc a.wr a.rd i.n -- i.cnt }
        C(tcpip_adapter_get_ip_info) //c ip-info@  { a.buf i.adapter# -- i.error }
-
+#endif
        C(open_dir)             //c open-dir       { -- a.dir }
        C(closedir)             //c close-dir      { a.dir -- }
        C(next_file)            //c next-file      { a.dir -- a.dirent }
@@ -386,8 +387,9 @@ cell ((* const ccalls[])()) = {
        C(gpio_intr_disable)         //c gpio_intr_disable          { i.handle -- i.err }
 
        C(esp_clk_cpu_freq)          //c esp-clk-cpu-freq           { -- i.hz }
+#ifdef FREQ    
        C(rtc_clk_cpu_freq_set)      //c rtc-clk-cpu-freq-set       { i.freq123 -- }
-
+#endif
        C(esp_now_open)              //c esp-now-open               { i.channel -- i.error? }
        C(esp_now_init)              //c esp-now-init               { -- i.error? }
        C(esp_now_deinit)            //c esp-now-deinit             { -- i.error? }
~~~

jonsmirl, 2023-03-11:
The current build procedure sort of works. But instead of compiling the files into build/esp32 copy them into sdk_main/components/cforth. Then call EspIDF to compile them.  Doing it that way opens the door to the esp32s2, esp32s3, esp32c3 etc compilers.  You can also remove all of the Espressif compiler support from the make system.

Maybe get that architecture first working on idf3.3 with esp32? Then I can help with moving forward onto esp32s3 and IDF 4.4.

jonsmirl, 2023-03-11:
Once everything is set up and working in sdk_main/components/cforth the build procedure can be extended to package the component and upload it to the Espressif component repository.

So cforth 'make' copies the correct stuff into sdk_main/components/cforth.
Next it calls the IDF (just assume it is installed) to build or build manually 'idf.py build'
This allows me to use 'idf.py set-config esp32s3' in the esp app directory.
Test and verify everything working
Then another make step would package the component and push it into the Espressif repository.

jonsmirl, 2023-03-11:
You don't even have to copy the files if you aren't generating them, you can make links.

quozl, 2023-03-11:
@Jos-Ven, I know nothing about a WIP branch, so I could not review https://github.com/MitchBradley/cforth/pull/90 ... are you now proposing that work for the master branch?

@jonsmirl, looks interesting, but I don't use the IDF directly.  I hide it behind platformIO.  As long as the classic and platformIO build methods keep working, I don't really have an issue.

In case you need it, GitHub's markdown has diff highlighting; three backticks followed by the letters "diff" on same line, renders like this;

```diff
 cell my_lwip_write(cell handle, cell len, void *adr)
 {
-    return (cell)lwip_write_r((int)handle, adr, (size_t)len);
+    return (cell)lwip_writev((int)handle, adr, (size_t)len);
 }
```

jonsmirl, 2023-03-12:
The two major differences between IDF3 and IDF4 -- switch from gnu make to cmake. And support for multiple CPU types including RISCV.  There is a python wrapper around everything just to collect the old separate pieces into a single command. I suspect switching over to an IDF component model will reduce the complexity of esp32 support in cforth since you can offload the complex multiple target build work onto the Espressif tools.

quozl, 2023-03-12:
Thanks, that's interesting.  PlatformIO has IDF4 now.  https://docs.platformio.org/en/latest/frameworks/espidf.html  I've not yet tried one of the RISCV boards.  I'll put that on my list.

jonsmirl, 2023-03-12:
It is 32b RISCV as used in the esp32c3 and esp32c6 not the 64b version in the big CPUs.

jonsmirl, 2023-03-12:
I've copied the needed files into my IDF 4.4 project. I can build just a few minor ESPIDF errors which I will touch up tomorrow.  It is complaining that this function is missing a return value, not sure what it wants.
~~~
cell set_esp_now_callback_rcv(QueueHandle_t hQueue)
{
    espnow_queue = hQueue;
    ESP_ERROR_CHECK( esp_now_register_recv_cb(q_data_cb) );
}
~~~
There are a lot of compiler warnings.
~~~
/home/jonsmirl/lowpan/main/forth/extend.c:371:3: note: in expansion of macro 'C'
   C(vTaskPrioritySet)          //c vTaskPrioritySet           { a.prio i.handle -- }
   ^
/home/jonsmirl/lowpan/main/forth/forth.h:100:19: warning: cast between incompatible function types from 'void (*)(cell)' {aka 'void (*)(int)'} to 'cell (*)()' {aka 'int (*)()'} [-Wcast-function-type]
 #define C(cname)  (cell (*)())cname,
                   ^
~~~
Files I copied...
~~~
jonsmirl@ares:~/lowpan/main/forth$ ls
compiler.c  consio.c   extend.c    forth.c   interface.c  prims.h        startapp.c  tmain.c
compiler.h  dict.h     fileio.c    forth.h   interface.h  rodict.c       syscall.c   userarea.h
config.h    dicthdr.h  floatops.c  fprims.h  lineedit.c   specialkeys.h  tdate.c     vars.h
~~~

Jos-Ven, 2023-03-12:
Hi, Thank you Jon for the extensive explanation.
I am under the impression that your changes will replace platfromIO by 
IDF 4.4
In the past I got into a lot of problems with Cforth.
The probable cause is that I am not using a standard developers board as
you can see at: http://home.kpn.nl/~josv/msmarth.html
The problems were solved by using platformIO.
I am afraid that switching back to IDF will get me in trouble again.

James Cameron wrote:
 > I know nothing about a WIP branch, so I could not review #90 ...
That is quite a problem. It means that we can not accept
modifications / improvements for the platforms we do not know about.
 > are you now proposing that work for the master branch?
No that has not been tested. It is just an extension for the WIP branch.
I needed an extra rs232 for a sps30 sensor to measure air quality.

Jos

jonsmirl, 2023-03-12:
I am compiling on the esp32s3 now. The attached file has the minor diffs needed. I achieved this by manually copying the above list of files into my esp32s3 app. This manual copy would be replaced by the component distribution mechanism. There are still a lot of compiler warnings about function signature mismatches. 

Next I have to get it running. This is more complex for me because I want to use forth in an embedded environment without spiffs. My plan is to first enable spiffs in my environment to make sure cforth is working, then I will remove it. For testing I also need to figure out how to make it share the serial port with my app.

[esps3.diff.txt](https://github.com/MitchBradley/cforth/files/10951446/esps3.diff.txt)

~~~diff
diff --git a/main/forth/consio.c b/main/forth/consio.c
index f44f34d..af3f84f 100644
--- a/main/forth/consio.c
+++ b/main/forth/consio.c
@@ -14,7 +14,7 @@ extern void init_filesystem(void);
 int isinteractive() {  return (1);  }
 int isstandalone() {  return (1);  }
 
-void raw_emit(unsigned char c)
+void raw_emit(char c)
 {
     uart_write_bytes(0, &c, 1);
 }
@@ -33,7 +33,7 @@ void cprint(const char *str, cell *up)
 }
 
 u_char key_is_avail = 0;
-u_char the_key;
+char the_key;
 
 int key_avail(cell *up)
 {
diff --git a/main/forth/extend.c b/main/forth/extend.c
index 254c246..6bce278 100755
--- a/main/forth/extend.c
+++ b/main/forth/extend.c
@@ -5,6 +5,8 @@
 #include "compiler.h"
 //#include "i2c-ifce.h"
 #include "interface.h"
+#include <freertos/FreeRTOS.h>
+
 
 cell version_adr(void)
 {
@@ -43,7 +45,7 @@ extern void adc1_get_voltage(void);
 extern void esp_adc_cal_raw_to_voltage(void);
 extern void esp_adc_cal_characterize(void);
 extern void esp_adc_cal_check_efuse(void);
-extern void hall_sensor_read(void);
+//extern void hall_sensor_read(void);
 
 extern void mcpwm_gpio_init(void);
 extern void mcpwm_init(void);
@@ -175,11 +177,11 @@ static QueueHandle_t GpioQueue;
 
 void IRAM_ATTR gpio_qhandler(void *arg)
 {
-  interrupt_disable();
+  portDISABLE_INTERRUPTS();
   int xHigherPriorityTaskWokenByPost=0;
   int qitem=xTaskGetTickCount();
   xQueueGenericSendFromISR(GpioQueue, &qitem, &xHigherPriorityTaskWokenByPost, 0);
-  interrupt_restore();
+  portENABLE_INTERRUPTS();
 }
 
 static void gpio_isr_qhandler_add(int gpio_num, QueueHandle_t hQueue)
@@ -241,6 +243,7 @@ cell set_esp_now_callback_rcv(QueueHandle_t hQueue)
 {
     espnow_queue = hQueue;
     ESP_ERROR_CHECK( esp_now_register_recv_cb(q_data_cb) );
+	return ESP_OK;
 }
 
 // ------------ End Additions
@@ -263,7 +266,7 @@ cell ((* const ccalls[])()) = {
  C(esp_adc_cal_characterize)    //c get-adc-chars { a.adc_chars i.vref i.bi_width i.atten i.adc_num -- i.res }
  C(esp_adc_cal_raw_to_voltage)  //c adc-mv        { a.adc_chars i.reading - i.voltage }
  C(esp_adc_cal_check_efuse)     //c check-efuse   { i.type -- i.res }
-	C(hall_sensor_read)     //c hall@         { -- i.voltage }
+	//C(hall_sensor_read)     //c hall@         { -- i.voltage }
 
 	C(my_uart_param_config)	//c uart-param-config	{ i.flow i.stop i.par i.#bits i.baud i.uart_num -- i.err}
 	C(uart_write_bytes)	//c uart-write-bytes	{ i.size a.src i.uart_num -- i.res }
diff --git a/main/forth/forth.h b/main/forth/forth.h
index 52be9d3..3dff01d 100755
--- a/main/forth/forth.h
+++ b/main/forth/forth.h
@@ -29,7 +29,7 @@ struct stacks {
   cell rp;
   cell rp0;
 };
-void switch_stacks(struct stacks *old, struct stacks *new, cell *up);
+void switch_stacks(struct stacks *old, struct stacks *newp, cell *up);
 
 #define FTHERROR(s)   alerror(s, sizeof(s)-1, up)
 
@@ -70,7 +70,7 @@ void restoremode(void);
 void title(cell *up);
 int lineedit(char *addr, int count, void *up);
 void emit(u_char c, cell *up);
-int system();
+//int system();
 void set_input();
 void exit(int);
 
diff --git a/main/forth/interface.h b/main/forth/interface.h
index 52ab719..13a8a33 100755
--- a/main/forth/interface.h
+++ b/main/forth/interface.h
@@ -62,7 +62,7 @@ void us(cell us);
 
 void esp_deep_sleep(uint64_t time_in_us);
 
-cell esp_get_free_heap_size(void);
+//cell esp_get_free_heap_size(void);
 
 typedef uint32_t TickType_t;
 typedef uint32_t portTickType;
@@ -84,7 +84,7 @@ typedef void (*TaskFunction_t)( void * );
         #define pdTASK_CODE TaskFunction_t
         #define configSTACK_DEPTH_TYPE uint16_t
 
-#define IRAM_ATTR __attribute__((section(".iram1")))
+//#define IRAM_ATTR __attribute__((section(".iram1")))
 int tskNO_AFFINITY = 0x7FFFFFFF;
 
 BaseType_t xTaskCreatePinnedToCore(
@@ -214,7 +214,7 @@ struct timeval {
 };
 #endif
 
-int gettimeofday(struct timeval *__restrict __p, void *__restrict __tz);
+//int gettimeofday(struct timeval *__restrict __p, void *__restrict __tz);
 void esp_sleep_enable_timer_wakeup(uint64_t time_in_us);
 void esp_light_sleep_start();
~~~

MitchBradley, 2023-03-12:
Re char vs unsigned char:

ISTM that a better fix to the raw_emit char/u_char problem is to 
a) move the declaration of raw_emit() from extend.c into interface.h and
b) Change it to `void raw_emit(unsigned char c);`, leaving the implementation in consio.c as unsigned char.  The argument to uart_write_bytes() is void*, so there is no requirement for char* from that end.

Re hall_sensor_read:

a) Commenting out C() lines doesn't always work right because the program that looks for them doesn't necessarily handle C comments properly.  It is better to just delete them and let the diff history tell you that they were once present.
b) ESP-IDF guards hall_sensor_read() with `#if CONFIG_IDF_TARGET_ESP32` so you could use that to supply a null implementation, instead of deleting the function.  Either way is okay.

Re system, esp_get_free_heap_size, IRAM_ATTR, and gettimeofday:
If the problem is that the definitions in interface.h conflict with ones in ESP-IDF headers, the best approach is to just delete the lines instead of commenting them out.  The history will tell you what the lines used to be.  Modern practice is not to leave around commented-out lines, instead relying on source control history to find out what used to be there, if necessary.

jonsmirl, 2023-03-13:
Hall should be ifdef ESP32 since the other chips don't have it. I was just commenting things out to show what I was doing. I will refine things more as I figure out what is going on. Right now I have the more basic problem of needing to rework console IO since my hardware is using USB for serial IO instead of the UART.  It prints a prompt but I can't read keys yet. There is a config option in sdkconfig so cforth can use the esp config variable to select the right console code.  CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
~~~
I (853) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
init_io

CForth built 2023-03-12 02:19 from 5bd25e9
Type a key within 2 seconds to interact
~~~
My current task list is 1) get console working 2) move the large alloc blocks into PSRAM  3) Put spiffs on an IFDEF since it interferes with my other code. I use a different storage system. 4) Define words to call some of my C APIs. 

Once those are done I will be able to fully test out the forth system. I have used forth before, but not since the 80's. My current project needs a very small scripting language and forth should work. The scripts are just a couple of lines, but I need to be able to alter them.  I need to locate the API for loading word definitions, then I can define words and call execute_word() when the events happen.

quozl, 2023-03-13:
So can I hear why there is a WIP branch?  I would expect a master branch to be used for tested and reviewed code, and personal clones to be used for showing work in progress.  I don't mind either way, but not sharing expectations can be confusing.

MitchBradley, 2023-03-13:
I think I created the WIP branch because I was trying out some stuff, and it sort of worked, then someone liked what was in it and based some work off of it.  Maybe I merged it back into master and maybe I didn't; I don't remember.  It's hard to keep track of everything I have going on.

Jos-Ven, 2023-03-14:
My smarthome uses it. The preemtive multitasking part works fine for me.

closed 2023-03-19 by jonsmirl