MitchBradley/cforth #102: 2, operator

jonsmirl, 2023-03-21:
I was trying examples on stack overflow. Is the 2, operator not implemented?
~~~
create myStringArray s" This" 2, s" is" 2, s" an" 2, s" array" 2, s" of" 2, s" strings" 2,
: myString ( u1 -- caddr u1 ) myStringArray swap 2 cells * + 2@ ;
3 myString type
~~~

MitchBradley, 2023-03-21:
I don't remember, but you can easily make it as 
: 2,  ( n1 n2 -- )  here  2 cells allot  2!  ;

jonsmirl, 2023-03-21:
Now there is the issue with s"
~~~
ok : 2, ( n1 n2 -- ) here 2 cells allot 2! ;
ok create myStringArray s" This" 2, s" is" 2, s" an" 2, s" array" 2, s" of" 2, s" strings" 2,
ok : myString ( u1 -- caddr u1 ) myStringArray swap 2 cells * + 2@ ;
ok 3 myString
ok .s
f7d9c9f4 5 
ok type
 * + 
ok 
~~~

jonsmirl, 2023-03-21:
So this stack overflow answer is specific to some forth implementation? maybe gforth?
https://stackoverflow.com/questions/49533788/sorting-an-array-of-strings-in-forth

I need to figure out how to identify these variations.

jonsmirl, 2023-03-21:
Thanks. your string-array version works without issue.
~~~
jonsmirl@ares:~/cforth/build/esp32$ ./forth
C Forth  Copyright (c) 2008 FirmWorks
ok fl ../../src/lib/stringar.fth
ok string-array tcp-errors
ok   ," No error"
ok   ," Out of memory"
ok   ," Buffer error"
ok   ," Timeout"
ok   ," Routing"
ok   ," In progress"
ok   ," Illegal value"
ok   ," Would block"
ok   ," Connection aborted"
ok   ," Connection reset"
ok   ," Connection closed"
ok   ," Not connected"
ok   ," Illegal argument"
ok   ," Already connected"
ok end-string-array
ok 5 tcp-errors count type
In progress
ok 7 tcp-errors count type 
Would block
ok 
~~~

closed 2023-03-21 by jonsmirl

MitchBradley, 2023-03-21:
I have no idea of the context that the person who answered the stackoverflow thing was working from.  He might have just been guessing.  2, is not in [the standard](https://forth-standard.org/standard/words) but its definition is obvious and I'm pretty sure that it is present in quite a few implementations.  S" persistence characteristics, though, are clearly set out in the standard.  The [core version of S"](https://forth-standard.org/standard/core/Sq) covers the case where S" is compiled inside another word; in that situation the string is persistent because the enclosing word provides a persistent container to hold the string data.  The [file extension version of S"](https://forth-standard.org/standard/file/Sq) covers the outside-a-definition case by saying "Store the resulting string in a *transient* buffer c-addr u."

The reasoning behind this is that Forth grew up in highly memory-constrained environments so there is a strong aversion to doing anything that will consume memory without your explicit knowledge.  In interpret state, when you say S" any number of times on one line, the string data can stay in the already-paid-for input buffer.  You can pass the data to, say, a file word on the same line and all is well.  But on the next input line, the old data is overwritten by the new line of data.  If you want to save S" string data in interpret state, it is up to you to copy it somewhere, perhaps into the dictionary with allot, or into dynamically allocated memory.

This, of course, is much less convenient that languages with garbage collection, but it does let you make very compact systems.