MitchBradley/cforth #103: my non-working LCD

jonsmirl, 2023-03-22:
These lines are enough to stop my LCD from working:
~~~
: load-startup-file  ( -- )  " start" included   ;

: app
   ['] load-startup-file catch drop
   quit
;
~~~
What is this doing when I run 'app'?

MitchBradley, 2023-03-22:
load-startup-file tries to read a file name "start" from the FLASH filesystem.  If you don't have a FLASH filesystem at all, I don't know what will happen because I don't know how you turned it off.  If you have such a filesystem, and there is no "start" file, that should be innocuous because if will fail to find the file, throw an error, and catch will respond to the throw and ignore it.

quit tries to invoke the Forth interactive interpreter on the console.  That will wait for input forever.  Perhaps that is why your LCD is not working, because quit never quits.  https://forth-standard.org/standard/core/QUIT - "Repeat the following".  Yes I know it is a bad name, but history.  It means "quit what you are doing and go back to the interactive interpreter".

jonsmirl, 2023-03-22:
It is the 'quit' word messing with the LCD. I removed everything else.

I will try putting forth in a standalone task. Something 'quit' is doing is some how messing with the LCD GPIO, DMA, interrupts, etc. The LCD is DMA based, it runs without a need for code other than to set it up.

MitchBradley, 2023-03-22:
Hell if I know.  I don't have your hardware.

Why are you calling quit?  Do you mean to run the interactive interpreter?  It blocks waiting for console input.

MitchBradley, 2023-03-22:
If you want to run the interactive interpreter in a task, you will need to add a task yield to the while loop in key().

jonsmirl, 2023-03-22:
I would like to have the interactive interpreter working for debugging. 

Where is the source for quit?  It could be blocking in a way that locks the system up. I am on freertos.

MitchBradley, 2023-03-22:
Per the above, the blocking point is at key().  Re the question about the source code for quit, grep is your friend.  But you probably won't need to look at it because it all boils down to waiting for input, which is key() in consio.c

jonsmirl, 2023-03-22:
I found it
~~~
: quit  ( -- )
   \ XXX We really should clean up any open input files here...
   0 complevel !
   rp0 @ rp!
   interact
   bye
;
~~~
I changed console io to this:
~~~
int key_avail(cell *up)
{
    if (key_is_avail) {
        return (cell)-1;
    }
    if(raw_poll(&the_key)) {
        key_is_avail = 1;
        return (cell)-1;
    }
    return 0;
}

int raw_poll(unsigned char *c)
{
    return uart_read_bytes(0, c, 1, 0);
}
~~~
That doesn't block, the timeout is 0.

But... that hard loop may be blocking the other tasks from running. Let me try with timeout of 1.

jonsmirl, 2023-03-22:
Changing timeout had not effect. I will try putting forth into its own task, but that will take me a while to code up.

If that doesn't work I can look at changing the input loop to sleep and wake up on an interrupt.

jonsmirl, 2023-03-22:
ESP IDF includes a FreeRTOS friendly implementation of linenoise. cforth using the same library, but ESP32 cforth is not using it. So it looks like I need to swap out linedit.c for the linenoise library.

MitchBradley, 2023-03-22:
Have you tried fixing key() like I suggested - twice?

`    while (!key_avail(up)) {
       vTaskDelay(1);
     }
`

jonsmirl, 2023-03-22:
Yes, I tried that a few comments back.

jonsmirl, 2023-03-22:
What I tried was:
 return uart_read_bytes(0, c, 1, 1);

Which should be the same thing

MitchBradley, 2023-03-22:
If you have a timeout of 1, key_avail will block for only 1 ms, then when it returns, key() will just call it again and there will be no yield.  uart_read_bytes does not do an internal task yield, regardless of the timeout.

Here is the documentation for uart_read_bytes(), from <ESP_IDF>/components/driver/include/driver/uart.h:
```
/**
 * @brief UART read bytes from UART buffer
 *
 * @param uart_num UART port number, the max port number is (UART_NUM_MAX -1).
 * @param buf     pointer to the buffer.
 * @param length  data length
 * @param ticks_to_wait sTimeout, count in RTOS ticks
 *
 * @return
 *     - (-1) Error
 *     - OTHERS (>=0) The number of bytes read from UART FIFO
 */
int uart_read_bytes(uart_port_t uart_num, void* buf, uint32_t length, TickType_t ticks_to_wait);
```
No mention of a task yield.

Here is the source code from <ESP_IDF>/components/driver/uart.c:
```
int uart_read_bytes(uart_port_t uart_num, void *buf, uint32_t length, TickType_t ticks_to_wait)
{
    ESP_RETURN_ON_FALSE((uart_num < UART_NUM_MAX), (-1), UART_TAG, "uart_num error");
    ESP_RETURN_ON_FALSE((buf), (-1), UART_TAG, "uart data null");
    ESP_RETURN_ON_FALSE((p_uart_obj[uart_num]), (-1), UART_TAG, "uart driver error");
    uint8_t *data = NULL;
    size_t size;
    size_t copy_len = 0;
    int len_tmp;
    if (xSemaphoreTake(p_uart_obj[uart_num]->rx_mux, (portTickType)ticks_to_wait) != pdTRUE) {
        return -1;
    }
    while (length) {
        if (p_uart_obj[uart_num]->rx_cur_remain == 0) {
            data = (uint8_t *) xRingbufferReceive(p_uart_obj[uart_num]->rx_ring_buf, &size, (portTickType) ticks_to_wait);
            if (data) {
                p_uart_obj[uart_num]->rx_head_ptr = data;
                p_uart_obj[uart_num]->rx_ptr = data;
                p_uart_obj[uart_num]->rx_cur_remain = size;
            } else {
                //When using dual cores, `rx_buffer_full_flg` may read and write on different cores at same time,
                //which may lose synchronization. So we also need to call `uart_check_buf_full` once when ringbuffer is empty
                //to solve the possible asynchronous issues.
                if (uart_check_buf_full(uart_num)) {
                    //This condition will never be true if `uart_read_bytes`
                    //and `uart_rx_intr_handler_default` are scheduled on the same core.
                    continue;
                } else {
                    xSemaphoreGive(p_uart_obj[uart_num]->rx_mux);
                    return copy_len;
                }
            }
        }
        if (p_uart_obj[uart_num]->rx_cur_remain > length) {
            len_tmp = length;
        } else {
            len_tmp = p_uart_obj[uart_num]->rx_cur_remain;
        }
        memcpy((uint8_t *)buf + copy_len, p_uart_obj[uart_num]->rx_ptr, len_tmp);
        UART_ENTER_CRITICAL(&(uart_context[uart_num].spinlock));
        p_uart_obj[uart_num]->rx_buffered_len -= len_tmp;
        uart_pattern_queue_update(uart_num, len_tmp);
        p_uart_obj[uart_num]->rx_ptr += len_tmp;
        UART_EXIT_CRITICAL(&(uart_context[uart_num].spinlock));
        p_uart_obj[uart_num]->rx_cur_remain -= len_tmp;
        copy_len += len_tmp;
        length -= len_tmp;
        if (p_uart_obj[uart_num]->rx_cur_remain == 0) {
            vRingbufferReturnItem(p_uart_obj[uart_num]->rx_ring_buf, p_uart_obj[uart_num]->rx_head_ptr);
            p_uart_obj[uart_num]->rx_head_ptr = NULL;
            p_uart_obj[uart_num]->rx_ptr = NULL;
            uart_check_buf_full(uart_num);
        }
    }

    xSemaphoreGive(p_uart_obj[uart_num]->rx_mux);
    return copy_len;
}
```
Again, no yield.

You need a yield, and the obvious place to put that is in key().  Alternatively, you could try
```
int key_avail(cell *up)
{
    if (key_is_avail) {
        return (cell)-1;
    }
    if(raw_poll(&the_key)) {
        key_is_avail = 1;
        return (cell)-1;
    }
    vTaskDelay(1 / portTICK_PERIOD_MS);
    return 0;
}
```

MitchBradley, 2023-03-22:
I don't know why this would break your LCD, but your code is wrong.  key_avail() is supposed to return 0 when no key is available.  It is okay to do a task yield but it is not okay for key_avail() to wait for a character, even if it yields to other tasks while waiting.

Why won't you try doing it the way I suggest?

MitchBradley, 2023-03-22:
Is it possible that Forth and your LCD code are in the same task?  execute_word("app", up) is never going to return.  It has to be in its own task, and your LCD code in another.

jonsmirl, 2023-03-23:
That comment prompted me into finding the problem. Running the "app" word causes forth to not exit and this tick handler not to run. I knew forth was causing it not to run and I said - fine, no big deal, the LVGL doc says it is the animation tick handler and I'm not using any animations.  Hmm... but what is an LVGL animation? Maybe they consider drawing the initial screen an animation? 

So I moved the tick handler into the forth key_avail loop -- and voila screen works again. So apparently the initial draw of the screen is done by this tick handler. What I was seeing is just the random bits in an uninitialized LCD buffer.

Thanks for the help. The fix is -  I need to split into two tasks.

~~~
    forth();

    while (1) {
        // raise the task priority of LVGL and/or reduce the handler period can improve the performance
        vTaskDelay(pdMS_TO_TICKS(10));
        // The task running lv_timer_handler should have lower priority than that running `lv_tick_inc`
        lv_timer_handler();
    }
~~~
~~~
int key_avail(cell *up)
{
    if (key_is_avail) {
        return (cell)-1;
    }
    if(raw_poll(&the_key)) {
        key_is_avail = 1;
        return (cell)-1;
    }
    vTaskDelay(1 / portTICK_PERIOD_MS);
    lv_timer_handler();

    return 0;
}
~~~

closed 2023-03-23 by jonsmirl