MitchBradley/cforth #105: callback_up

jonsmirl, 2023-03-25:
What callback_up doing, can I have more then one? I am calling execute_word from more than one task. The first task initialized forth and created callback_up;  In a different task I call execute_word and give it that same pointer. But then I type 'bye' on the console it dies here:
0x420aa319: doccall at /home/jonsmirl/lowpan/components/forth/forth.c:1561
this line:     case '\0': ret = *sp++; break;
And then reboots. I suspect this is due to returning into the other task?

So can I have one task running in interactive mode and another task doing execute_word()? There are locks in place around execute_word. In the interactive task I release that lock during the key_avail poll. Can I leave forth sitting in key_avail in one task while another task is doing execute_word()?

MitchBradley, 2023-03-25:
You are using cforth past its design parameters.  The typical usage, and the design goal, is to have Forth be the main routine that does all the work, perhaps using calling into leaf C routines - which do something then return - as necessary.

It is just not set up to be called re-entrantly from multiple tasks.  If I had a specific need to make something like that work in a definite system context, I could probably accomplish it because I know where all the bodies are buried, but at present it is not set up for that in the general case.

I think you need to simplify your assumptions.

MitchBradley, 2023-03-25:
"up" is the base address of the per-task data (the "user area" in historical Forth terminology").  If you know what you are doing, you can use that to achieve some degree of parallelism, but you must be careful to ensure that everything that has to be per-task is allocated in the user area instead of in the Forth dictionary.  This is an advanced use case that does not show up too often, so the nitty gritty details are not well documented or tested.

jonsmirl, 2023-03-25:
My desire was to leave a forth prompt on the UART and then have the device continue to function. In that scenario the prompt is always active.

So plan B I can use the ESP console to start forth and not release the lock in key_avail. Doing that will freeze the device. That should let me do the testing I want to do, but it is not very friendly for a more casual person who just wants to play around.  I will also need to sort out why BYE does not work from the second task.

MitchBradley, 2023-03-25:
The casual person does not use Forth as a worker subroutine in the way you are doing, but rather uses Forth as the "main" program.  Your use scenario is certainly valid, but it is highly unusual in the Forth world, which is why the system is not tuned to make that easy.

jonsmirl, 2023-03-25:
Another idea... if  forth is looping in key_avail, how does forth deal with asynchronous events while in that state? Instead of doing execute_word in my first task, can I send a freertos message to the second task? Is there a way to receive and execute a message like that while waiting in the key_avail loop?

MitchBradley, 2023-03-25:
You could modify key_avail so that, if no key is available, it looks for other events and dispatches based on them.

MitchBradley, 2023-03-25:
In fact that is the way I usually do it when I need to perform other work in the background.

jonsmirl, 2023-03-25:
Thanks, I will give that a try and let you know if I succeed.

closed 2023-03-25 by jonsmirl