MitchBradley/cforth #107: Multiple copies of forth

jonsmirl, 2023-03-29:
If I init forth twice, can I use the two instances independently?
I want two completely isolated instances only sharing the read-only data.

No global variables holding state? All state is in the dynamically allocated memory?

quozl, 2023-03-29:
My gut feel is that it would be a big job, but possible.  You would have to root out each global variable.

MitchBradley, 2023-03-29:
Most state is in the user area.  VARIABLE, USER, VALUE, DEFER are all implemented in the user area.  In order to get a shared variable, you pretty much have to use CREATE , DOES> @

That said, there are things that can trip you up, like pictured numeric output and the input buffer.

MitchBradley, 2023-03-29:
init_compiler is the thing that establishes the addresses of all the RAM areas.  It is called by prepare_dictionary which is called by init_forth.  You should study how it works and figure out how to call it twice, or make a second variant of it.

jonsmirl, 2023-03-29:
This second copy is going to talk to Google Real-time DB, it is not user facing. It is going to monitor a bunch of things and then log them up to Google. Google RTDB has a REST interface and it is only going to put things into the db. Phone apps are used to look at the data. The things it is monitoring are already protected from multitasking issues.