MitchBradley/cforth #120: Unable to se pins 23 and 33 on ESP32 ( M5STACK)

jemo07, 2023-09-10:
Hello Mitch, first of all, thank you for this amazing forth, I can't believe I did not find it until now! 

I having an issue with setting up the M5STACK to use it's built in SPI ILI9342/41 

This is as far as I got ( well after bouncing on the ESP32 lack of good documentation just to find the registers.) 
```forth
\ Define GPIO pins
23 constant MOSI_PIN
18 constant CLK_PIN
14 constant CS_PIN
27 constant DC_PIN
33 constant RST_PIN

\ Set GPIO pins as outputs
MOSI_PIN p-out
CLK_PIN p-out
CS_PIN p-out
DC_PIN p-out
RST_PIN p-out
``` 

I get the following error: 
```
ok \ Set GPIO pins as outputs
ok MOSI_PIN p-out
E (8381457) gpio: GPIO can only be used as input mode
[8381473][E][esp32-hal-gpio.c:130] __pinMode(): GPIO config failed
ok CLK_PIN p-out
[8381485][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
ok CS_PIN p-out
ok DC_PIN p-out
E (8381473) gpio: GPIO can only be used as input mode
[8381489][E][esp32-hal-gpio.c:130] __pinMode(): GPIO config failed
ok RST_PIN p-out
[8751271][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
ok 
```
Did search on the offending code but I don't see the connection ( my limitations in c ) 
https://github.com/espressif/arduino-esp32/blob/720be7599f23ccf26a09bb371af833b44cf4627f/cores/esp32/esp32-hal-gpio.c#L91
I also uploaded to code to another ESP32, and the same issue ( all build with pio ). 

I extracted the following: 
https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf
```text
GP0 = Pin(0)
GP1 = Pin(1) # TXD1, I/O/T (Input/Output/Touch sensor)
GP2 = Pin(2) # GPIO, ADC2_CH2/T2, I/O/T
GP3 = Pin(3) # RXD1, I/O/T
GP4 = Pin(4) # CS TF Card
GP5 = Pin(5) # GPIO, I/O/T
GP6 = Pin(6)
GP7 = Pin(7)
GP8 = Pin(8)
GP9 = Pin(9)
GP10 = Pin(10)
GP11 = Pin(11)
GP12 = Pin(12) # IIS_SK, ADC2_CH5, I/O/T
GP13 = Pin(13) # IIS_WS, ADC2_CH4/T4, I/O/T
GP14 = Pin(14) # CS
GP15 = Pin(15) # IIS_OUT, ADC2_CH3/T3, I/O/T
GP16 = Pin(16) # RXD2, I/O/T, I/O/T
GP17 = Pin(17) # TXD2, I/O/T
GP18 = Pin(18) # ILI9342C LCD CLK, SCK, I/O/T
GP19 = Pin(19) # MISO TF Card, MISO, I/O/T
GP21 = Pin(21) # SDA, Grove Port-A (Red), IPS306 charging/discharging circuit
GP22 = Pin(22) # SCL, Grove Port-A (Red), IPS306 charging/discharging circuit
GP23 = Pin(23) # MOSI TF Card, ILI9342C LCD MOSI/MISO, MOSI, I/O/T
GP25 = Pin(25) # DAC Pin, connected to speaker, DAC_1, ADC2_CH8, I/O/T
GP26 = Pin(26) # Grove Port-B (Black), DAC/ADC, DAC_2, ADC2_CH9, I/O/T
GP27 = Pin(27) # ILI9342C LCD DC
GP32 = Pin(32) # ILI9342C LCD BL
GP33 = Pin(33) # ILI9342C LDC RST
GP34 = Pin(34) # IIS_IN, ADC1_CH6, I
GP35 = Pin(35) # ADC, ADC1_CH7, I
GP36 = Pin(36) # Grove Port-B (Black), DAC/ADC, I
GP37 = Pin(37) # Button C
GP38 = Pin(38) # Button B
GP39 = Pin(39) # Button A
```
and for further interest on the M5stack pinout, this can be found here: 

https://github.com/m5stack/M5Unified

Any pointers would be greatly appreciated.

MitchBradley, 2023-09-10:
Are you in hex mode when you define the pins?  If so, the GPIO numbers are being misinterpreted.  I recommend using explicit base prefixes like #23 (decimal) or $23 (hex).

quozl, 2023-09-10:
What _he_ said.  C Forth defaults to base 16.  ;-)

jemo07, 2023-09-11:
Duhhhhhhhh! Thank you...

closed 2023-09-11 by jemo07