MitchBradley/cforth #123: Building a new App. ESP32

jemo07, 2023-09-20:
Hello all, 

I just did a clean install of the git and the ESP32 IDF. 

I used the pio method of building the app and this works for esp32. 

make 
Getting esp-idf
(cd /home/jemo/DevForth \
&& wget https://github.com/espressif/esp-idf/releases/download/5.1/esp-idf-5.1.zip \
&& echo Unzipping esp-idf \
&& unzip -q esp-idf-5.1.zip \
&& rm esp-idf-5.1.zip \
&& python -m pip install --user -r /home/jemo/esp/esp-idf/requirements.txt \
&& if [ ! -e /home/jemo/esp/esp-idf/tools/toolchain_versions.mk ]; then grep SUPPORTED_TOOLCHAIN /home/jemo/esp/esp-idf/make/project.mk | grep := | sed -e s/SUPPORTED/CURRENT/ -e s/VERSIONS/VERSION/ -e s/crosstool-ng-// -e s/DESC/DESC_SHORT/ >/home/jemo/esp/esp-idf/tools/toolchain_versions.mk; fi \
)
--2023-09-20 15:12:40--  https://github.com/espressif/esp-idf/releases/download/5.1/esp-idf-5.1.zip

This cahnge to the sdk.mk files resolves this part of the error: 
ESP_IDF_ARCHIVE = esp-idf-v$(ESP_IDF_VERSION).zip
ESP_IDF_URL = https://dl.espressif.com/github_assets/espressif/esp-idf/releases/download/v$(ESP_IDF_VERSION)/$(ESP_IDF_ARCHIVE)

the URL is incorrect. 

the second issue is: 
/bin/sh: 6: python: not found
(mkdir -p /home/jemo/DevForth/toolchain-- \
&& cd /home/jemo/DevForth/toolchain-- \
&& wget https://dl.espressif.com/dl/xtensa-esp32-elf-linux64--.tar.gz \
&& tar xvf xtensa-esp32-elf-linux64--.tar.gz \
&& rm xtensa-esp32-elf-linux64--.tar.gz \
)
--2023-09-20 15:26:35--  https://dl.espressif.com/dl/xtensa-esp32-elf-linux64--.tar.gz
Resolving dl.espressif.com (dl.espressif.com)... 108.157.109.52, 108.157.109.22, 108.157.109.92, ...
Connecting to dl.espressif.com (dl.espressif.com)|108.157.109.52|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-09-20 15:26:35 ERROR 404: Not Found.

make: *** [../../src/app/esp32/sdk.mk:46: /home/jemo/DevForth/toolchain--/xtensa-esp32-elf/bin] Error 8
 you can see where the is a place holder `--` that does not fully build out the correct path. 

Well, looks I was not successful, even after setting up direct paths, somethig always breaks trying to build esp32 with make.

Jos-Ven, 2023-09-22:
The following might also interest you:
From: https://docs.espressif.com/projects/esp-idf/en/release-v3.3/get-started-cmake/index.html
The CMake-based build system will become the default build system in ESP-IDF V4.0. 
The existing GNU Make based build system will be deprecated in ESP-IDF V5.0.

quozl, 2023-09-22:
And that's one of the reasons I chose PlatformIO; somebody else is dealing with the complexity of upstream changes, encouraged and supported by a large number of developers who use their work.  All I have to do is wait for someone to do it for me.

Jos-Ven, 2023-09-23:
Version 5.1 is way to complex. Version 3.3.6 was easy.
Just change the version number in the following 2 files:
~/cforth/build/esp32/Makefile
~/cforth/src/app/esp32/sdk.mk

When the SDK is compiled you get a number of questions.
I answered them with the default choice.
Except for:
> Bootloader: 2. Rev 1 (ESP32_REV_MIN_1) (NEW)  2 # My Chip is ESP32-D0WD (revision 1)
Enable SO_LINGER processing (LWIP_SO_LINGER) [N/y/?] (NEW) Y
> 2. Default alloc mode (MBEDTLS_DEFAULT_MEM_ALLOC) (NEW) 2

When the ESP32 boots I see:
bootloader chip revision: 1:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6972
load:0x40078000,len:12664
load:0x40080400,len:6740
entry 0x4008077c
I (73) boot: Chip Revision: 1
I (33) boot: ESP-IDF v3.3.6 2nd stage bootloader
I (33) boot: compile time 12:13:30
I (33) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot:  3 storage          Unknown data     01 82 00110000 000f0000
I (90) boot: End of partition table
I (94) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x28210 (164368) map
I (161) esp_image: segment 1: paddr=0x00038238 vaddr=0x3ffbdb60 size=0x0280c ( 10252) load
I (165) esp_image: segment 2: paddr=0x0003aa4c vaddr=0x40080000 size=0x00400 (  1024) load
I (168) esp_image: segment 3: paddr=0x0003ae54 vaddr=0x40080400 size=0x051bc ( 20924) load
I (185) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x79f78 (499576) map
I (361) esp_image: segment 5: paddr=0x000b9f98 vaddr=0x400855bc size=0x10a0c ( 68108) load
I (390) esp_image: segment 6: paddr=0x000ca9ac vaddr=0x400c0000 size=0x00034 (    52) load
I (402) boot: Loaded app from partition at offset 0x10000
I (403) boot: Disabling RNG early entropy source...

CForth built 2023-09-23 10:04 from 61ada4e
Type a key within 2 seconds to interact
ok
\ Still testing...

jemo07, 2023-09-23:
Thank you @Jos-Ven, I’ll try to reproduce your success.