MitchBradley/cforth #124: ESP32, how to properlu build with PIO and exted.c

jemo07, 2023-09-20:
Hello, 

I can successfully create an ESP32  image using the provided [`src/app/esp32` ](https://github.com/MitchBradley/cforth/tree/master/src/app/esp32) yet when I flash cforth to the board, it cant seem to address the provide words in the `extend.c` file, in this case, [`src/app/esp32/extend.c`](https://github.com/MitchBradley/cforth/blob/master/src/app/esp32/extend.c) which is part of the repository. I have tried @Jos-Ven  repository, and I can't seem to find the words in my build. 
Here is an example of the tested words for the sake of the conversation which I know are not in the dictionary, 
``` C
C(open_dir)		//c open-dir       { -- a.dir }
	C(closedir)		//c close-dir      { a.dir -- }
	C(next_file)		//c next-file      { a.dir -- a.dirent }
	C(dirent_size)		//c file-bytes     { a.dir -- i.size }
	C(dirent_name)		//c file-name      { a.dir -- a.name }
	C(rename_file)		//c rename-file    { $.old $.new -- }
	C(delete_file)		//c delete-file    { $.name -- }
```	
This is the test to try to test for these words: 

``` text
ok 
ok close-dir

close-dir ?
Error at: close-dir |  

Aborted
ok next-file

next-file ?
Error at: next-file |  

Aborted
ok file-name 

file-name ?
Error at: file-name  |  

Aborted
ok delete-file

delete-file ?
Error at: delete-file |  

Aborted
ok 

``` 

I just built the again an came across some warnings  see the snippet : 
``` color 
Compiling .pio/build/esp32/src/cforth/embed/consio.c.o
src/app/embed-linux/tmain.c: In function 'forth':
src/app/embed-linux/tmain.c:12:18: warning: passing argument 1 of 'execute_word' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     execute_word("app", up);  // Call the top-level application word
                  ^~~~~
In file included from src/app/embed-linux/tmain.c:4:
src/cforth/compiler.h:38:31: note: expected 'char *' but argument is of type 'const char *'
 extern int execute_word(char *s, cell *up);
                         ~~~~~~^
Compiling .pio/build/esp32/src/cforth/embed/mallocembed.c.o
In file included from src/cforth/embed/consio.c:12:
src/cforth/embed/consio.c: In function 'read_dictionary':
src/cforth/embed/consio.c:56:56: warning: passing argument 1 of 'alerror' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 void read_dictionary(char *name, cell *up) {  FTHERROR("No file I/O\n");  }
                                                        ^~~~~~~~~~~~~~~
src/cforth/forth.h:34:31: note: in definition of macro 'FTHERROR'
 #define FTHERROR(s)   alerror(s, sizeof(s)-1, up)
                               ^
src/cforth/embed/consio.c:44:20: note: expected 'char *' but argument is of type 'const char *'
 void alerror(char *str, int len, cell *up)
              ~~~~~~^~~
In file included from src/cforth/embed/consio.c:12:
src/cforth/embed/consio.c: In function 'write_dictionary':
[... 89 lines cut ...]
src/cforth/forth.h:34:31: note: in definition of macro 'FTHERROR'
 #define FTHERROR(s)   alerror(s, sizeof(s)-1, up)
                               ^
In file included from src/cforth/forth.c:13:
src/cforth/compiler.h:25:27: note: expected 'char *' but argument is of type 'const char *'
 extern void alerror(char *str, int len, cell *up);
                     ~~~~~~^~~
src/platform/arduino/extend.c: In function 'build_date_adr':
<command-line>: warning: return discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
src/platform/arduino/extend.c:5:12: note: in expansion of macro 'CFORTH_DATE'
     return CFORTH_DATE;
            ^~~~~~~~~~~
src/platform/arduino/extend.c: In function 'version_adr':
<command-line>: warning: return discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
src/platform/arduino/extend.c:10:12: note: in expansion of macro 'CFORTH_VERSION'
     return CFORTH_VERSION;
            ^~~~~~~~~~~~~~
Building .pio/build/esp32/bootloader.bin
Generating partitions .pio/build/esp32/partitions.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
```
Thank you!

quozl, 2023-09-21:
The PlatformIO build uses a smaller extend.c in platform/arduino not the one with the ESP-IDK build.  You can either change the build, or move what you need across.

C Forth is not a software package with a single build method, but a collection of build methods according to requirements of various developers.  The respective developers support their build methods.  You will note a build tree separate to the source tree, and that's the main way we support this style.

Thanks for the warnings.  If you have a fix, let me know.

@Jos-Ven's repository if used should be used with the WIP branch not the master branch.  @Jos-Ven can explain further.

Jos-Ven, 2023-09-21:
Hi, There was a problem with extend.c in the past.
That was solved by modifying: 
src/cforth/forth.c
and  src/lib/xmodem.fth

As far as I could test my repository works when you use:

git clone  https://github.com/Jos-Ven/cforth --branch WIP
cd ~/cforth/build/esp32  
make clean # When things go wrong. EG: renaming a build
COMPORT=/dev/ttyUSB0 make flash

Perhaps **make clean** might solve your #123 problem

jemo07, 2023-09-21:
Hello @Jos-Ven thak you for shining in, maybe there is a step that I might be missing, here is my output, 
``` test
 ✘ jemo@jemo-MBP  ~/DevForth  git clone https://github.com/Jos-Ven/cforth --branch WIP Jos-cforth
Cloning into 'Jos-cforth'...
remote: Enumerating objects: 6560, done.
remote: Counting objects: 100% (1052/1052), done.
remote: Compressing objects: 100% (435/435), done.
remote: Total 6560 (delta 694), reused 887 (delta 612), pack-reused 5508
Receiving objects: 100% (6560/6560), 7.47 MiB | 8.20 MiB/s, done.
Resolving deltas: 100% (4196/4196), done.
``` 
This is the build process: 
``` text 
 jemo@jemo-MBP  ~/DevForth  cd Jos-cforth/build/esp32 
 jemo@jemo-MBP  ~/DevForth/Jos-cforth/build/esp32   WIP  make clean
Getting esp-idf
(cd /home/jemo/DevForth \
&& wget https://github.com/espressif/esp-idf/releases/download/v3.1.4/esp-idf-v3.1.4.zip \
&& echo Unzipping esp-idf \
&& unzip -q esp-idf-v3.1.4.zip \
&& rm esp-idf-v3.1.4.zip \
&& python -m pip install --user -r /home/jemo/DevForth/esp-idf-v3.1.4//requirements.txt \
&& if [ ! -e /home/jemo/DevForth/esp-idf-v3.1.4//tools/toolchain_versions.mk ]; then grep SUPPORTED_TOOLCHAIN /home/jemo/DevForth/esp-idf-v3.1.4//make/project.mk | grep := | sed -e s/SUPPORTED/CURRENT/ -e s/VERSIONS/VERSION/ -e s/crosstool-ng-// -e s/DESC/DESC_SHORT/ >/home/jemo/DevForth/esp-idf-v3.1.4//tools/toolchain_versions.mk; fi \
)
--2023-09-21 19:22:43--  https://github.com/espressif/esp-idf/releases/download/v3.1.4/esp-idf-v3.1.4.zip
Resolving github.com (github.com)... 140.82.121.4
Connecting to github.com (github.com)|140.82.121.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/65899476/93be5400-64f5-11e9-8e47-b591c4f0b6b5?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230921%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230921T172243Z&X-Amz-Expires=300&X-Amz-Signature=19388e80bc7f0073e92359bf437ccd8c03e8a07a387f144f6b6e1def210c40d7&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=65899476&response-content-disposition=attachment%3B%20filename%3Desp-idf-v3.1.4.zip&response-content-type=application%2Foctet-stream [following]
--2023-09-21 19:22:43--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/65899476/93be5400-64f5-11e9-8e47-b591c4f0b6b5?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230921%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230921T172243Z&X-Amz-Expires=300&X-Amz-Signature=19388e80bc7f0073e92359bf437ccd8c03e8a07a387f144f6b6e1def210c40d7&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=65899476&response-content-disposition=attachment%3B%20filename%3Desp-idf-v3.1.4.zip&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.110.133, 185.199.108.133, 185.199.109.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 270785709 (258M) [application/octet-stream]
Saving to: ‘esp-idf-v3.1.4.zip’

esp-idf-v3.1.4.zip  100%[===================>] 258.24M  21.7MB/s    in 12s     

2023-09-21 19:22:55 (21.7 MB/s) - ‘esp-idf-v3.1.4.zip’ saved [270785709/270785709]

Unzipping esp-idf
/bin/sh: 6: python: not found
```

This is the test for python: 

``` text 
 jemo@jemo-MBP  ~/DevForth/Jos-cforth/build/esp32   WIP  python --version
Python 3.10.12
 jemo@jemo-MBP  ~/DevForth/Jos-cforth/build/esp32   WIP  
```

Jos-Ven, 2023-09-21:
Hi, My Python version is: Python 3.11.2

See https://www.debugpoint.com/install-python-3-11-ubuntu/ at the end for installation
then retry the installation with:
make clean 
make
If that fails try to run requirements.txt in esp-idf-v3.1.4 
Like:
pip install -r ../../../esp-idf-v3.1.4/requirements.txt