MitchBradley/cforth #125: Can I run the PlatformIO builds on a Raspberry Pi CM4 with PiOS?

znmeb, 2023-10-20:
I'm trying to build cforth for a Raspberry Pi Pico on a Raspberry Pi CM4 running the latest version of PiOS 64 bit (the one just released, based on Debian "bookworm")

I got the PlatformIO core installed, but when I get to the "pio run" step, it crashes trying to build some packages with "-m32".

```
❯ ~/.platformio/penv/bin/pio run
Processing host_makename (platform: native)
-------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/host_makename/src/cforth/makename.o
gcc: error: unrecognized command-line option ‘-m32’
*** [.pio/build/host_makename/src/cforth/makename.o] Error 1
========================================================== [FAILED] Took 1.78 seconds ==========================================================

Processing host_meta (platform: native)
-------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/host_meta/src/cforth/compiler.o
Compiling .pio/build/host_meta/src/cforth/dictfile.o
gcc: error: unrecognized command-line option ‘-m32’
Compiling .pio/build/host_meta/src/cforth/getc-kbd.o
gcc: error: unrecognized command-line option ‘-m32’
*** [.pio/build/host_meta/src/cforth/compiler.o] Error 1
Compiling .pio/build/host_meta/src/cforth/io.o
*** [.pio/build/host_meta/src/cforth/dictfile.o] Error 1
gcc: error: unrecognized command-line option ‘-m32’
*** [.pio/build/host_meta/src/cforth/getc-kbd.o] Error 1
gcc: error: unrecognized command-line option ‘-m32’
*** [.pio/build/host_meta/src/cforth/io.o] Error 1
========================================================== [FAILED] Took 1.30 seconds ==========================================================

Processing host_forth (platform: native)
-------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/host_forth/src/app/host-serial/extend-posix.o
Compiling .pio/build/host_forth/src/cforth/compiler.o
gcc: error: unrecognized command-line option ‘-m32’
Compiling .pio/build/host_forth/src/cforth/dictfile.o
gcc: error: unrecognized command-line option ‘-m32’
*** [.pio/build/host_forth/src/app/host-serial/extend-posix.o] Error 1
Compiling .pio/build/host_forth/src/cforth/floatops.o
gcc: error: unrecognized command-line option ‘-m32’
*** [.pio/build/host_forth/src/cforth/compiler.o] Error 1
*** [.pio/build/host_forth/src/cforth/dictfile.o] Error 1
gcc: error: unrecognized command-line option ‘-m32’
*** [.pio/build/host_forth/src/cforth/floatops.o] Error 1
========================================================== [FAILED] Took 1.35 seconds ==========================================================

Processing host_makeccalls (platform: native)
-------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/host_makeccalls/src/cforth/makeccalls.o
gcc: error: unrecognized command-line option ‘-m32’
*** [.pio/build/host_makeccalls/src/cforth/makeccalls.o] Error 1
========================================================== [FAILED] Took 1.28 seconds ==========================================================

Processing target_makebi (platform: native)
-------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/target_makebi/src/cforth/embed/makebi.o
gcc: error: unrecognized command-line option ‘-m32’
*** [.pio/build/target_makebi/src/cforth/embed/makebi.o] Error 1
========================================================== [FAILED] Took 1.26 seconds ==========================================================

Environment      Status    Duration
---------------  --------  ------------
host_makename    FAILED    00:00:01.784
host_meta        FAILED    00:00:01.301
host_forth       FAILED    00:00:01.352
host_makeccalls  FAILED    00:00:01.282
target_makebi    FAILED    00:00:01.259
===================================================== 5 failed, 0 succeeded in 00:00:06.978 =====================================================
```

I'm guessing this only runs on `x86_64` Linux at the moment - is that correct? If so, how difficult would it be to make it run on an `arm64` system? The `libc6-dev-*-cross` packages exist; I think it's a matter of detecting the host OS and changing the `gcc` flags.

quozl, 2023-10-20:
Interesting, thanks.  Yes, my tests were on x86_64 Linux.  You need to find the gcc option on your system for building a program for 32-bit, as well as how to run the program.  I don't have a system handy right at the moment.  Have a look at https://jensd.be/1126/linux/cross-compiling-for-arm-or-aarch64-on-debian-or-ubuntu

znmeb, 2023-10-21:
> Interesting, thanks. Yes, my tests were on x86_64 Linux. You need to find the gcc option on your system for building a program for 32-bit, as well as how to run the program. I don't have a system handy right at the moment.

I've found a couple of them but it doesn't look like that's defined in a single place - I'd have to do a full refactoring of the build processes. It'd be easier for me to try to get the builds working on Windows, either native or in MSYS2. WSL and virtual machines are out; they can't access the devices.

MitchBradley, 2023-10-21:
I don't have an ARM64 system either so I don't know what it would take to make it work. My sense is that it is doable, but would require sooner wrangling. Build systems can be very tedious, riddled with obscure "minor" details that can be hard too track down.

MitchBradley, 2023-10-21:
I build with WSL a lot. You don't need to access the target devices during the build process. You just have to make sure that target device access is encapsulated in words that do not get executed until the target app runs on the target system.

znmeb, 2023-10-21:
> I build with WSL a lot. You don't need to access the target devices during the build process. You just have to make sure that target device access is encapsulated in words that do not get executed until the target app runs on the target system.

Ah, OK. That helps a lot! This is the first time I've attempted to use PlatformIO. Is the workflow:

1. Build in WSL. This produces a binary object that needs to be deployed to the microcontroller.,
2. Copy the binary object from the WSL filesystem to the Windows filesystem, and then
3. Deploy and test the binary using PlatformIO on Windows?

quozl, 2023-10-21:
We have multiple build methods in CForth.  PlatformIO isn't the only method, and has only arrived recently.  If PlatformIO can't do what you need, you can try building using the Pico SDK.  Have a look in our build directories for other ways to build CForth.

However, sticking to PlatformIO, and assuming target microcontroller is the Raspberry Pi Pico RP2040, and you've a working Windows system, another path ought to be to use PlatformIO on Windows, with changes to `platformio.ini` and `scripts/` that change my Linux build flags and filesystem paths.

CForth is a complex thing to start with though, because of how it also runs on the build system to make the dictionary.  Have you got a blinky going? ;-)

znmeb, 2023-10-21:
The path of least resistance for me may be to (reluctantly) dual-boot my Windows laptop with Ubuntu 22.04 LTS or even buy a cheap Intel headless SBC as an Arch Linux server. I was going with PlatformIO because it looked like an easier way to go beyond the Raspberry Pi Pico to other boards. That said, I did get the host `cforth` to build with no issues on the Raspberry PiOS machine using `build/host-serial-linux64`.

closed 2023-10-21 by znmeb