MitchBradley/cforth #129: STM32F103C8T6 Build error due to missing extend.c file

hinsech475, 2024-01-06:
H to your support team.
I am currently trying to build CFORTH for an STM32F103C8T6 clone. It was tested with Mecristp-Diagnostic and it has 128K of flash and is a true STM32F108C8T6 with a hidden 64KB of flash memory.

I am builing from an up to date Linux MINT PC on an AMD CPU to wich I added the following:
sudo apt-get update
sudo apt-get -y install gcc-arm-none-eabi
sudo apt-get install gcc-multilib

My CFORTH folder is  /home/christian/Documents/cforth.
I downloaded the file en.stsw-stm32054_v3-6-0.zip and unpacked it in /home/christian/Documents..

My 1st attempt to build failed and gave me the following results:

root@Christian-Mint-CH1:/home/christian/Documents/cforth/build/arm-stm32f103# make
TCC ../../src/platform/arm-stm32f103/startup_stm32f10x.c
TCC ../../../STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_usart.c
TCC ../../../STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_rcc.c
TCC ../../../STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_gpio.c
TCC ../../../STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_adc.c
TCC ../../../STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/STM32F10x_StdPeriph_Driver/src/misc.c
TCC ../../src/platform/arm-stm32f103/systick.c
CC ../../src/cforth/makename.c
CPP ../../src/cforth/forth.c
MAKENAME
TCC ../../src/platform/arm-stm32f103/tmain.c
TCC ../../src/cforth/embed/mallocembed.c
TCC ../../src/platform/arm-stm32f103/consoleio.c
TCC ../../../STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/CMSIS/CM3/DeviceSupport/ST/STM32F10x/system_stm32f10x.c
TCC ../../src/platform/arm-stm32f103/gpio.c
TCC ../../src/platform/arm-stm32f103/adc.c
TCC ../../src/cforth/forth.c
TCC ../../src/cforth/compiler.c
TCC ../../src/cforth/syscall.c
TCC ../../src/cforth/floatops.c
TCC ../../src/cforth/lineedit.c
TCC ../../src/cforth/embed/startapp.c
TCC ../../src/cforth/embed/consio.c
CC ../../src/cforth/forth.c
CC ../../src/cforth/compiler.c
CC ../../src/cforth/syscall.c
CC ../../src/cforth/floatops.c
CC ../../src/cforth/makeccalls.c
CC ../../src/cforth/extend.c
CC ../../src/cforth/main.c
CC ../../src/cforth/io.c
CC ../../src/cforth/nullbi.c
CC ../../src/cforth/dictfile.c
CC ../../src/cforth/mallocl.c
CC ../../src/cforth/lineedit.c
CC ../../src/cforth/linux-kbd.c
MAKING FORTH
CC main.o io.o nullbi.o dictfile.o mallocl.o lineedit.o linux-kbd.o forth.o compiler.o syscall.o floatops.o extend.o -o forth
CC ../../src/cforth/meta.c
CC ../../src/cforth/getc-kbd.c
CC meta.o
./meta ../../src/cforth/interp.fth kernel.dic
./forth kernel.dic ../../src/cforth/load.fth
make: *** Aucune règle pour fabriquer la cible « ../../src/app/arm-stm32f103/extend.c », nécessaire pour « tccalls.fth ». Arrêt.
root@Christian-Mint-CH1:/home/christian/Documents/cforth/build/arm-stm32f103#
I now only get the following error at the end or what I think is the end":
make: *** Aucune règle pour fabriquer la cible « ../../src/app/arm-stm32f103/extend.c », nécessaire pour « tccalls.fth ». Arrêt.
My tests also revealed that the STM32F10x_StdPeriph_Lib library folder has to be named STM32F10x_StdPeriph_Lib_V3.5.0 and should be located in cforth parent folder (../cforth) according to the makefile entry:
STMLIB ?= $(TOPDIR)/../STM32F10x_StdPeriph_Lib_V3.5.0
with TOPDIR=../.. In our case it is: /home/christian/Documents/cforth/build/arm-stm32f103
and it is defined this folder makefile.
I tried STM32F10x_StdPeriph_Lib_V3.5.0 and STM32F10x_StdPeriph_Lib_V3.6.0. Both give the same error.

After some investigation I found that there was no /home/christian/Documents/cforth/src/app/arm-gd32f103/extend.c file.

Since I did not know what this file sould contain for an STM32F103C8T6, I created in there an extend.c file with just the following comment text to allowed me to continue the MAKE process.

In my 2nd attempt to build CFORTH for an STM32F103C8T6, I now get the following resuls.
root@Christian-Mint-CH1:/home/christian/Documents/cforth/build/arm-stm32f103# make clean
root@Christian-Mint-CH1:/home/christian/Documents/cforth/build/arm-stm32f103# make
TCC ../../src/platform/arm-stm32f103/startup_stm32f10x.c
TCC ../../../STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_usart.c
TCC ../../../STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_rcc.c
TCC ../../../STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_gpio.c
TCC ../../../STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/STM32F10x_StdPeriph_Driver/src/stm32f10x_adc.c
TCC ../../../STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/STM32F10x_StdPeriph_Driver/src/misc.c
TCC ../../src/platform/arm-stm32f103/systick.c
CC ../../src/cforth/makename.c
CPP ../../src/cforth/forth.c
MAKENAME
TCC ../../src/platform/arm-stm32f103/tmain.c
TCC ../../src/cforth/embed/mallocembed.c
TCC ../../src/platform/arm-stm32f103/consoleio.c
TCC ../../../STM32F10x_StdPeriph_Lib_V3.5.0/Libraries/CMSIS/CM3/DeviceSupport/ST/STM32F10x/system_stm32f10x.c
TCC ../../src/platform/arm-stm32f103/gpio.c
TCC ../../src/platform/arm-stm32f103/adc.c
TCC ../../src/cforth/forth.c
TCC ../../src/cforth/compiler.c
TCC ../../src/cforth/syscall.c
TCC ../../src/cforth/floatops.c
TCC ../../src/cforth/lineedit.c
TCC ../../src/cforth/embed/startapp.c
TCC ../../src/cforth/embed/consio.c
CC ../../src/cforth/forth.c
CC ../../src/cforth/compiler.c
CC ../../src/cforth/syscall.c
CC ../../src/cforth/floatops.c
CC ../../src/cforth/makeccalls.c
CC ../../src/cforth/extend.c
CC ../../src/cforth/main.c
CC ../../src/cforth/io.c
CC ../../src/cforth/nullbi.c
CC ../../src/cforth/dictfile.c
CC ../../src/cforth/mallocl.c
CC ../../src/cforth/lineedit.c
CC ../../src/cforth/linux-kbd.c
MAKING FORTH
CC main.o io.o nullbi.o dictfile.o mallocl.o lineedit.o linux-kbd.o forth.o compiler.o syscall.o floatops.o extend.o -o forth
CC ../../src/cforth/meta.c
CC ../../src/cforth/getc-kbd.c
CC meta.o
./meta ../../src/cforth/interp.fth kernel.dic
./forth kernel.dic ../../src/cforth/load.fth
./makeccalls <../../src/app/arm-stm32f103/extend.c >tccalls.fth
./forth forth.dic tccalls.fth ../../src/app/arm-stm32f103/app.fth
Test for SFR Passed
Test for RAM Passed

gpio-open ?
Error at:    out_pp #13 gpioc gpio-open  |  to led-gpio
make: *** [../../src/cforth/embed/targets.mk:106 : app.dic] Erreur 1
root@Christian-Mint-CH1:/home/christian/Documents/cforth/build/arm-stm32f103# 

It looks like I need a proper extend.c file. 

I noticed that most /home/christian/Documents/cforth/src/app/xxx-microcontroller/extend.c files do not exist except for the esp8266 and esp32.

*** What sould the proper extend.c file be for an STM32F108C8T6? ***

I appreciate your help.

quozl, 2024-01-06:
`src/platform/arm-stm32f103/extend.c` has `gpio-open`.  I'm not sure why the build is looking into `src/app`.

Jos-Ven, 2024-01-07:
Hi, I am confused by the location of STM32F10x_StdPeriph_Lib_V3.5.0
From the makefile:
TOPDIR=../..
STMLIB ?= $(TOPDIR)/../STM32F10x_StdPeriph_Lib_V3.5.0

When I am in the build directory at:
/home/pi/cf/cforth/build/arm-stm32f103
Then 3 levels UP gets me into:
2.root@acer~/arm-stm32f103$ cd ..
2.root@acer~/build$ cd ..
2.root@acer~/cforth$ cd ..
2.root@acer~/cf$

Then I think make expects the STM32F10x_StdPeriph_Lib_V3.5.0 
to be there and not in the ~/cforth/build directory

@quozl What is your opinion?

MitchBradley, 2024-01-07:
I think that build was set up so that the std periph lib was in a peer directory of the top level cforth directory.  That would make sense given that my computer has a GitHub folder in which I keep various repos from github, including cforth and vendor toolsets.