MitchBradley/cforth #130: Tried to execute a null token

jonsmirl, 2024-02-08:
I have cforth embedded in my esp32 app and I've somehow broken extension functions.  Can you please give me some hints on what to look for? I've probably done something dumb but I can't see what it is.

I am failing on the first call to build-date
~~~
CForth built Tried to execute a null token
~~~
~~~
: .built  ( -- )  'build-date cscount type  ;

: banner  ( -- )
   cr ." CForth built " .built
   ."  from " .commit
   cr
;
~~~
in extend.c
~~~
	C(build_date_adr)       //c 'build-date     { -- a.value }
	C(version_adr)          //c 'version        { -- a.value }
~~~
my debug doesn't print
~~~
cell build_date_adr(void)
{
   ESP_LOGI(TAG, "JDS build addr");
    extern char build_date[];
    return (cell)build_date;
}
~~~
tccalls.fth
~~~
#21 ccall: 'build-date     { -- a.value }
#22 ccall: 'version        { -- a.value }
~~~
I am calling init_forth()
~~~
    init_io(0, (char **)0, (cell *)callback_up);   // Perform platform-specific initialization
    callback_up = (void *)init_forth();
~~~
dict.h and dicthdr.h are there
~~~
-rw-rw-r-- 1 jonsmirl jonsmirl 395136 Feb  8 01:30 dict.h
-rw-rw-r-- 1 jonsmirl jonsmirl     51 Feb  8 01:30 dicthdr.h
~~~
~~~
# Forth settings
idf_build_set_property(COMPILE_OPTIONS "-DBITS32" APPEND)
idf_build_set_property(COMPILE_OPTIONS "-DFLOATING" APPEND)
idf_build_set_property(COMPILE_OPTIONS "-DMORE_FP" APPEND)
idf_build_set_property(COMPILE_OPTIONS "-DNOSYSCALL" APPEND)
idf_build_set_property(COMPILE_OPTIONS "-DTARGET" APPEND)
idf_build_set_property(COMPILE_OPTIONS "-DMAXDICT=0x4000" APPEND)
~~~

quozl, 2024-02-08:
If the debug doesn't print, my first guess is that some other source file is being used where you think extend.c is being used.  In those situations, I use a link map to find where the symbol was resolved.

MitchBradley, 2024-02-09:
Are you trying to execute 'build-date before tccsls.fth is loaded?

jonsmirl, 2024-02-09:
Likely, since tccsls.fth is not part of my system. 

I will try to add it.

jonsmirl, 2024-02-09:
I don't see tccsls.fth anywhere, where is it?

From top of my app.fth

fl misc.fth
fl dl.fth
fl random.fth
fl ilog2.fth
fl tek.fth
fl stringar.fth

jonsmirl, 2024-02-09:
Isn't tccalls.fth getting turned into rodict.c via dict.h and dicthdr.h?

I am calling prepare_builtin_dictionary(), I checked in the debugger.

MitchBradley, 2024-02-09:
tccalls.fth is generated by makeccalls via a build rule, then compiled into app.dic

quozl, 2024-02-09:
I would remove `banner` and use the ok prompt to introspect, with `see` and `dump`, to find out what the state of the built system is.  On the other hand, if you don't have an ok prompt, debugger is the next thing to try.

jonsmirl, 2024-02-09:
I replaced the call to "app" with this, and it works.

So something is wrong with my "app" word.
~~~
    forth_init();
    //execute_word("app", callback_up);
    
    char *cmd =
R"FORTH(
: .commit1  ( -- )  'version cscount type  ;

: .built1  ( -- )  'build-date cscount type  ;

: banner1  ( -- )
   cr ." CForth built " .built1
   ."  from " .commit1
   cr
;

banner1

)FORTH";

    spush((int)cmd, getCallback_up());
    spush(strlen(cmd), getCallback_up());
    execute_word("evaluate", getCallback_up());

~~~

jonsmirl, 2024-02-09:
Is this a valid app.fth to make my 'app' word for esp32?
[app.zip](https://github.com/MitchBradley/cforth/files/14217419/app.zip)

quozl, 2024-02-09:
Nothing surprises me in your app.fth.  It looks just like src/app/esp32/app.fth

jonsmirl, 2024-02-09:
This works....
~~~
    forth_init();
    //execute_word("app", callback_up);
    
    char *cmd =
R"FORTH(
: .commit1  ( -- )  'version cscount type  ;

: .built1  ( -- )  'build-date cscount type  ;

: banner1  ( -- )
   cr ." CForth built " .built1
   ."  from " .commit1
   cr
;

banner1

)FORTH";

    spush((int)cmd, getCallback_up());
    spush(strlen(cmd), getCallback_up());
    execute_word("evaluate", getCallback_up());
~~~
and this gives null token. I don't understand why.
"CForth built Tried to execute a null token"
~~~
void forth_main()
{
    forth_init();
    //execute_word("app", callback_up);

    char *cmd =
R"FORTH(

banner

)FORTH";

    spush((int)cmd, getCallback_up());
    spush(strlen(cmd), getCallback_up());
    execute_word("evaluate", getCallback_up());

    /* A task should NEVER return */
    vTaskDelete(NULL);
}
~~~

quozl, 2024-02-09:
Use `see` to decompile banner and compare with banner1.

jonsmirl, 2024-02-09:
~~~
void forth_interact(void)
{
    ESP_LOGI(TAG, "forth_interact enter");
    execute_word("app", callback_up);
    ESP_LOGI(TAG, "forth_interact exit");
}

void forth_main()
{
    forth_init();
    //execute_word("app", callback_up);

    char *cmd =
R"FORTH(

: .commit1  ( -- )  'version cscount type  ;

: .built1  ( -- )  'build-date cscount type  ;

: banner1  ( -- )
   cr ." CForth built " .built1
   ."  from " .commit1
   cr
;

banner1

see banner
see banner1

)FORTH";

    spush((int)cmd, getCallback_up());
    spush(strlen(cmd), getCallback_up());
    execute_word("evaluate", getCallback_up());

    /* A task should NEVER return */
    vTaskDelete(NULL);
}
~~~
prints this
~~~
CForth built I (699) extend: JDS build addr
2023-03-18 17:31 from I (700) extend: JDS version addr
1a6805a
: banner 
   Tried to execute a null token
: banner1 
   Tried to execute a null token
I (6202) main_task: Returned from app_main()

~~~
~~~

jonsmirl, 2024-02-09:
Maybe I am not including everything I need into rodict.c
~~~
./makeccalls <../extend.c >tccalls.fth
./forth forth.dic tccalls.fth app.fth
./makebi app.dic
~~~

quozl, 2024-02-09:
Using fix-build-warnings branch and PlatformIO, I just built for an Adafruit Feather ESP32, and this is what I get for output;

```log
CForth built 2024-02-09 05:11 from ad1819e
ok see banner
: banner
   ." CForth built " .built ."  from " .commit
   cr
;
ok see .built
: .built
   'build-date cscount type
;
ok see 'build-date
ccall: 'build-date    (Body: $0   ....-a.. )
ok 'build-date 20 dump
           0  1  2  3  4  5  6  7  \/  9  a  b  c  d  e  f  01234567v9abcdef
3f40c2f0  bc 00 00 00 4f 00 00 00  32 30 32 34 2d 30 32 2d  <...O...2024-02-
3f40c300  30 39 20 30 35 3a 31 31  00 61 64 31 38 31 39 65  09 05:11.ad1819e
3f40c310  00 00 00 00 10 45 0d 40  18 45 0d 40 f4 51 0d 40  .....E.@.E.@tQ.@
ok █
```

So yes, it may be something to do with your build.

jonsmirl, 2024-02-09:
It is certainly something to do with my build. I have embedded cforth into a larger ESP32 app as a scripting language. The app uses 'evaluate' to run snippets of forth.  What I'm trying to do now is enable interactive mode and failing in my attempts.

quozl, 2024-02-09:
Have your app call `forth()` rather than `evaluate`?

jonsmirl, 2024-02-09:
forth() is doing the same thing:
~~~
void forth(void)
{
    init_io(0, (char **)0, (cell *)callback_up);   // Perform platform-specific initialization
    callback_up = (void *)init_forth();
    execute_word("app", callback_up);  // Call the top-level application word
}
~~~

My target hardware has more flash to work with now. Tomorrow I will try turning on a file system and running ESP32 cforth the normal way. But first I will have to figure out how to get the cforth git tree off from IDF-3 and onto IDF-5 since my hardware is ESP32S3 and IDF-3 doesn't support the S3.

quozl, 2024-02-09:
Oh, that won't work, you've probably got a forth() that calls execute_word on your app.  Hmm.  Hard to know without seeing your entire build, and that would take a while.

jonsmirl, 2024-02-09:
I suspect the problem lies here:
~~~
./makeccalls <../extend.c >tccalls.fth
./forth forth.dic tccalls.fth app.fth
./makebi app.dic
~~~
The read only dictionary is not including everything it needs, but I don't know how to fix it.

quozl, 2024-02-09:
I can't remember if PlatformIO supports S3 yet.  If so, branch of our master, use PlatformIO to build for S3?

quozl, 2024-02-09:
Can you check that makeccalls, forth, and makebi run without error?  I've sometimes made mistakes in app.fth in a way that prevents app.dic from being written.  Check the file dates and times?

quozl, 2024-02-09:
Does tccalls.fth contain what you expect?

quozl, 2024-02-09:
Also consider a way to test forth.dic ... if it has a problem injected by earlier build, app.dic will almost certainly inherit it.

jonsmirl, 2024-02-09:
tccalls.fth looks fine. I am not getting any errors on generation. 
That is the LVGL API, the scripts draw screens.
~~~
#0 ccall: lv_scr_act		{ -- i.screen }
#1 ccall: lv_scr_load		{ i.screen -- }
#2 ccall: lv_obj_create	{ i.parent -- i.object }
#3 ccall: lv_obj_set_pos	{ i.y i.x i.object -- }
#4 ccall: lv_obj_set_size	{ i.h i.w i.object -- }
#5 ccall: lv_obj_set_width { i.w i.object -- }
#6 ccall: lv_obj_set_user_data { i.user_data i.object -- }
#7 ccall: lv_obj_align	{ i.y i.x i.align i.object -- }
#8 ccall: lv_obj_center	{ i.object -- }
#9 ccall: lv_obj_add_style { i.selector i.style i.object -- }
#10 ccall: lv_obj_add_event_cb { i.user_data i.filter i.event_cb i.object -- i.desc }
#11 ccall: lv_btn_create	{ i.parent -- i.button }
#12 ccall: lv_label_create	{ i.parent -- i.label }
#13 ccall: lv_label_set_text { $text i.label -- }
#14 ccall: lv_label_set_long_mode { i.mode i.label -- }
#15 ccall: get_style 		{ i.index i.page -- i.style }
#16 ccall: lv_style_init 	{ i.style -- }
#17 ccall: lv_style_set_bg_opa { i.opa i.style -- }
#18 ccall: lv_style_set_bg_color { i.color i.style -- }
#19 ccall: lv_style_set_border_width { i.width i.style -- }
#20 ccall: lv_style_set_border_color { i.color i.style -- }
#21 ccall: 'build-date     { -- a.value }
#22 ccall: 'version        { -- a.value }
#23 ccall: ms              { i.ms -- }
#24 ccall: get-msecs       { -- i.ms }
#25 ccall: restart         { -- }
#26 ccall: i2c-open  { i.scl i.sda -- i.error? }
#27 ccall: i2c-close  { -- }
#28 ccall: i2c-write-read { a.wbuf i.wsize a.rbuf i.rsize i.slave i.stop -- i.err? }
#29 ccall: i2c-b@     { i.reg i.slave i.stop -- i.b }
#30 ccall: i2c-b!     { i.value i.reg i.slave -- i.error? }
#31 ccall: i2c-be-w@  { i.reg i.slave i.stop -- i.w }
#32 ccall: i2c-le-w@  { i.reg i.slave i.stop -- i.w }
#33 ccall: i2c-be-w!  { i.value i.reg i.slave -- i.error? }
#34 ccall: i2c-le-w!  { i.value i.reg i.slave -- i.error? }
#35 ccall: gpio-pin@  { i.gpio# -- i.flag }
#36 ccall: gpio-pin!  { i.level i.gpio# -- }
#37 ccall: gpio-toggle { i.gpio# -- }
#38 ccall: gpio-is-output { i.gpio# -- }
#39 ccall: gpio-is-output-open-drain { i.gpio# -- }
#40 ccall: gpio-is-input { i.gpio# -- }
#41 ccall: gpio-is-input-pullup { i.gpio# -- }
#42 ccall: gpio-is-input-pulldown { i.gpio# -- }
#43 ccall: gpio-mode { i.pullup? i.direction i.gpio# -- }
#44 ccall: wifi-mode@ { -- i.mode }
#45 ccall: log-level! { i.level $component -- }
#46 ccall: socket         { i.proto i.type i.family -- i.handle }
#47 ccall: bind           { i.len a.addr i.handle -- i.error }
#48 ccall: setsockopt     { i.len a.addr i.optname i.level i.handle -- i.error }
#49 ccall: getsockopt     { i.len a.addr i.optname i.level i.handle -- i.error }
#50 ccall: connect        { i.len a.adr i.handle -- i.error }
#51 ccall: stream-connect { i.timeout $.portname $.hostname -- i.handle }
#52 ccall: lwip-write     { a.buf i.size i.handle -- i.count }
#53 ccall: lwip-read      { a.buf i.size i.handle -- i.count }
#54 ccall: lwip-close     { i.handle -- }
#55 ccall: lwip-listen    { i.backlog i.handle -- i.handle }
#56 ccall: lwip-accept    { a.addrlen a.addr i.handle -- i.error }
#57 ccall: start-server   { i.port -- i.error }
#58 ccall: lwip-select    { i.sec a.exc a.wr a.rd i.n -- i.cnt }
#59 ccall: open-dir       { -- a.dir }
#60 ccall: close-dir      { a.dir -- }
#61 ccall: next-file      { a.dir -- a.dirent }
#62 ccall: file-bytes     { a.dir -- i.size }
#63 ccall: file-name      { a.dir -- a.name }
#64 ccall: rename-file    { $.old $.new -- }
#65 ccall: delete-file    { $.name -- }
#66 ccall: fs-avail       { -- i.bytes }
#67 ccall: m-emit         { i.char -- }
#68 ccall: errno          { -- i.errno }
#69 ccall: strerror       { i.errno -- $.msg }
#70 ccall: gpio-matrix-out { i.inven i.invout i.fun i.pin -- }
#71 ccall: gpio-matrix-in  { i.invert i.fun i.pin -- }
#72 ccall: mcpwm_gpio_init  { i.gpio# i.io_signal i.pwm# -- e.err? }
#73 ccall: mcpwm_init  { a.conf i.timer# i.pwm# -- e.err? }
#74 ccall: mcpwm_set_frequency  { i.freq i.timer# i.pwm# -- e.err? }
#75 ccall: mcpwm_set_duty_in_us  { i.duty i.op# i.timer# i.pwm# -- e.err? }
#76 ccall: mcpwm_set_duty_type { i.duty# i.op# i.timer# i.pwm# -- i.err? }
#77 ccall: mcpwm_get_frequency { i.timer# i.pwm# -- i.freq }
#78 ccall: mcpwm_set_signal_high { i.op# i.timer# i.pwm# -- i.err? }
#79 ccall: mcpwm_set_signal_low { i.op# i.timer# i.pwm# -- i.err? }
#80 ccall: mcpwm_start { i.timer# i.pwm# -- i.err? }
#81 ccall: mcpwm_stop { i.timer# i.pwm# -- i.err? }
#82 ccall: set-alarm   { i.xt i.ms -- }
#83 ccall: repeat-alarm   { i.xt i.ms -- }
#84 ccall: set-alarm-us   { i.xt i.us -- }
#85 ccall: repeat-alarm-us   { i.xt i.us -- }
#86 ccall: us { i.us -- }
~~~

quozl, 2024-02-09:
I continue to speculate.  Have you added or removed any ccall?  An inconsistent ccall table means the wrong function is called for the number passed.

jonsmirl, 2024-02-09:
I am not sure where forth.dic came from. It is over a year old in my git.

the tccalls are right, this works and prints what is expected.
The problem is more likely in something like forth.dic.
~~~
    forth_init();
    //execute_word("app", callback_up);
    
    char *cmd =
R"FORTH(
: .commit1  ( -- )  'version cscount type  ;

: .built1  ( -- )  'build-date cscount type  ;

: banner1  ( -- )
   cr ." CForth built " .built1
   ."  from " .commit1
   cr
;

banner1

)FORTH";

    spush((int)cmd, getCallback_up());
    spush(strlen(cmd), getCallback_up());
    execute_word("evaluate", getCallback_up());
~~~

jonsmirl, 2024-02-09:
I took forth.dic out of the current cforth tree and it made no difference.

quozl, 2024-02-09:
Guess still have to find out what is different at execution time between `banner1` and `banner`, especially `.built1` and `.built`.

quozl, 2024-02-09:
> I took forth.dic out of the current cforth tree and it made no difference.

Yes, that also happened to me just now.  My build was happening fine but there was no app.dic nor forth.dic file.  Underneath, my system had a gcc upgrade, but the 32-bit package for libgcc was not installed.

jonsmirl, 2024-02-10:
The interpreter is going off into the weeds, I am adding printf to track down why.

    forth_init();
    execute_word("banner", callback_up);
~~~
I (690) rodict: romvars 0x3c0668a5
I (690) rodict: sizeof(romvars) 2b4
I (690) rodict: ramvars 0x3fc93ef8
I (691) rodict: ramdict 0x3c083a68
I (695) rodict: sizeof(token_t) 4
I (699) interpreter: Token 14398
I (703) interpreter: scr 217
I (707) interpreter: Token 1395
I (711) interpreter: scr 222
I (714) interpreter: Token 86

I (718) interpreter: Token 2176
I (722) interpreter: scr 217
I (725) interpreter: Token 2164
I (729) interpreter: scr 217
I (733) interpreter: Token 22
I (736) interpreter: Token 28
I (740) interpreter: Token 2143
I (744) interpreter: scr 217
I (748) interpreter: Token 70
I (751) interpreter: Token 1455
I (755) interpreter: scr 217
I (759) interpreter: Token 13
I (762) interpreter: Token 13
I (766) interpreter: Token 79
I (769) interpreter: Token 5
I (773) interpreter: Token 37
I (777) interpreter: Token 2093
I (780) interpreter: scr 217
I (784) interpreter: Token 2088
I (788) interpreter: scr 218
I (791) interpreter: Token 133
I (795) interpreter: Token 6
I (799) interpreter: Token 5
I (802) interpreter: Token 2088
I (806) interpreter: scr 218
I (810) interpreter: Token 47
I (813) interpreter: Token 2
I (817) interpreter: Token 79
I (821) interpreter: Token 17
I (824) interpreter: Token 79
I (828) interpreter: Token 16
I (831) interpreter: Token 27
I (835) interpreter: Token 16
I (839) interpreter: Token 21
I (842) interpreter: Token 79
I (846) interpreter: Token 1400
I (850) interpreter: scr 222
I (853) interpreter: Token 3216
I (857) interpreter: scr 217
I (861) interpreter: Token 3195
I (865) interpreter: scr 217
I (868) interpreter: Token 13
I (872) interpreter: Token 5
I (875) interpreter: Token 12
I (879) interpreter: Token 79
I (883) interpreter: Token 139
I (886) interpreter: Token 73
I (890) interpreter: Token 49
I (894) interpreter: Token 3185
I (898) interpreter: scr 217
I (901) interpreter: Token 133
I (905) interpreter: Token 1277
I (909) interpreter: scr 221
I (912) interpreter: Token 65
[... 145 lines cut ...]
I (1461) interpreter: Token 85
tI (1465) interpreter: Token 79
I (1468) interpreter: Token 141
I (1472) interpreter: Token 73
I (1476) interpreter: Token 49
I (1480) interpreter: Token 3185
I (1484) interpreter: scr 217
I (1487) interpreter: Token 133
I (1491) interpreter: Token 1277
I (1495) interpreter: scr 221
I (1499) interpreter: Token 65
I (1502) interpreter: Token 1391
I (1506) interpreter: scr 222
I (1510) interpreter: Token 85
 I (1514) interpreter: Token 79
I (1517) interpreter: Token 141
I (1521) interpreter: Token 79
I (1525) interpreter: Token 79
I (1529) interpreter: Token 3683840
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420035cb
0x420035cb: panic_handler at /home/jonsmirl/esp/esp-idf/components/esp_system/port/panic_handler.c:145 (discriminator 3)
~~~

jonsmirl, 2024-02-10:
This is the working banner1 trace:
~~~
tI (4300) interpreter: Token 79
I (4304) interpreter: Token 141
I (4308) interpreter: Token 73
I (4312) interpreter: Token 49
I (4315) interpreter: Token 3185
I (4319) interpreter: scr 217
I (4323) interpreter: Token 133
I (4327) interpreter: Token 1277
I (4331) interpreter: scr 221
I (4334) interpreter: Token 65
I (4338) interpreter: Token 1391
I (4342) interpreter: scr 222
I (4345) interpreter: Token 85
 I (4349) interpreter: Token 79
I (4353) interpreter: Token 141
I (4357) interpreter: Token 79
I (4361) interpreter: Token 79
I (4364) interpreter: Token 14390
I (4368) interpreter: scr 217
I (4372) interpreter: Token 13174
~~~
This is the failing banner trace
~~~
tI (4300) interpreter: Token 79
I (4304) interpreter: Token 141
I (4308) interpreter: Token 73
I (4312) interpreter: Token 49
I (4315) interpreter: Token 3185
I (4319) interpreter: scr 217
I (4323) interpreter: Token 133
I (4327) interpreter: Token 1277
I (4331) interpreter: scr 221
I (4334) interpreter: Token 65
I (4338) interpreter: Token 1391
I (4342) interpreter: scr 222
I (4345) interpreter: Token 85
 I (4349) interpreter: Token 79
I (4353) interpreter: Token 141
I (4357) interpreter: Token 79
I (4361) interpreter: Token 79
I (4364) interpreter: Token 3683840
~~~

From the first trace the token is 14390 which is 0x3836
From the second trace the token is 3683840 which is 0x383600

So an extra bye of zero has somehow gotten into my romdict.

quozl, 2024-02-10:
I agree.  Following written before I saw your last, showing how I validated what you said.

Check your prims.h for the token values.  On my builds, 85 is EMIT, 79 is UNNEST and 141 is PAREN_LOOP.  It's probably the same.  So after it prints the "CForth built " string, returning to caller, the next token is invalid.  That would be `'build-date`, I guess.

I suggest using the host forth against the target's dictionary, and examine the state of the dictionary.  I'm using different build tools, but it should be relevant;

```log
$ .pio/build/host_forth/program .pio/build/target_makebi/target.dic 
C Forth  Copyright (c) 2008 FirmWorks
ok see .built
: .built 
   'build-date cscount type 
;
ok see 'build-date
ccall: 'build-date    (Body: $0   ....-a.. ) 
ok words
... 'version      'build-date   acall:        ...
ok ' 'version .x
f7e6e6c0 
ok ' 'build-date .x
f7e6e6a4 
ok ' acall: .x 
f7e6e670 
ok $f7e6e6c0 $f7e6e6a4 - .x
1c 
ok ' 'build-date $1c dump
           0  1  2  3 \/  5  6  7   8  9  a  b  c  d  e  f  0123v56789abcdef
f7e6e6a0  98 25 00 00 8f 25 00 00  00 00 00 00 2d 61 00 00  .%...%......-a..
f7e6e6b0  00 00 00 27 76 65 72 73  69 6f 6e 08 a5 25 00 00  ...'version.%%..
ok 
```

In there should be the ccall number.  Verify it.  Then next step is to figure out why the number can't be dereferenced and the function executed.

jonsmirl, 2024-02-10:
~~~
jonsmirl@ares:~/l5/components/forth/regen$ ./forth app.dic
C Forth  Copyright (c) 2008 FirmWorks
ok see .built
: .built 
   'build-date cscount type 
;
ok see 'build-date
ccall: 'build-date    (Body: $15   ....-a.. ) 
ok ' 'version .x
f7babe04 
ok ' 'build-date .x
f7babde8 
ok ' acall: .x
f7ba99c0 
ok $f7babe04 $f7babde8 - .x
1c 
ok ' 'build-date $1c dump
           0  1  2  3  4  5  6  7  \/  9  a  b  c  d  e  f  01234567v9abcdef
f7babde0  61 74 65 0b 6f 33 00 00  63 2a 00 00 15 00 00 00  ate.o3..c*......
f7babdf0  2d 61 00 00 00 00 00 27  76 65 72 73 69 6f 6e 08  -a.....'version.
f7babe00  76 33 00 00 63 2a 00 00  16 00 00 00 2d 61 00 00  v3..c*......-a..
ok 
~~~

quozl, 2024-02-10:
Sorry, I've no idea.  It's just like mine.

jonsmirl, 2024-02-10:
is the instruction pointer messed up?

working...
~~~
I (4907) interpreter: scr 221
I (4911) interpreter: Token 65
I (4915) interpreter: Token 1391
I (4919) interpreter: scr 222
I (4922) interpreter: Token 85
 I (4926) interpreter: Token 79
I (4930) interpreter: UNNEST 0x3c059fa5
8d f0 4f 00 61 90 d9 b2 71 4f   --> bytes at *rp
I (4937) interpreter: Token 141
I (4941) interpreter: Token 79
I (4945) interpreter: UNNEST 0x3c058f55
4f 00 80 d9 83 22 69 69 4f 00 
I (4952) interpreter: Token 79
I (4956) interpreter: UNNEST 0x3c083a90
36 80 06 6f 2e 73 4f 62 65 0c 
I (4963) interpreter: Token 14390
~~~
failing
~~~
 I (4926) interpreter: Token 79
I (4930) interpreter: UNNEST 0x3c059fa5
8d f0 4f 00 61 90 d9 b2 71 4f 
I (4937) interpreter: Token 141
I (4941) interpreter: Token 79
I (4945) interpreter: UNNEST 0x3c058f55
4f 00 80 d9 83 22 69 69 4f 00 
I (4952) interpreter: Token 79
I (4956) interpreter: UNNEST 0x3c064e5c
00 00 00 72 00 00 00 00 2d 85 
I (4963) interpreter: Token 3683840
~~~

/*$p unnest */  case UNNEST:
    ESP_LOGI(TAG, "UNNEST %p", *rp);
    for (int i = 0; i < 10; i++)
        printf("%02x ", *(unsigned char *)(*rp + i));
    printf("\n");

jonsmirl, 2024-02-10:
My dicts are here:
I (690) rodict: romvars 0x3c06695d
I (690) rodict: sizeof(romvars) 2b4
I (690) rodict: ramvars 0x3fc93ef8
I (691) rodict: ramdict 0x3c083a68
I (695) rodict: sizeof(token_t) 4

This does not appear to be pointing into a dict....
I (4956) interpreter: UNNEST 0x3c064e5c

quozl, 2024-02-10:
I agree.  Something broke the Forth return stack?

jonsmirl, 2024-02-10:
working
~~~
 I (5289) interpreter: Token 79
I (5293) interpreter: UNNEST 0x3c059fa9
8d f0 4f 00 61 90 d9 b2 71 4f 
0x3c059fa9 0xffffffff 0x3c083a8e 0x3c059f99 0x3c058f59 
I (5305) interpreter: Token 141
I (5309) interpreter: Token 79
I (5312) interpreter: UNNEST 0x3c058f59
4f 00 80 d9 83 22 69 69 4f 00 
0x3c058f59 0x3c083a90 0x3c0586a5 0x3c05874d 0x3c0588cd 
I (5325) interpreter: Token 79
I (5328) interpreter: UNNEST 0x3c083a90
36 80 06 6f 2e 73 4f 62 65 0c 
0x3c083a90 0x3c0586a5 0x3c05874d 0x3c0588cd 0xf7deaff0 
I (5341) interpreter: Token 14390
~~~
broken
~~~
 I (5289) interpreter: Token 79
I (5293) interpreter: UNNEST 0x3c059fa9
8d f0 4f 00 61 90 d9 b2 71 4f 
0x3c059fa9 0xffffffff 0x3c064e5f 0x3c059f99 0x3c058f59 
I (5305) interpreter: Token 141
I (5309) interpreter: Token 79
I (5312) interpreter: UNNEST 0x3c058f59
4f 00 80 d9 83 22 69 69 4f 00 
0x3c058f59 0x3c064e60 0x3c0586a5 0x3c05874d 0x3c0588cd 
I (5325) interpreter: Token 79
I (5328) interpreter: UNNEST 0x3c064e60
00 00 00 72 00 00 00 00 2d 85 
0x3c064e60 0x3c0586a5 0x3c05874d 0x3c0588cd 0xf7db4ff0 
I (5341) interpreter: Token 3683840
ESP-ROM:esp32s3-20210327
~~~

I build the dic like this;
./makeccalls <../extend.c >tccalls.fth
./forth forth.dic tccalls.fth app.fth
./makebi app.dic

The address of 'build-date is not the same between banner and banner1.
It is correct in banner1, and wrong in banner.

So when app.dic was made, how did it know the address of 'build-date?

jonsmirl, 2024-02-10:
My ccall numbers:
#21 ccall: 'build-date     { -- a.value }
#22 ccall: 'version        { -- a.value }

Up above when we did the see
ok see 'build-date
ccall: 'build-date    (Body: $15   ....-a.. ) 

So they match which is why banner1 works.

So why is banner messed up? it is like it is using the wrong ccall numbers

i

quozl, 2024-02-10:
> So when app.dic was made, how did it know the address of 'build-date?

tccalls.fth creates the dictionary entries, which contain the number.  The extern ccalls[] contains the address.

quozl, 2024-02-10:
A link map can confirm the object file and therefore the source file for ccalls.

MitchBradley, 2024-02-10:
Be careful not to put junk in your ccalls table that you think the compiler will ignore.  makeccalls.c might not ignore it the same way as the compiler does.  There is a step to run it through the preprocessor in order to sanitize it, but that is not foolproof.  When in doubt, make the ccalls table "clean", without ifdefs or commented-out sections.

MitchBradley, 2024-02-10:
In particular, commenting out sections of the ccalls table with #ifdef or #if will probably work, but commenting it out with /* .. */ probably will not work.

jonsmirl, 2024-02-10:
I don't think this has anything to do with the external functions, instead it appears to be a problem with the way my app.dic is built. I ran three tests, and in the last one it still fails and it is using no external functions.

One thing I did, I did not build forth.dic myself. Instead I used the cforth tree esp32 build to make it, and then I copied it into my tree.

To me this seems to be implying some sort of issue with rodict.c, or a problem with 'makebi'?

top of my app.fth
~~~
\ Load file for application-specific Forth extensions

fl misc.fth
fl dl.fth
fl random.fth
fl ilog2.fth
fl tek.fth
fl stringar.fth

warning @ warning off
: bye standalone?  if  restart  then  bye  ;
warning !

: .commit  ( -- )  'version cscount type  ;

: .built  ( -- )  'build-date cscount type  ;

: banner  ( -- )
   cr ." CForth built " .built
   ."  from " .commit
   cr
;
~~~
which was compiled like this
~~~
./makeccalls <../extend.c >tccalls.fth
./forth forth.dic tccalls.fth app.fth
./makebi app.dic
~~~

This is of trace of
    execute_word(".built", callback_up);
which works, note that at 752 it calls opcode 176 CCALL
~~~
I (690) rodict: romvars 0x3c0668fd
I (690) rodict: sizeof(romvars) 2b4
I (690) rodict: ramvars 0x3fc93ef8
I (691) rodict: ramdict 0x3c083a68
I (695) rodict: sizeof(token_t) 4
I (699) interpreter: Token 14390
I (703) interpreter: scr 217
I (707) interpreter: Token 13174
I (711) interpreter: scr 10851
I (714) interpreter: Token 10
I (718) interpreter: Token 2585
I (722) interpreter: scr 217
I (725) interpreter: Token 72
I (729) interpreter: Token 79
I (733) interpreter: UNNEST 0x3c06167d
0c 30 b0 4f 00 6c 60 d9 54 8f 
0x3c06167d 0x3c064dc9 0x3fc93b1c 0xfefefefe 0xfefefefe 
I (745) interpreter: Token 12
I (749) interpreter: Token 48
I (752) interpreter: Token 176
I (756) interpreter: doccall -a
I (760) extend: JDS build addr
I (763) interpreter: Token 79
I (767) interpreter: UNNEST 0x3c064dc9
be 78 4f 00 6e 36 d9 73 80 0d 
0x3c064dc9 0x3fc93b1c 0xfefefefe 0xfefefefe 0xaaaab432 
I (779) interpreter: Token 10686
I (783) interpreter: scr 217
[... 752 lines cut ...]
cell *
prepare_builtin_dictionary(int dictmax)
{
    const u_char *p;
    const u_char *lastp;
    u_char *q;
    p = romvars;
    lastp = romvars + sizeof(romvars);
    q = (u_char *)ramvars;

ESP_LOGI(TAG, "romvars %p", p);
ESP_LOGI(TAG, "sizeof(romvars) %x", sizeof(romvars));
ESP_LOGI(TAG, "ramvars %p", q);
    while (p < lastp) {
        *q++ = *p++;
    }
    ramdict = heap_caps_malloc(MAXDICT, MALLOC_CAP_SPIRAM);
ESP_LOGI(TAG, "ramdict %p", ramdict);
ESP_LOGI(TAG, "sizeof(token_t) %x", sizeof(token_t));
    init_compiler(romdict, ramdict, 
                  (token_t)(sizeof(romdict) / sizeof(token_t)),
                  ramdict, ramdict + MAXDICT, ramvars);
    return ramvars;
}
~~~

quozl, 2024-02-11:
Starting to smell like a rogue write somewhere.  Try dumping the ccalls table and dictionary fragments at interpreter start and at the time the bad token is encountered?

jonsmirl, 2024-02-12:
I'll start a new issue with better info

closed 2024-02-12 by jonsmirl