MitchBradley/cforth #131: PAREN_LOOP on ESP32S3?

jonsmirl, 2024-02-12:
Is PAREN_LOOP not working for the ESP32S3? Is this line doing what it is supposed to be doing?
~~~
    // Loop termination: clean up return stack and skip branch offset
    rp = (token_t **)  ((char *)rp + 2*sizeof(cell) + sizeof(branch_t *));
~~~
~~~
I (3612) interpreter: Token 1391
I (3616) interpreter: switch(token) *rp 0x3c059fa9 ip 0x3c059f29
0x3c059fa9 0xffffffff 0x3c064e5f 0x3c059f99 0x3c058f59 
I (3628) interpreter: scr 222
I (3631) interpreter: Token 85
I (3635) interpreter: switch(token) *rp 0x3c059fa9 ip 0x3c059f29
0x3c059fa9 0xffffffff 0x3c064e5f 0x3c059f99 0x3c058f59 
 I (3647) interpreter: Token 79
I (3650) interpreter: switch(token) *rp 0x3c059fa9 ip 0x3c059f2d
0x3c059fa9 0xffffffff 0x3c064e5f 0x3c059f99 0x3c058f59 
I (3662) interpreter: UNNEST *rp 0x3c059fa9 ip 0x3c059f2d
0x3c059fa9 0xffffffff 0x3c064e5f 0x3c059f99 0x3c058f59 
I (3673) interpreter: Token 141
I (3677) interpreter: switch(token) *rp 0xffffffff ip 0x3c059fad
0xffffffff 0x3c064e5f 0x3c059f99 0x3c058f59 0x3c064e60 
I (3689) interpreter: PAREN_LOOP *rp 0x3c058f59 ip 0x3c059fb1
0x3c058f59 0x3c064e60 0x3fc93b1c 0xfefefefe 0xfefefefe 
I (3700) interpreter: Token 79
I (3704) interpreter: switch(token) *rp 0x3c058f59 ip 0x3c059fb5
0x3c058f59 0x3c064e60 0x3fc93b1c 0xfefefefe 0xfefefefe 
I (3715) interpreter: UNNEST *rp 0x3c058f59 ip 0x3c059fb5
0x3c058f59 0x3c064e60 0x3fc93b1c 0xfefefefe 0xfefefefe 
I (3726) interpreter: Token 79
I (3730) interpreter: switch(token) *rp 0x3c064e60 ip 0x3c058f5d
0x3c064e60 0x3fc93b1c 0xfefefefe 0xfefefefe 0xaaaa8432 
I (3742) interpreter: UNNEST *rp 0x3c064e60 ip 0x3c058f5d
0x3c064e60 0x3fc93b1c 0xfefefefe 0xfefefefe 0xaaaa8432 
I (3753) interpreter: Token 3683840
I (3757) interpreter: switch(token) *rp 0x3fc93b1c ip 0x3c064e64
0x3fc93b1c 0xfefefefe 0xfefefefe 0xaaaa8432 0x3c083a60 
~~~
where I am printing....
~~~
doprim:
#ifdef JDS
    ESP_LOGI(TAG, "Token %d", token);
    ESP_LOGI(TAG, "switch(token) *rp %p ip %p", *rp, ip);
    for (int i = 0; i < 5; i++)
        printf("%p ", *(rp + i));
    printf("\n");
#endif
    switch (token) {
    case 0:
~~~
~~~

/*$p unnest */  case UNNEST:
#ifdef JDS
    ESP_LOGI(TAG, "UNNEST *rp %p ip %p", *rp, ip);
    for (int i = 0; i < 5; i++)
        printf("%p ", *(rp + i));
    printf("\n");
#endif
 /*$p exit */    case EXIT:    ip = *rp++;     next;
/*$p execute */ case EXECUTE:
~~~
~~~
/*$p (loop) */  case PAREN_LOOP:
    if (++(*(cell *)rp) != 0) {
        branch;
        next;
    }
    // Loop termination: clean up return stack and skip branch offset
    rp = (token_t **)  ((char *)rp + 2*sizeof(cell) + sizeof(branch_t *));
    ++ip;
#ifdef JDS
    ESP_LOGI(TAG, "PAREN_LOOP *rp %p ip %p", *rp, ip);
    for (int i = 0; i < 5; i++)
        printf("%p ", *(rp + i));
    printf("\n");
#endif
    next;
~~~

MitchBradley, 2024-02-12:
I reformatted the dump to make it easier for me to read as shown below, with explanations.  I don't see anything wrong.  What do you think is wrong?
```
Token 1391
switch(token)   ip 59f29  rs: 59fa9 -1 64e5f 59f99 58f59 
scr 222
(token 1391 apparently has no effect on the return stack and does not advance IP)

Token 85
switch(token)   ip 59f29  rs: 59fa9 -1 64e5f 59f99 58f59 
\ token 85 apparently has no effect on the return stack; IP advances by 4

Token 79
switch(token)   ip 59f2d  rs: 59fa9 -1 64e5f 59f99 58f59 
UNNEST          ip 59f2d  rs: 59fa9 -1 64e5f 59f99 58f59 
\ UNNEST pops the return stack into IP and IP advances by 4

Token 141
switch(token)   ip 59fad  rs:       -1 64e5f 59f99 58f59 64e60 
PAREN_LOOP      ip 59fb1  rs:                      58f59 64e60 0x3fc93b1c
\ PAREN_LOOP increments the top of the return stack, turning the -1 into 0
\ Since it is now 0, the loop terminates, removing 3 items from the return
\ stack.  Those items were -remaining_loop_count, limit_value, loop_end_ip
\ IP is advanced by 4

Token 79
switch(token)   ip 59fb5  rs:                      58f59 64e60 0x3fc93b1c
UNNEST          ip 59fb5  rs:                      58f59 64e60 0x3fc93b1c
\ UNNEST pops the return stack into IP and IP advances by 4

Token 79
switch(token)   ip 58f5d  rs:                            64e60 0x3fc93b1c
UNNEST          ip 58f5d  rs:                            64e60 0x3fc93b1c
\ UNNEST pops the return stack into IP and IP advances by 4

Token 3683840
switch(token)   ip 64e64  rs:                                  0x3fc93b1c
```

MitchBradley, 2024-02-12:
The process of sending data dumps of stuff that might be relevant is not converging to a solution.  I think that, in order to solve this, you would need to provide the exact source you are using and the exact steps to trigger the problem, starting from zero.

jonsmirl, 2024-02-12:
0x3c064e60 does not point to a valid forth op code.

Do you have an ESP32S3 dev board?  I can put up a git tree trimmed down to just the forth stuff.

MitchBradley, 2024-02-12:
I have several -S3s.

MitchBradley, 2024-02-12:
Oh, I see, the IP is usually odd whereas it becomes even at the end.  There might be something funny about the way that xtensa represents pointers.  I will investigate.

jonsmirl, 2024-02-12:
I am trying to build forth as a installable component using the Espressif IDF.  I am fine with sharing the code and effort back if you want it.  Once the ESP IDF components works, it should work without change for all Espressif CPUs. The cforth tree is using ESP IDF 3.x and many ESP CPUs are only supported in IDF 4.x and 5.x. I am using IDF 5.1.2

I will start trimming the tree down and put up a URL when I have it tested.

MitchBradley, 2024-02-12:
> Once the ESP IDF components works, it should work without change for all Espressif CPUs.

Well, almost.  There are some differences that are hard to abstract-out.  ESP-IDF does an heroic job, but doesn't always succeed.

jonsmirl, 2024-02-12:
My top suspicion is that there is a struct which need to be marked 'unaligned' and the compiler is aligning it.

MitchBradley, 2024-02-12:
My suspicion  is the opposite - the dictionary starting address needs to be aligned and it is not.

jonsmirl, 2024-02-12:
These are the memory allocation from the same run, romvars is unaligned.
~~~
I (690) rodict: romvars 0x3c066961
I (690) rodict: sizeof(romvars) 2b4
I (690) rodict: ramvars 0x3fc93ef8
I (691) rodict: ramdict 0x3c083a68
I (695) rodict: sizeof(token_t) 4
~~~

jonsmirl, 2024-02-12:
It works with this fix....
~~~
const __attribute__((aligned(32))) u_char romdict[] = {
#include "dict.h"
};
~~~

MitchBradley, 2024-02-12:
bingo

MitchBradley, 2024-02-12:
first alignment problem since about 15 years ago.  Actually, I don't remember the last one.

jonsmirl, 2024-02-12:
Thanks for the help, I'm going to force align all of those includes in romdict.h. It may just be luck the other ones are aligned.  But the compiler is correct, there is no rule that says array of uchar has to be aligned.

How about a component for ESP IDF? is there interest in that? The cforth tree would likely need a script which copies all of the relevant files into the correct location to build the component, and then you'd use the IDF to build it. Right now I have manually copied everything. Maybe your master builds could push it here: https://components.espressif.com/

MitchBradley, 2024-02-12:
A component might be useful - I might use it myself - but I don't have time to support it.  I am spending more than the time I have available supporting FluidNC.

jonsmirl, 2024-02-12:
I will keep slowly working on the component. Right now I am cheating by using the cforth tree to do all of the host bits and then I manually copy them over.  A finished component needs to be self contained.

closed 2024-02-12 by jonsmirl

jonsmirl, 2024-02-12:
BTW, I think what exposed this is that I am on Espressif IDF 5.1.2 and cforth is using IDF 3.1.4. I tried changing the cforth tree to use IDF 5.x but I couldn't easily sort out the build process since the core IDF has changed a lot. The newer compiler in IDF 5.1.2 appears to be able to pack things more efficiently in flash which resulted in the byte aligned arrays.  

cforth itself compiles on IDF 5.1.2 (with this fix), it is just all of the extension function which changed. I don't need most of them and simply deleted the unnecessary ones.

I had been running cforth on the S3 for about a year before I even discovered this issue. I only found it when I turned on interactive mode which uses romvars, previously I was not using any of the romvars functions in my embedded application which runs words via the API.  I am looking at reworking my app from API based to message based. If it is message based I will be able to run interact mode at the same time, something I currently can't do. (the loop waiting for keyboard input will also check for messages).

MitchBradley, 2024-02-12:
Migrating FluidNC from IDF 3 to IDF 5 was a nightmare.

jonsmirl, 2024-02-12:
On thing stopping a full IDF component for cforth is that I don't know how to mix host builds with target builds in ESP CMake.  I don't know how to tell the ESP component Cmake to build makebi, makecalls and forth. I do know how to run the apps, I am missing the knowledge of how to build them mixed in with target compilations. As a work around I build with the cforth tree and then copy the executable into the my tree. I can probably exec() off 'make' and build then that way, but I haven't tried doing that yet. Plus a component in the ESP library has to work on Mac, Windows and Linux so that makefile has to work on all three.

MitchBradley, 2024-02-12:
I don't know how to do that either.  CForth originated in the 80's, constrained by the compiler and build technology of the time, and was optimized for speed and compactness on machines where those things mattered.  Some of the tools like makeccalls could have been written in, for example, sed, but then building on Windows would have been troublesome.  The one cross-platform constant at the time this particular iteration of the build methodology arose was the ability to compile and execute a C program.

I have recently simplified quozl's PlatformIO build method to use Python scripts in place of makename and makeccalls.  That works because, since PlatformIO is written in Python, you can assume the availability of a suitable Python interpreter.

quozl, 2024-02-12:
Yay, alignment.  Well done.