MitchBradley/cforth #136: w/o  open-file

Jos-Ven, 2024-08-14:
Hi, I encountered perhaps a minor deviation from the standard.
The following tests shows it:
 
: show-test ( - )
  s" HELLO.TXT" r/o open-file   
   cr drop pad 200 2 pick read-file drop pad swap type  close-file drop ;

: test-write  ( - )
  s" HELLO.TXT" r/w create-file drop 
  s" Hello   on disk " 2 pick write-file drop close-file drop   
     show-test
  s" HELLO.TXT"    r/w  open-file drop  
  s" Hello 1" 2 pick write-file drop close-file drop   
     show-test
  s" HELLO.TXT"    w/o  open-file drop  
  s" Hello 2" 2 pick write-file drop  close-file drop
     show-test ;    test-write

Cforth:
Hello   on disk
Hello 1 on disk
Hello 1 on disk Hello 2 ok 

Here Cforth on an ESP32 appends Hello 2 when w/o is used.
Instead of overwriting the text at the start of the file. 
Is that intensionally? Should we keep it?

On other systems:
Win32Forth:
Hello   on disk 
Hello 1 on disk 
Hello 2 on disk  ok

Gforth:
Hello   on disk
Hello 1 on disk
Hello 2 on disk  ok

Kind regards Jos

MitchBradley, 2024-08-14:
It seems like a bug.  Please fix it if you want to.

Jos-Ven, 2024-08-24:
Hi, I found a solution.
It will be submitted as soon as when the SDcard extension is ready.

closed 2024-08-24 by Jos-Ven