MitchBradley/cforth #141: lastacf from words in flash memory

Jos-Ven, 2025-03-10:
Hi, When I define the following test in RAM like: 
: test ( - ) [ lastacf ] literal >name$ type ;

Then test shows indeed 'test' and see test shows:
: test
   $3ffb5dac >name$ type
;

When it is flashed then test shows:

: test
   $f7b60430 >name$ type
;
Now test crashes Cforth.
It seems lastacf can not be used here.
Is there a way to make test work in flash memory?

Thanks in advance, Jos

MitchBradley, 2025-03-10:
Can you explain what you are trying to accomplish?   "[ lastacf ] literal" freezes the address of something that is probably in RAM at the time it is created, but then if you try to move it to FLASH, the address would have to be different.   I can't figure out your objective.

MitchBradley, 2025-03-10:
You want something that displays its own name?

Jos-Ven, 2025-03-10:
Hi, Thanks, I made too complicated.
Kind regards, Jos.

closed 2025-03-10 by Jos-Ven

MitchBradley, 2025-03-10:
Just to close this out - the way to do what was originally asked is

: test [ lastacf xtliteral ] >name$ type ;

xtliteral is like literal, but it handles dictionary relocation and is not immediate.

The explicit-name chain is a better and clearer solution to the actual problem.

Jos-Ven, 2025-03-11:
Thanks again. Now my chains also work when they are defined in flash.
The reason I am still working on them is that they are able
to report stack-errors over TCP to another connected Forth-system.